evcc-io/evcc #4597: Easee Home: request canceled (Client Timeout)

josefzahner, 2022-09-27:
### Describe the bug

I'm getting constantly the error below, of course with different Ampere values. Overall it seems to work anyhow.

`max charge current 16A: Post \"https://api.easee.cloud/api/chargers/EHWW2UEM/settings\": net/http: request canceled (Client.Timeout exceeded while awaiting headers)`

[image]

### Steps to reproduce

Charging Value change

### Configuration details

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 10s # control cycle interval

sponsortoken: <my-token>

meters:
- type: template
  template: sma-home-manager 
  usage: grid  
  host: 192.168.20.181  
  name: grid1
- type: template
  template: sma-inverter 
  usage: pv  
  host: 192.168.20.182  
  password: xxx  
  name: pv2
- type: template
  template: sma-inverter 
  usage: pv  
  host: 192.168.20.183  
  password: xxx  
  name: pv3

chargers:
- type: template
  template: easee 
  user: home@xxxxxxxx.xyz
  password: xxx
  charger: EHxxxxxx
  name: wallbox4

loadpoints:
- title: Wallbox
  charger: wallbox4
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  vehicle: GLC350e

vehicles:
- name: GLC350e
  type: mercedes
  title: Mercedes GLC350e
  vin: <my-VIN>
  clientId: <my-clientID>
  clientSecret: <my-secret>
  cache: 5m
  phases: 1

site:
  title: My Site
  meters:
    grid: grid1
    pvs:
    - pv2
    - pv3

hems:
  type: sma
  AllowControl: false

tariffs:
  currency: CHF
  grid:
    type: fixed
    price: 0.2048 # 0,297 [currency]/kWh

  feedin:
    type: fixed
    price: 0.08 # [currency]/kWh

### Log details

[main  ] INFO 2022/09/27 11:16:32 evcc 0.103
[main  ] INFO 2022/09/27 11:16:32 using config file: /etc/evcc.yaml
[main  ] INFO 2022/09/27 11:16:32 listening at :7070
[semp  ] WARN 2022/09/27 11:16:39 SEMP_BASE_URL unspecified, using http://192.168.20.185:7070 instead
[site  ] INFO 2022/09/27 11:16:39 site config:
[site  ] INFO 2022/09/27 11:16:39   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2022/09/27 11:16:39     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2022/09/27 11:16:39     pv 1:      power ✓ energy ✓ currents ✓
[site  ] INFO 2022/09/27 11:16:39     pv 2:      power ✓ energy ✓ currents ✓
[site  ] INFO 2022/09/27 11:16:39   vehicles:
[site  ] INFO 2022/09/27 11:16:39     vehicle 1: range ✓ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2022/09/27 11:16:39 loadpoint 1:
[lp-1  ] INFO 2022/09/27 11:16:39   mode:        pv
[lp-1  ] INFO 2022/09/27 11:16:39   charger:     power ✓ energy ✗ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2022/09/27 11:16:39   meters:      charge ✓
[lp-1  ] INFO 2022/09/27 11:16:39     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2022/09/27 11:16:39 phase timer inactive
[lp-1  ] DEBUG 2022/09/27 11:16:39 pv timer inactive
[lp-1  ] INFO 2022/09/27 11:16:39 vehicle updated: unknown -> Mercedes GLC350e
[lp-1  ] DEBUG 2022/09/27 11:16:39 max charge current: 6A
[main  ] INFO 2022/09/27 11:16:39 ensure the oauth client redirect/callback is configured for Mercedes GLC350e: http://evcc.local:7070/oauth/vehicles/1/callback
[server] FATAL 2022/09/27 11:16:39 remove /tmp/evcc: operation not permitted
[site  ] DEBUG 2022/09/27 11:16:39 ----
[lp-1  ] DEBUG 2022/09/27 11:16:39 charge power: 1448W
[site  ] DEBUG 2022/09/27 11:16:39 pv power: 2862W

### What type of operating system are you running?

Linux

### Version

evcc version 0.103

josefzahner, 2022-09-27:
Momentan funktionierts wieder. Scheint also eher ein temporäres Problem auf Seite easee gewesen zu sein. Wäre trotzdem spannend zu wissen was genau es damit auf sich hat. Auf der Easee Status Seite sehe ich nichts davon, das etwas offline gewesen wäre.

andig, 2022-09-27:
Siehe https://github.com/evcc-io/evcc/discussions/4413

closed 2022-09-27 by andig