evcc-io/evcc #4609: Hyundai "cache" error

VolkerK62, 2022-09-28:
### Describe the bug

Vehicle wird als unavailable angezeigt

### Steps to reproduce

evcc aktualisiert und neu gestartet

### Configuration details

```
vehicles:
- name: auto
  type: template
  template: hyundai
  title: (T)Raumschiff      
  user: xxx  # Benutzername um auf das Gerät zuzugreifen
  password: xxx  # Passwort des Benutzerkontos
  #vin: xxx
  capacity: 72
  language: de            
  phases: 3 
  ##onIdentify:
  mode: pv
  targetSoC: 90
```

### Log details

```
pi@raspberrypi:~ $ evcc vehicle -l trace
[main  ] INFO 2022/09/28 09:13:56 evcc 0.103 (f06922b2)
[main  ] INFO 2022/09/28 09:13:56 using config file: /etc/evcc.yaml
[mqtt  ] INFO 2022/09/28 09:13:57 connecting evcc-398452958 at tcp://localhost:1883
[mqtt  ] DEBUG 2022/09/28 09:13:57 tcp://localhost:1883 connected
.SoC: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'hyundai': 1 error(s) decoding:

* error decoding 'Cache': time: invalid duration "<no value>"
Capacity: 0kWh
```

### What type of operating system are you running?

Linux

### Version

0.103 aktuelles nightly

VolkerK62, 2022-09-28:
downgrade auf 0.103 stable -> Problem behoben

andig, 2022-09-28:
Fehler kommt durch neues Handling "deprecated" Parametern. Lustigerweise sollte dieser Parameter gar nicht deprecated sein- das macht es so verwirrend.

closed 2022-09-28 by andig