evcc-io/evcc #4627: Sungrow Wechselrichter - Energy

Daniel-Peeeh, 2022-09-29:
### Describe the bug

Ich weiß nicht ob dieser Wert überhaupt verwendet wird, aber mir ist aufgefallen dass der absolute Wert der produzierten kWh um eine Kommastelle falsch ausgegeben wird. 

Beispiel:

Im WR: 1500 kWh
EVCC: 150.0 kWh

### Steps to reproduce

-

### Configuration details

- type: template
  name: sungrow
  template: sungrow-inverter
  usage: pv
  modbus: tcpip
  id: 1
  host: XXX
  port: 502

### Log details

sungrow
-------
Power:  3132W
Energy: 150.0kWh

### What type of operating system are you running?

Linux

### Version

evcc version 0.104.1

Daniel-Peeeh, 2022-09-29:
[main  ] INFO 2022/09/29 07:39:30 evcc 0.104.1
[main  ] INFO 2022/09/29 07:39:30 using config file: /etc/evcc.yaml

sungrow
-------
[modbus] TRACE 2022/09/29 07:39:30 modbus: send 00 01 00 00 00 06 01 04 13 a6 00 02
[modbus] TRACE 2022/09/29 07:39:30 modbus: recv 00 01 00 00 00 07 01 04 04 01 af 00 00
[modbus] TRACE 2022/09/29 07:39:30 modbus: send 00 02 00 00 00 06 01 04 13 8b 00 02
[modbus] TRACE 2022/09/29 07:39:30 modbus: recv 00 02 00 00 00 07 01 04 04 06 5f 00 00
Power:  431W
Energy: 163.1kWh

panzerdev, 2022-09-30:
Hallo, ich habe einen Sungrow SG8.0RT Wechselrichter und ich glaube den scale von `0.1` zu entfernen war ein Fehler für das Register `5098`.

Beim `5003` passt das ohne scale.

# Register 5098
[image]

[image]

# Register  5003

[image]

[image]

premultiply, 2022-09-30:
Danke, ändere ich dann dort nochmal.

panzerdev, 2022-09-30:
@premultiply Ich hab leider keinen Sungrow hybrid wechselrichter um das nachzuprüfen... wenn ich demnächst mal einen vor mir hab schau ich mal ob die skalierungen passen.

premultiply, 2022-09-30:
Ist jetzt streng nach Modbus-Dokumentation.
War da vorher nur zu Übereifrig und hab es nicht nochmal mit der Doku gegengecheckt. Letztendlich war nur 5003 falsch.

PR #4643 (2022-09-29): Sungrow: remove energy scale
Fix #4627

/cc @Daniel-Peeeh
PR #4648 (2022-09-30): Sungrow: Fix energy scale
Fix #4627