evcc-io/evcc #4641: maxCurrent1p nur für Zielladen

MarkusGH, 2022-09-29:
openWB mit Phasenumschaltung.
Fahrzeug, das 1-phasig mit 20 A, 2-phasig mit je 16 A Laden kann.
Wenn in der Config  maxCurrent = 20A eingestellt ist funktionieren das Laden und die Phasenumschaltung richtig, aber bei Zielladen wird von einer Ladeleistung von 20A x 2 Phasen = 9,2 kW ausgegangen, real stehen jedoch nur 16A  x 2 Phasen = 7,4 kW zur Verfügung.
Die Folge ist dass das Fahrzeug zu der gewünschten Uhrzeit nicht den gewählten SoC erreicht.

Ich weiß, dass https://github.com/evcc-io/evcc/pull/4072 wegen zu vieler Edge Cases verworfen wurde, die Frage ist, ob man mit maxCurrent1p alleine (ggf. nur für Zielladen verwendet) genauso viele Edge Cases produziert.

patbab, 2022-10-02:
Würde es nicht zumindest im konkreten Fall funktionieren, wenn man einfach maxCurrent = 16A einstellt?
evcc müsste ja trotzdem "hochschalten" können 1p 16A -> 2p 8A bzw. "herunterschalten! 2p 6A -> 1p 12A, und damit das ganze Regelspektrum ausnutzen?

Oder anders gesagt, die 1p 20A zu nutzen, macht ja nur Sinn, wenn kein 2p zur Verfügung steht.

Patrick

MarkusGH, 2022-10-02:
Das ist tatsächlich mein aktueller Workaround, allerdings führt das dazu dass wegen 1A Granularität dann zwischen 3,7 und 4.6 kW nur in 460W Schritten geregelt werden kann. Bei meiner Anlage ist das aber leider tatsächlich bezüglich Überschuss ein häufiger Leistungsbereich.

MarkusGH, 2022-10-02:
Stelle gerade fest dass der von mir gewählte Titel eigentlich falsch ist - wenn man ausschließlich das Problem des falsch berechneten Zielladens lösen möchte müsste man für das Vehicle einen Parameter maxChargePower o.Ä. einführen.

naltatis, 2022-10-29:
Ja, würde ich auch so sehen. Die Zielladeberechnung muss wissen was die maximale Leistung von Wallbox und Fahrzeug ist. Aktuell wird davon ausgegangen, dass das Fahrzeug so schnell laden kann wie an der Wallbox konfiguriertet. Das ist nicht nur ein 1p3p Problem sondern tritt auch auf wenn man an einer 32A@3p Wallbox mit einem 11kW Auto zielgeladen wird. Da muss aktuell dann maxCurrent am Fahrzeug definiert werden damit das richtig klappt.

Eine Alternative zur Konfiguration könnte auch eine Erkennung sein. Der Algorithmus könnte bspw explizit einmal vorher die maximale Ladegeschwindigkeit ausprobieren. Oder wir speichern den max Wert den wir zum Fahrzeug beim "schnell laden" beobachtet haben in die DB und nutzen das dann für den Planner.

MarkusGH, 2022-10-29:
maxCurrent hat halt den Pferdefuß, dass es auch den maximalen Ladestrom bei einphasigem Laden beeinflusst.

premultiply, 2022-10-29:
Sich die Spitzenleistung der Session zu merken könnte ggf. schon mal eine Teilhilfslösung sein. Gute Idee!

MarkusGH, 2022-10-29:
Dann müsste man aber auch die Leistung kurz auf 3p/maxCurrent setzen um herauszufinden was das Auto kann...

andig, 2022-11-03:
> Ich weiß, dass https://github.com/evcc-io/evcc/pull/4072 wegen zu vieler Edge Cases verworfen wurde, die Frage ist, ob man mit maxCurrent1p alleine (ggf. nur für Zielladen verwendet) genauso viele Edge Cases produziert.

Ich sehe dafür keinen guten Case. Eine Sonderfunktion fürs Zielladen ist nicht zielführend (haha). Da sollten wir lieber nochmal drüber nachdenken ob es keine Chance gibt, die Edge Cases zu reduzieren und eine gesamtheitliche Lösung zu bauen.

Wontfix.

closed 2022-11-03 by andig