evcc-io/evcc #4652: möglicher Fehler bei prioritySoC

VolkerK62, 2022-09-30:
### Describe the bug

Ich habe `bufferSoC` gar nicht definiert und `prioritySoC` auf 100 gesetzt, weil ich nicht will, dass die Hausbatterie zum Laden genutzt wird.
Sobald der SoC 100% erreicht hat, verschwindet im Log der Hinweis, dass die Batterie Priorität hat und `battery power` wird zu `site power` addiert.
Siehe log-file (`residualpower` steht auf 200)
-260W battery -230W grid +200W residual = -290W site

Ist im code möglicherweise an der Stelle ein "kleiner als" und nicht ein "kleiner gleich" definiert?

Ich habe jetzt `prioritySoC` auf 101 gesetzt ;-)
Dann hat die Batterie auch mit 100% noch Priorität.

### Steps to reproduce

-

### Configuration details

-

### Log details

```
Sep 30 15:46:15 raspberrypi evcc[1604]: [site  ] DEBUG 2022/09/30 15:46:15 ----
Sep 30 15:46:15 raspberrypi evcc[1604]: [lp-1  ] DEBUG 2022/09/30 15:46:15 charge power: 1700W
Sep 30 15:46:15 raspberrypi evcc[1604]: [site  ] DEBUG 2022/09/30 15:46:15 pv power: 3041W
Sep 30 15:46:15 raspberrypi evcc[1604]: [site  ] DEBUG 2022/09/30 15:46:15 battery power: -260W
Sep 30 15:46:15 raspberrypi evcc[1604]: [site  ] DEBUG 2022/09/30 15:46:15 grid power: -230W
Sep 30 15:46:15 raspberrypi evcc[1604]: [site  ] DEBUG 2022/09/30 15:46:15 battery soc 0: 100%
Sep 30 15:46:15 raspberrypi evcc[1604]: [site  ] DEBUG 2022/09/30 15:46:15 site power: -290W
Sep 30 15:46:15 raspberrypi evcc[1604]: [lp-1  ] DEBUG 2022/09/30 15:46:15 charge currents: [8.1 0 0]A
Sep 30 15:46:15 raspberrypi evcc[1604]: [lp-1  ] DEBUG 2022/09/30 15:46:15 detected phases: 1p
Sep 30 15:46:15 raspberrypi evcc[1604]: [lp-1  ] DEBUG 2022/09/30 15:46:15 charger status: C
Sep 30 15:46:15 raspberrypi evcc[1604]: [lp-1  ] DEBUG 2022/09/30 15:46:15 next soc poll remaining time: 59m49s
Sep 30 15:46:15 raspberrypi evcc[1604]: [lp-1  ] DEBUG 2022/09/30 15:46:15 soc estimated: 86.25% (vehicle: 86.00%)
Sep 30 15:46:15 raspberrypi evcc[1604]: [lp-1  ] DEBUG 2022/09/30 15:46:15 vehicle soc: 86%
Sep 30 15:46:15 raspberrypi evcc[1604]: [lp-1  ] DEBUG 2022/09/30 15:46:15 vehicle range: 410km
Sep 30 15:46:15 raspberrypi evcc[1604]: [lp-1  ] DEBUG 2022/09/30 15:46:15 vehicle target soc: 100%
Sep 30 15:46:15 raspberrypi evcc[1604]: [lp-1  ] DEBUG 2022/09/30 15:46:15 pv charge current: 10.3A = 9A + 1.26A (-290W @ 1p)
Sep 30 15:46:15 raspberrypi evcc[1604]: [lp-1  ] DEBUG 2022/09/30 15:46:15 max charge current: 10A
Sep 30 15:46:15 raspberrypi evcc[1604]: [site  ] DEBUG 2022/09/30 15:46:15 telemetry: charge: Δ5/5Wh @ 1700W
Sep 30 15:46:25 raspberrypi evcc[1604]: [site  ] DEBUG 2022/09/30 15:46:25 ----
```

### What type of operating system are you running?

Linux

### Version

0.104.2

andig, 2022-09-30:
Ich weiss nicht, ob ich die Frage verstehe, aber:

	// if battery is charging below prioritySoC give it priority
	if socs < site.PrioritySoC && batteryPower < 0 {
		site.log.DEBUG.Printf("giving priority to battery charging at soc: %.0f%%", socs)
		batteryPower = 0
	}

Das Problem löst Du, indem Du für `prioritySoC` einfach gar nichts einstellst (=0). Der Defaultwert schaltet das Feature immer ab.

VolkerK62, 2022-09-30:
Ah. Dann hab ich die Doku falsch verstanden. Ich dachte, wenn es deaktiviert ist (= 0), dass die Batterie dann gar keine prio hat.

andig, 2022-09-30:
Dann hab ich wohl Deine Frage nicht verstanden. Ich glaub oben muss ein <= hin, oder?

VolkerK62, 2022-09-30:
Alles gut. Ich dachte, wenn ich es rauslasse, dass die Batterie dann gar keine Prio hat. 
Da habe ich zu kompliziert gedacht: wenn eine Priorität deaktiviert ist, gibt es keine Priorität, also wird die Batterie immer genutzt  ;-)

Und ich dachte, dass der Grenzwert noch dazuzählt, also <=. 
Aber oben heißt es ja "below", also nur <.

Wenn es nciht in der Konfig aufgeführt ist, kann ich es trotzdem (bei Bedarf) per API setzen?

andig, 2022-09-30:
Nee, dafür nicht. Lass uns doch einfach „up to“ draus machen?

VolkerK62, 2022-10-01:
Es gibt kein Problem. Wenn dann gibt es nur ein Problem in der Doku.
Der Text ist verwirrend. Bei beiden (buffer und priority) heißt es, dass oberhalb des angegebenen Level die Batterie genutzt wird. Wenn man es weglässt, steht beim buffer "entspricht > 100%", bei priority steht "entspricht 0%".
Das habe ich falsch interpretiert. Da meine Batterie immer > 0% ist, dachte ich, dass sie dann immer genutzt wird.
In dem Kontekt müsste in der Doku bei priority auch "entspricht > 100%" stehen. 

Aber wenn wir schon mal dabei sind ...
> Wenn ich will dass die Batterie bis einschl. 80% Prio hat stimmt der Code, denn ab 80 wäre sie aus der Prio raus. 

Das "einschl." stimmt dann aber nicht. Wobei das jetzt schon in Richtung Haare spalten geht ;-)

Das sollte im Normalfall kein Problem sein.
Meine Besonderheit ist: Ich bekomme meinen SoC nicht vom Gerät, sonder hole ihn mir (alle 10 Minuten) aus der Webseite des Herstellers (die nur alle 15 Minuten aktualisiert wird). Da habe ich teilweise SoC-Sprünge von mehr als 10%. 

Wegen mir können wir hier zu machen.

closed 2022-10-01 by VolkerK62