evcc-io/evcc #4658: Timeout/Problem während connect zu BMW

manny4566, 2022-09-30:
### Describe the bug

Problem mit BMW API/URL ?
Kann es sein, dass die BMW Schnittstelle Probleme verursacht oder der timeout zu klein ist für die Abfrage ?

vehicle not available: cannot create vehicle 'template': cannot create vehicle 'bmw': cannot get vehicles: Get "https://cocoapi.bmwgroup.com/eadrax-vcs/v1/vehicles?apptimezone=120&appDateTime=1664565216712": dial tcp: lookup cocoapi.bmwgroup.com: i/o timeout

### Steps to reproduce

BMW Fahrzeug anlegen

### Configuration details

vehicles:
- type: template
  template: bmw
  title: BMWi3
  user: xxx
  password: xxx
  vin: xxx
  capacity: 42.2
  name: ev2

### Log details

[main  ] INFO 2022/09/30 21:15:50 evcc 0.104.2
[main  ] INFO 2022/09/30 21:15:50 using config file: /etc/evcc.yaml
[main  ] INFO 2022/09/30 21:15:50 listening at :7070
[lp-1  ] WARN 2022/09/30 21:15:54 locking phase config to 3p for switchable charger
[site  ] INFO 2022/09/30 21:15:54 site config:
[site  ] INFO 2022/09/30 21:15:54   meters:      grid ✓ pv ✗ battery ✗
[site  ] INFO 2022/09/30 21:15:54     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2022/09/30 21:15:54   vehicles:
[site  ] INFO 2022/09/30 21:15:54     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2022/09/30 21:15:54 loadpoint 1:
[lp-1  ] INFO 2022/09/30 21:15:54   mode:        pv
[lp-1  ] INFO 2022/09/30 21:15:54   charger:     power ✓ energy ✗ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2022/09/30 21:15:54   meters:      charge ✓
[lp-1  ] INFO 2022/09/30 21:15:54     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2022/09/30 21:15:54 phase timer inactive
[lp-1  ] DEBUG 2022/09/30 21:15:54 pv timer inactive
[main  ] FATAL 2022/09/30 21:15:54 listen tcp :7070: bind: address already in use
[site  ] DEBUG 2022/09/30 21:15:54 ----
[lp-1  ] DEBUG 2022/09/30 21:15:54 charge power: 0W

### What type of operating system are you running?

Linux

### Version

evcc version 0.104.2

manny4566, 2022-09-30:
Nachtrag:
Folgende Fehler kommen ebenfalls sporadisch:

```
root@evcc-srv:~# evcc -l debug vehicle
[main  ] INFO 2022/09/30 21:24:07 evcc 0.104.2
[main  ] INFO 2022/09/30 21:24:07 using config file: /etc/evcc.yaml
.SoC:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'bmw': cannot get vehicles: unexpected status: 429
Capacity: 0kWh
```

```
root@evcc-srv:~# evcc -l debug vehicle
[main  ] INFO 2022/09/30 21:25:10 evcc 0.104.2
[main  ] INFO 2022/09/30 21:25:10 using config file: /etc/evcc.yaml
.SoC:          unexpected status: 429
Charge status: unexpected status: 429
Range:         unexpected status: 429
Odometer:      unexpected status: 429
Capacity:      42kWh

```

premultiply, 2022-09-30:
> lookup cocoapi.bmwgroup.com: i/o timeout

lookup ->DNS-Problem.
Netzwerkbasisdienste prüfen? Pi-Hole oder anderer eigener DNS-Dienst in Verwendung?

closed 2022-09-30 by premultiply

manny4566, 2022-09-30:
> > lookup cocoapi.bmwgroup.com: i/o timeout
> 
> lookup ->DNS-Problem. Netzwerkbasisdienste prüfen? Pi-Hole oder anderer eigener DNS-Dienst in Verwendung?

Die DNS Auflösung funktioniert:

`root@evcc-srv:~# nslookup cocoapi.bmwgroup.com
Server:         192.168.0.204
Address:        192.168.0.204#53

Non-authoritative answer:
cocoapi.bmwgroup.com    canonical name = btceuprd-apim.azure-api.net.
btceuprd-apim.azure-api.net     canonical name = apimgmttmyqtzzigmxorkjl5gsn6zlzyrlqfjgi6ezdk3hxdfv.trafficmanager.net.
apimgmttmyqtzzigmxorkjl5gsn6zlzyrlqfjgi6ezdk3hxdfv.trafficmanager.net   canonical name = btceuprd-apim-westeurope-01.regional.azure-api.net.
btceuprd-apim-westeurope-01.regional.azure-api.net      canonical name = btceuprd-apim.westeurope.cloudapp.azure.com.
Name:   btceuprd-apim.westeurope.cloudapp.azure.com
Address: 20.82.36.137
`