evcc-io/evcc #4660: ToDo: Get signed phase currents from all gridmeters

premultiply, 2022-09-30:
All (grid) meter templates have to be checked and enhanced to get signed phase currents where possible.

Requirement for load management https://github.com/evcc-io/evcc/pull/3453

Positive sign for import on phase, negative sign for export.

Signed values can be made by mul(phasecurrent, sign(phasepower)) where they are not natively signed.

Hofyyy, 2022-10-27:
Ich kann das für Victron prüfen und testen.

premultiply, 2022-11-11:
@andig Vorschlag: Wir erweitern die Meter so, dass es zusätzlich zu den currents auch optional phasepowers gibt.
Diese werden dann (wenn vorhanden) zusätzlich abgefragt und deren Vorzeichen wird dann auf den Betrag der jeweiligen currents angewendet.
Damit sparen wir uns die komplizierte und umfangreiche Rechnerei innerhalb der Templates, die Prinzipbedingt auch noch zu Mehrfachabfragen der currents führen würde und haben ein deutlich cleaneres Interface.
Zudem können wir bei vielen Metern die entsprechenden Werte wahrscheinlich auch automatisiert über die mbmd-Aliase holen ohne sonstige größere Änderungen.

andig, 2022-11-11:
Geht bestimmt. Wobei sign auch nicht zu Mehrfachabfragen führen dürfte. Hat aber den Vorteil, dass der Rest von evcc mit den bekannten Strömen weiter arbeiten kann.

PR #5539 (2022-12-29): Extended metering
Fix https://github.com/evcc-io/evcc/issues/5369

This PR extends the metering API and adds more optional measurement values ​​to all meters.
It adds **voltages** and **powers** per phase.

**Voltages** per phase can be particularly useful in situations where wallboxes are powered with single or three-phase power (via manual switch-disconnector) without automatic phase switching. The phase configuration present can thus be recognized at any time via the phase voltages present without having to adapt the basic configuration in each case.

The **phase powers** are required becau [... cut]
closed 2022-12-29 by premultiply