evcc-io/evcc #4672: fortlaufender ERROR bei `evcc discuss`

VolkerK62, 2022-10-01:
### Describe the bug

Nach Eingabe von `evcc discuss` öffnet sich ein Browser Fenster. In dem Komandozeilenfenster im Hintergrund läuft währenddessen die angefügte Fehlermeldung, solange das Browserfenster im Vordergrund ist. Holt man das Komandozeilenfenster nach vorne, stoppen die Fehlermeldungen. Schließt man das Browserfenster kommt man wieder zum prompt.

### Steps to reproduce

-

### Configuration details

-

### Log details

```
pi@raspberrypi:~ $ evcc discuss
[main  ] INFO 2022/10/01 18:54:47 evcc 0.104.2 (a9600a4b)
[main  ] INFO 2022/10/01 18:54:47 using config file: /etc/evcc.yaml
[6057:6057:1001/185448.467229:ERROR:component_loader.cc(193)] Failed to parse extension manifest.
[6110:6110:1001/185448.980086:ERROR:gpu_init.cc(486)] Passthrough is not supported, GL is egl, ANGLE is 
[6057:6210:1001/185449.272999:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[6057:6210:1001/185449.273790:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.UPower.GetDisplayDevice: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[6057:6210:1001/185449.274508:ERROR:object_proxy.cc(623)] Failed to call method: org.freedesktop.UPower.EnumerateDevices: object_path= /org/freedesktop/UPower: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.UPower was not provided by any .service files
[6110:6110:1001/185452.617915:ERROR:sync_control_vsync_provider.cc(139)] Calculated bogus refresh interval=0.984313 s, last_timebase_=440936032391 bogo-microseconds, timebase=440937016704 bogo-microseconds, last_media_stream_counter_=440936, media_stream_counter=440937
[6110:6110:1001/185453.603715:ERROR:sync_control_vsync_provider.cc(139)] Calculated bogus refresh interval=0.986255 s, last_timebase_=44
```

### What type of operating system are you running?

Linux

### Version

0.104.2 (a9600a4b)

andig, 2022-10-01:
Kein evcc Fehler ;)

closed 2022-10-01 by andig

VolkerK62, 2022-10-02:
ok. Dann ist das evtl der Grund für mein Verzögerungsproblem.
Hast du einen Reparatur-Tip für mich?

VolkerK62, 2022-10-02:
ist gelöst. Hardwarebeschleunigung in Chromium deaktiviert und schon läuft es flüssig.