evcc-io/evcc #4677: [nightly] panic: runtime error: invalid memory address or nil pointer dereference

ecfmau, 2022-10-02:
### Describe the bug

Suddenly, since this night, the evcc nightly build crashes at startub with the following message:

panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x50 pc=0x155e3b3]

### Steps to reproduce

Just starting evcc produces the crash

### Configuration details

Sponsor token, usernames and passwords have been redacted. This config file worked fine yesterday afternoon.
[evcc.yaml.txt](https://github.com/evcc-io/evcc/files/9691999/evcc.yaml.txt)

### Log details

[evcc-crash.log](https://github.com/evcc-io/evcc/files/9692001/evcc-crash.log)

### What type of operating system are you running?

Docker container

### Version

nightly build (needed for elli wallbox)

andig, 2022-10-02:
Closed in 7920b28a3b787e2d85f069061af1b50a55001999

closed 2022-10-02 by andig

ecfmau, 2022-10-02:
Thanks, evcc is now starting again but the connection to my elli wallbox does not work anymore (tls: failed to parse certificate from server: x509: invalid basic constraints b). I'll analyze some more and open another issue if the problem seems not to be on my setup. Have a nice day.

andig, 2022-10-02:
For time being you're still stuck with the special builds, we're investigating why nightly doesn't work.

ecfmau, 2022-10-02:
Ok then I'll have to wait until the nightly works again, I can't use a special build because I need it in Docker.