evcc-io/evcc #4703: Add the total charged energy per load point to the rest/mqtt API

pail23, 2022-10-03:
**Is your feature request related to a problem? Please describe.**
Currently the chargedEnergy during a loading session is published on the API. It would be very helpful to have a total charged energy (sum of all session) published on the API in order to simplify energy consumption tracking outside evcc. Some Chargers like keba also provide this value on their API.

**Describe the solution you'd like**
Publish a totalChargedEnergy per loadpoint which sums up all energy charged.

andig, 2022-10-03:
That would depend on something like https://github.com/evcc-io/evcc/issues/4159. We can revisit once that is done. For time being wontfix.

pail23, 2022-10-14:
@andig 
Now that #4159 is closed, I ask you to reconsider this issue. I did a quick prototype by adding
`lp.publish("totalChargedEnergy", lp.chargeMeterTotal())`
after 
`lp.publish("chargedEnergy", lp.chargedEnergy)` (currently line 1505 in loadpoint.go)

By adding this line, I achieved what I would like to have. I'm happy to create a PR in case this would help to move it forward. Having the value available through mqtt would be good enough for me, I don't need it in the rest API.

Many thanks for reconsidering.

closed 2022-10-18 by andig