evcc-io/evcc #4716: Mercedes not working anumore

stefaanbolle, 2022-10-04:
### Describe the bug

Since upgrading a couple of nightlies, the Mercedes vehicle is not detected anymore. It used to work fine with this config. The vehicle is an EQA.

### Steps to reproduce

Restart evcc

### Configuration details

evcc.yaml (truncated)
```
network:
  schema: http
  host: 192.168.1.23
  port: 7070

interval: 20s # control cycle interval
sponsortoken: abcd
telemetry: true
plant: Home

vehicles:
- name: Tesla
  type: template
  template: tesla
  title: Tesla
  accessToken: abcd
  refreshToken: efgh
  vin: L1234
  capacity: 82
  phases: 3
  mode: minpv
  minSoC: 20
  targetSoC: 90
  minCurrent: 6
  maxCurrent: 16
  identifiers:
  - 5678
  cloud: false
- name: EQA
  type: mercedes
  title: EQA
  vin: VIN1234
  clientId: abcd
  clientSecret: efgh
  cache: 2m
  onIdentify:
    mode: minpv
    minSoC: 20
    targetSoC: 90
    minCurrent: 6
    maxCurrent: 16
  identifiers:
   - 1234
```

Mercedes Me API
```
API Key: ****************
OAuth Credentials (Authorization Code Flow)
Client ID: abcd
Client Secret: efgh
Redirect URLs
http://192.168.1.23:7070/oauth/vehicles/1/callback
http://192.168.1.29:8091/mb-callback
```

### Log details

evcc vehicle
```
[main  ] INFO 2022/10/04 07:53:25 evcc 0.104.2 (06fa7490)
[main  ] INFO 2022/10/04 07:53:25 using config file: /home/pi/evcc.yaml
Tesla
-------
SoC:           72%
Charge status: C
Target SoC:    90%
Identifiers:   [5678]
OnIdentified:  MaxCurrent:16, MinSoC:20, TargetSoC:90, Mode:minpv, MinCurrent:6

EQA-300
-------
.SoC:         Get "https://api.mercedes-benz.com/vehicledata/v2/vehicles/W1N9N0JB6PJ044333/containers/electricvehicle": oauth2: token expired and refresh token is not set
Range:        Get "https://api.mercedes-benz.com/vehicledata/v2/vehicles/W1N9N0JB6PJ044333/containers/electricvehicle": oauth2: token expired and refresh token is not set
Capacity:     0kWh
Identifiers:  [1234]
OnIdentified: MaxCurrent:16, MinSoC:20, TargetSoC:90, Mode:minpv, MinCurrent:6

```

### What type of operating system are you running?

Linux

### Version

evcc version 0.104.2 (06fa7490)

premultiply, 2022-10-04:
> `[main  ] INFO 2022/10/04 07:53:25 using config file: /home/pi/evcc.yaml`

Wirklich?

stefaanbolle, 2022-10-04:
> 

Nope, real path is different

andig, 2022-10-04:
Geht nur evcc vehicle nicht oder generell auch der Login übers UI?

stefaanbolle, 2022-10-04:
I can login via the WebUI, but the vehicle remains undetected (red SoC bar at 20%). Login works (red circle in menu disappears after login.
However evcc vehicle continues returning
```
.SoC:         Get "https://api.mercedes-benz.com/vehicledata/v2/vehicles/VIN1234/containers/electricvehicle": oauth2: token expired and refresh token is not set
Range:        Get "https://api.mercedes-benz.com/vehicledata/v2/vehicles/VIN1234/containers/electricvehicle": oauth2: token expired and refresh token is not set
```
once connected to (and badged at) the charger

andig, 2022-10-04:
Zugangsdaten für info@evcc.io?

stefaanbolle, 2022-10-04:
> 

What do you mean with this?

andig, 2022-10-04:
> What do you mean with this?

Wie immer: ohne Zugangsdaten keine Fehleranalyse möglich...

stefaanbolle, 2022-10-04:
Ah ok, I can give access, what do you exactly need? I can pb you with the needed info

stefaanbolle, 2022-10-04:
@VolkerK62 indeed seems to be the same problem

Macrisu, 2022-10-04:
Same problem here, but I do not have access.
It started maybe two days ago?
SoC 20% (real 84%)

andig, 2022-10-05:
> "Unfortunately we have an outage since 10/01/22 for all BYOCAR API´s. We will let you know as soon as this issue has been resolved."

> Die MB API ist wohl derzeit gestört. Dann bleibt nur Abwarten.

stefaanbolle, 2022-10-05:
That makes things clear, thanks Andig!

ilion1, 2022-10-05:
Server error 500 and 403 was logged also with older evcc versions since last weekend and is confirmed by Mercedes: [https://developer.mercedes-benz.com/status](url) _"We’re currently facing issues with our car-related data products (Vehicle Status, Electric Vehicle status, Lock status, Pay-as-you-drive Insurance and Fuel Status) as well as with their respective BYOCAR solutions. We are aware of this and are working on it."_

closed 2022-10-05 by stefaanbolle

ilion1, 2022-10-07:
The API is working again (status all green on https://developer.mercedes-benz.com/status ) and SOC is displayed correctly in EVCC for Mercedes EV.