evcc-io/evcc #4723: vehicle soc: unexpected status: 500

GR-70, 2022-10-04:
### Describe the bug

Hallo,

seit Update von v0.103.2 auf v0.104.2 geht mein Mercedes Api in diesem großartigen Projekt nicht mehr richtig: Der Soc wird nicht mehr im UI angezeigt. Anmeldung über UI auf Mercedes Api geht jedoch.
Der Charger ist ein PMCC Eebus.

Weiß jemand Rat?

Vielen Dank.

### Steps to reproduce

?

### Configuration details

# open evcc at http://evcc.local:7070

network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http

  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. Host must be set to the docker host's name
  host: 192.168.178.150

  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

  # per default, evcc should be available at http://evcc.local:7070

log: info
levels:
  cache: error

interval: 10s # control cycle interval

sponsortoken: xxxxx

meters:
- type: template
  template: sma-home-manager 
  usage: grid  
  host: 192.168.178.135  
  name: shm
- type: template
  template: sma-inverter 
  usage: pv  
  host: 192.168.178.45  
  password: xxxxxx 
  name: pv3

chargers:
- type: template
  template: pmcc
  ski: xxxxxxxx
  name: pmcc
- name: SmaEvCharger
  type: template
  template: smaevcharger
  host: 192.168.178.47 # IP-Adresse oder Hostname
  user: xxxxx # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
  password: xxxxxxx # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)  

vehicles:
- name: mercedes
  type: mercedes
  title: mercedes
  vin: Wxxxxxxx
  clientId: dbxxxxxxx
  ClientSecret: vxxxxxxx
  cache: 10m
- type: template
  template: vw
  title: T
  user: xxxxxxx
  password: xxxxxx
  vin: WVxxxxxxx
  capacity: 13
  cloud: true
  name: T

loadpoints:
- title: 9A
  charger: pmcc
  vehicle: mercedes
  mode: pv
- title: 9
  charger: SmaEvCharger
  vehicle: T7
  mode: pv
  phases: 1
  mincurrent: 6
  maxcurrent: 32
  resetOnDisconnect: true

site:
  title: Mein Zuhause
  meters:
    grid: shm
    pvs:
     - pv3

hems:
  type: shm
  allowcontrol: false
  

eebus:
  certificate:
      private: |
          -----BEGIN EC PRIVATE KEY-----
          xxxxx
          -----END EC PRIVATE KEY-----
      public: |
          -----BEGIN CERTIFICATE-----
         xxxxx
          -----END CERTIFICATE-----

### Log details

s. o.

### What type of operating system are you running?

Linux

### Version

v0.104.2

VolkerK62, 2022-10-04:
evtl #4716 ??

andig, 2022-10-05:
https://github.com/evcc-io/evcc/issues/4716#issuecomment-1268124889

closed 2022-10-05 by andig