evcc-io/evcc #4730: evcc hängt für mehrere Minuten

VolkerK62, 2022-10-05:
### Describe the bug

MIr ist vor ein paar Tagen aufgefallen, dass evcc für mehrere Minuten "eingefroren" war.
Anschließend kam im UI eine Fehlermeldung, irgendwas mit vehicle und 408.
Im log ist eine Pause von mehreren Minuten zu sehen, in der auf eine Antwort vom Auto gewartet wurde.
Das das Auto einen Fehler bringt ist klar, da gab es wohl mit Bluelink Probleme.

Aber ist es normal, dass dann der ganze Prozess stillsteht?

siehe log
16:29:37 - Pause - 16:30:29 - Pause - 16:32:24 geht es normal weiter
18:11:57 - Pause - 18:13:53 - Pause - 18:15:49 geht es normal weiter

### Steps to reproduce

-

### Configuration details

-

### Log details

```
Sep 30 16:29:27 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 16:29:27 ----
Sep 30 16:29:27 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 16:29:27 charge power: 0W
Sep 30 16:29:27 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 16:29:27 pv power: 1410W
Sep 30 16:29:27 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 16:29:27 battery power: -66W
Sep 30 16:29:27 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 16:29:27 grid power: -907W
Sep 30 16:29:27 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 16:29:27 battery soc 0: 100%
Sep 30 16:29:27 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 16:29:27 giving priority to battery charging at soc: 100%
Sep 30 16:29:27 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 16:29:27 site power: -807W
Sep 30 16:29:27 raspberrypi evcc[19472]: [hyundai] TRACE 2022/09/30 16:29:27 GET https://prd.eu-ccapi.hyundai.com:8080/api/v1/spa/vehicles/718cba00-f613-468b-9b32-4befc32058da/status/latest
Sep 30 16:29:37 raspberrypi evcc[19472]: [hyundai] TRACE 2022/09/30 16:29:37 {"retCode":"F","resCode":"5031","resMsg":"Unavailable remote control - Service Temporary Unavailable","msgId":"98be3a36-2e42-4bba-a584-6ec623dbd1df"}
Sep 30 16:29:37 raspberrypi evcc[19472]: [lp-1  ] ERROR 2022/09/30 16:29:37 vehicle odometer: unexpected status: 503
Sep 30 16:29:37 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 16:29:37 charge currents: [0 0 0]A
Sep 30 16:29:37 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 16:29:37 charger status: B
Sep 30 16:29:37 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 16:29:37 charge start energy: 986kWh
Sep 30 16:29:37 raspberrypi evcc[19472]: [lp-1  ] INFO 2022/09/30 16:29:37 car connected
Sep 30 16:29:37 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 16:29:37 pv timer elapse
Sep 30 16:29:37 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 16:29:37 pv timer inactive
Sep 30 16:29:37 raspberrypi evcc[19472]: [hyundai] TRACE 2022/09/30 16:29:37 GET https://prd.eu-ccapi.hyundai.com:8080/api/v1/spa/vehicles/718cba00-f613-468b-9b32-4befc32058da/status/latest
Sep 30 16:30:29 raspberrypi evcc[19472]: [hyundai] TRACE 2022/09/30 16:30:29 {"retCode":"F","resCode":"5031","resMsg":"Unavailable remote control - Service Temporary Unavailable","msgId":"75331c80-67c7-48a6-b012-164729874bdd"}
Sep 30 16:30:29 raspberrypi evcc[19472]: [hyundai] TRACE 2022/09/30 16:30:29 GET https://prd.eu-ccapi.hyundai.com:8080/api/v1/spa/vehicles/718cba00-f613-468b-9b32-4befc32058da/status
Sep 30 16:32:24 raspberrypi evcc[19472]: [hyundai] TRACE 2022/09/30 16:32:24 {"retCode":"F","resCode":"4081","resMsg":"Request timeout. - Request timeout.","msgId":"50e4e6dc-9997-447b-b3cb-7ce00d1bf3a2"}
Sep 30 16:32:24 raspberrypi evcc[19472]: [lp-1  ] ERROR 2022/09/30 16:32:24 vehicle soc: unexpected status: 408
Sep 30 16:32:24 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 16:32:24 pv charge current: 3.51A = 0A + 3.51A (-807W @ 1p)
Sep 30 16:32:24 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 16:32:24 pv enable timer reset
Sep 30 16:32:24 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 16:32:24 pv timer inactive
Sep 30 16:32:24 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 16:32:24 ----
```

```
Sep 30 18:11:57 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 18:11:57 ----
Sep 30 18:11:57 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 18:11:57 charge power: 0W
Sep 30 18:11:57 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 18:11:57 pv power: 123W
Sep 30 18:11:57 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 18:11:57 battery power: 280W
Sep 30 18:11:57 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 18:11:57 grid power: 68W
Sep 30 18:11:57 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 18:11:57 battery soc 0: 98%
Sep 30 18:11:57 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 18:11:57 site power: 448W
Sep 30 18:11:57 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 18:11:57 charge currents: [0 0 0]A
Sep 30 18:11:57 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 18:11:57 charger status: A
Sep 30 18:11:57 raspberrypi evcc[19472]: [lp-1  ] INFO 2022/09/30 18:11:57 car disconnected
Sep 30 18:11:57 raspberrypi evcc[19472]: [lp-1  ] INFO 2022/09/30 18:11:57 vehicle updated: unknown -> (T)Raumschiff
Sep 30 18:11:57 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 18:11:57 set charge mode: pv
Sep 30 18:11:57 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 18:11:57 pv timer elapse
Sep 30 18:11:57 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 18:11:57 pv timer inactive
Sep 30 18:11:57 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 18:11:57 set target soc: 90
Sep 30 18:11:57 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 18:11:57 set charge mode: pv
Sep 30 18:11:57 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 18:11:57 set min current: 6
Sep 30 18:11:57 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 18:11:57 set max current: 16
Sep 30 18:11:57 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 18:11:57 set min soc: 0
Sep 30 18:11:57 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 18:11:57 set target soc: 90
Sep 30 18:11:57 raspberrypi evcc[19472]: [hyundai] TRACE 2022/09/30 18:11:57 GET https://prd.eu-ccapi.hyundai.com:8080/api/v1/spa/vehicles/718cba00-f613-468b-9b32-4befc32058da/status/latest
Sep 30 18:13:53 raspberrypi evcc[19472]: [hyundai] TRACE 2022/09/30 18:13:53 {"retCode":"F","resCode":"4081","resMsg":"Request timeout. - Request timeout.","msgId":"991807c1-d6eb-4e14-bb0f-953b4058934b"}
Sep 30 18:13:53 raspberrypi evcc[19472]: [lp-1  ] ERROR 2022/09/30 18:13:53 vehicle soc: unexpected status: 408
Sep 30 18:13:53 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 18:13:53 ----
Sep 30 18:13:53 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 18:13:53 charge power: 0W
Sep 30 18:13:53 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 18:13:53 pv power: 120W
Sep 30 18:13:54 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 18:13:54 battery power: 270W
Sep 30 18:13:54 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 18:13:54 grid power: 59W
Sep 30 18:13:54 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 18:13:54 battery soc 0: 98%
Sep 30 18:13:54 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 18:13:54 site power: 430W
Sep 30 18:13:54 raspberrypi evcc[19472]: [hyundai] TRACE 2022/09/30 18:13:54 GET https://prd.eu-ccapi.hyundai.com:8080/api/v1/spa/vehicles/718cba00-f613-468b-9b32-4befc32058da/status/latest
Sep 30 18:15:49 raspberrypi evcc[19472]: [hyundai] TRACE 2022/09/30 18:15:49 {"retCode":"F","resCode":"4081","resMsg":"Request timeout. - Request timeout.","msgId":"e7d70f8e-92f0-44fc-935c-c3d53058bc19"}
Sep 30 18:15:49 raspberrypi evcc[19472]: [lp-1  ] ERROR 2022/09/30 18:15:49 vehicle odometer: unexpected status: 408
Sep 30 18:15:49 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 18:15:49 charge currents: [0 0 0]A
Sep 30 18:15:49 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 18:15:49 charger status: A
Sep 30 18:15:49 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 18:15:49 next soc poll remaining time: 56m8s
Sep 30 18:15:49 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 18:15:49 ----
Sep 30 18:15:49 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 18:15:49 charge power: 0W
Sep 30 18:15:49 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 18:15:49 pv power: 116W
Sep 30 18:15:49 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 18:15:49 battery power: 240W
Sep 30 18:15:49 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 18:15:49 grid power: 58W
Sep 30 18:15:49 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 18:15:49 battery soc 0: 97%
Sep 30 18:15:49 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 18:15:49 site power: 398W
Sep 30 18:15:49 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 18:15:49 charge currents: [0 0 0]A
Sep 30 18:15:49 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 18:15:49 charger status: A
Sep 30 18:15:49 raspberrypi evcc[19472]: [lp-1  ] DEBUG 2022/09/30 18:15:49 next soc poll remaining time: 56m8s
Sep 30 18:15:49 raspberrypi evcc[19472]: [site  ] DEBUG 2022/09/30 18:15:49 ----
```

### What type of operating system are you running?

Linux

### Version

0.104.2

andig, 2022-10-05:
// api is unbelievably slow when retrieving status
	v.Client.Timeout = 120 * time.Second

VolkerK62, 2022-10-05:
> Works as expected:

Gut. Alles klar.

closed 2022-10-05 by VolkerK62