evcc-io/evcc #4754: API reset to default

VolkerK62, 2022-10-08:
Je nach Situation kann man gewisse Einstellungen per API ändern.
Wenn man aber wieder den default (= evcc.yaml) eingestellt haben möchte, muss man entweder `restart` machen oder per  `api/state`  die aktuellen Einstellungen anschauen und selektiv ändern.
Schön wäre ein `api/default` mit dem wieder alles auf die Konfig-Einstellungen zurückgesetzt wird.

naltatis, 2022-10-08:
Kannst du das etwas konkreter benennen? Un welche Einstellungen geht es? Einige Dinge werden ja heute beim Neustart nur zurückgesetzt weil wir bislang keine Persistenz hatten.

Sobald wir bspw MinSoC, Buffer, ... über die UI konfigurierbar machen würde ich den zuletzt eingestellten Wert als den korrekten sehen und auch nicht mehr aus der config lesen.

VolkerK62, 2022-10-08:
Speziell: residualpower, min/max current, priority/buffer SoC, Da sehe ich aktuell nicht, was eingestellt ist.
Bei mode, phases, min/target SoC sehe ich die aktuellen EInstellungen im UI.

andig, 2022-10-08:
> Schön wäre ein api/default mit dem wieder alles auf die Konfig-Einstellungen zurückgesetzt wird.

Das ist ziemlich unüblich. Wenn Du sie ändern kannst kannst du sie auch wieder zurück ändern. Was spricht dagegen?

VolkerK62, 2022-10-09:
>Was spricht dagegen?

Nichts. War nur eine Idee um es einfacher zu gestalten.

andig, 2022-10-09:
Danke dir. Da das auch sonst siw nirgendwo gemacht wird diesmal wontfix.

closed 2022-10-09 by andig