evcc-io/evcc #4759: fehlende session im Ladelog

VolkerK62, 2022-10-08:
### Describe the bug

Laut Ersparnisanzeige habe ich heute 4,5 kWh geladen. Das ist auch stimmig.
Ich hatte 3 Ladevorgänge.
Laut Ladelog aber nur 2 Vorgänge mit in Summe 2,8 kWh. 
Ein kompletter Ladevorgang fehlt.

### Steps to reproduce

-

### Configuration details

-

### Log details

Das Ende des fehlenden Ladevorgang.

```
Okt 08 10:59:40 raspberrypi evcc[20751]: [site  ] DEBUG 2022/10/08 10:59:40 ----
Okt 08 10:59:40 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:40 charge power: 2800W
Okt 08 10:59:40 raspberrypi evcc[20751]: [site  ] DEBUG 2022/10/08 10:59:40 pv power: 5219W
Okt 08 10:59:40 raspberrypi evcc[20751]: [site  ] DEBUG 2022/10/08 10:59:40 battery power: -1144W
Okt 08 10:59:40 raspberrypi evcc[20751]: [site  ] DEBUG 2022/10/08 10:59:40 grid power: -434W
Okt 08 10:59:40 raspberrypi evcc[20751]: [site  ] DEBUG 2022/10/08 10:59:40 battery soc 0: 20%
Okt 08 10:59:40 raspberrypi evcc[20751]: [site  ] DEBUG 2022/10/08 10:59:40 giving priority to battery charging at soc: 20%
Okt 08 10:59:40 raspberrypi evcc[20751]: [site  ] DEBUG 2022/10/08 10:59:40 site power: -134W
Okt 08 10:59:40 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:40 charge currents: [13.3 0 0]A
Okt 08 10:59:40 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:40 detected phases: 1p
Okt 08 10:59:40 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:40 charger status: C
Okt 08 10:59:40 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:40 next soc poll remaining time: 2h59m49s
Okt 08 10:59:40 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:40 soc estimated: 56.37% (vehicle: 56.00%)
Okt 08 10:59:40 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:40 vehicle soc: 56%
Okt 08 10:59:40 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:40 vehicle range: 259km
Okt 08 10:59:40 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:40 vehicle target soc: 100%
Okt 08 10:59:40 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:40 pv charge current: 14.6A = 14A + 0.584A (-134W @ 1p)
Okt 08 10:59:40 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:40 max charge current: 14A
Okt 08 10:59:40 raspberrypi evcc[20751]: [site  ] DEBUG 2022/10/08 10:59:40 telemetry: charge: Δ8/8Wh @ 2800W
Okt 08 10:59:50 raspberrypi evcc[20751]: [site  ] DEBUG 2022/10/08 10:59:50 ----
Okt 08 10:59:50 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:50 charge power: 0W
Okt 08 10:59:50 raspberrypi evcc[20751]: [site  ] DEBUG 2022/10/08 10:59:50 pv power: 5207W
Okt 08 10:59:50 raspberrypi evcc[20751]: [site  ] DEBUG 2022/10/08 10:59:50 battery power: -1171W
Okt 08 10:59:50 raspberrypi evcc[20751]: [site  ] DEBUG 2022/10/08 10:59:50 grid power: -1912W
Okt 08 10:59:50 raspberrypi evcc[20751]: [site  ] DEBUG 2022/10/08 10:59:50 battery soc 0: 20%
Okt 08 10:59:50 raspberrypi evcc[20751]: [site  ] DEBUG 2022/10/08 10:59:50 giving priority to battery charging at soc: 20%
Okt 08 10:59:50 raspberrypi evcc[20751]: [site  ] DEBUG 2022/10/08 10:59:50 site power: -1612W
Okt 08 10:59:50 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:50 charge currents: [0 0 0]A
Okt 08 10:59:50 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:50 charger status: A
Okt 08 10:59:50 raspberrypi evcc[20751]: [lp-1  ] INFO 2022/10/08 10:59:50 car disconnected
Okt 08 10:59:50 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:50 set charge mode: pv
Okt 08 10:59:50 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:50 set min current: 6
Okt 08 10:59:50 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:50 set max current: 16
Okt 08 10:59:50 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:50 set min soc: 0
Okt 08 10:59:50 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:50 set target soc: 90
Okt 08 10:59:50 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:50 final charge energy: 1.8kWh
Okt 08 10:59:50 raspberrypi evcc[20751]: [lp-1  ] INFO 2022/10/08 10:59:50 stop charging <-
Okt 08 10:59:50 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:50 soc estimated: 56.37% (vehicle: 56.00%)
Okt 08 10:59:50 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:50 vehicle soc: 56%
Okt 08 10:59:50 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:50 vehicle range: 259km
Okt 08 10:59:50 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:50 vehicle target soc: 100%
Okt 08 10:59:50 raspberrypi evcc[20751]: [lp-1  ] WARN 2022/10/08 10:59:50 charger out of sync: expected enabled, got disabled
Okt 08 10:59:50 raspberrypi evcc[20751]: [lp-1  ] ERROR 2022/10/08 10:59:50 charger: charging station refused to start transaction
Okt 08 10:59:50 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:50 charger disable
Okt 08 10:59:50 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 10:59:50 wake-up timer: stop
Okt 08 11:00:00 raspberrypi evcc[20751]: [site  ] DEBUG 2022/10/08 11:00:00 ----
Okt 08 11:00:00 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 11:00:00 charge power: 0W
Okt 08 11:00:00 raspberrypi evcc[20751]: [site  ] DEBUG 2022/10/08 11:00:00 pv power: 5200W
Okt 08 11:00:00 raspberrypi evcc[20751]: [site  ] DEBUG 2022/10/08 11:00:00 battery power: -1151W
Okt 08 11:00:00 raspberrypi evcc[20751]: [site  ] DEBUG 2022/10/08 11:00:00 grid power: -2632W
Okt 08 11:00:00 raspberrypi evcc[20751]: [site  ] DEBUG 2022/10/08 11:00:00 battery soc 0: 20%
Okt 08 11:00:00 raspberrypi evcc[20751]: [site  ] DEBUG 2022/10/08 11:00:00 giving priority to battery charging at soc: 20%
Okt 08 11:00:00 raspberrypi evcc[20751]: [site  ] DEBUG 2022/10/08 11:00:00 site power: -2332W
Okt 08 11:00:00 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 11:00:00 charge currents: [0 0 0]A
Okt 08 11:00:00 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 11:00:00 charger status: A
Okt 08 11:00:00 raspberrypi evcc[20751]: [lp-1  ] DEBUG 2022/10/08 11:00:00 next soc poll remaining time: 2h59m49s
Okt 08 11:00:10 raspberrypi evcc[20751]: [site  ] DEBUG 2022/10/08 11:00:10 ----
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.104.2 (e43a2e19)

andig, 2022-10-08:
@VolkerK62 das ist 1 Minute Log. Das sagt leider gar nix. Es bräuchte das Log der Ladevorgänge, ohne ist nix zu machen.

VolkerK62, 2022-10-08:
OK. Hier der komplette Zeitraum.
[sessionlog0810.txt](https://github.com/evcc-io/evcc/files/9740092/sessionlog0810.txt)

andig, 2022-10-08:
Hast du mal nach Fehlermeldungen geschaut?

VolkerK62, 2022-10-09:
Ja. Nur die üblichen vom Charger, die oft kommen beim Start/Ende der Ladung.
```
Okt 08 09:48:00 raspberrypi evcc[20751]: [lp-1  ] WARN 2022/10/08 09:48:00 charger logic error: disabled but charging
Okt 08 10:59:50 raspberrypi evcc[20751]: [lp-1  ] ERROR 2022/10/08 10:59:50 charger: charging station refused to start transaction
Okt 08 11:44:30 raspberrypi evcc[20751]: [lp-1  ] ERROR 2022/10/08 11:44:30 charger: charging station refused to start transaction
Okt 08 12:27:50 raspberrypi evcc[20751]: [lp-1  ] WARN 2022/10/08 12:27:50 charger logic error: disabled but charging
Okt 08 13:09:10 raspberrypi evcc[20751]: [lp-1  ] WARN 2022/10/08 13:09:10 charger logic error: disabled but charging
```

premultiply, 2022-10-09:
Was hast du nochmal für eine Box?

VolkerK62, 2022-10-09:
DaheimLaden

andig, 2022-10-09:
Ich kanns nachvollziehen. Bei mir wurde auch grad eine Session verschluckt. Scheint, als würde sie gar nicht erst angelegt werden. Untersuchung tbd.

Chris591, 2022-10-09:
Hab es gestern auch mal getestet.
Leider fehlt auch bei mir ein Eintrag im log. Ich schauen gleich mal ob er bei mir was geloggt hat.
Aber ich hab das zielladen benutzt und kurz bevor das Auto voll war durch das Auto getrennt. Vielleicht hilft das ja.

andig, 2022-10-09:
Ich muss nochmal zurück rudern. Kein Fehler bei mir, aber folgende Beobachtung:

Der aktuelle Ladevorgang taucht nicht im Log und nicht in der Datenbank auf sondern wir erst geschrieben, wenn er endet. Könnte das das Problem sein bei Euch? Falls einer zwischendurch fehlt: wurde da evcc mal beendet?

Chris591, 2022-10-09:
ne wurde nicht neugestartet.
Die Datenbank wurde erstellt - und bei nicht jedem stop charging vom lp-3 wird ein update gemacht.

```
Oct  9 07:15:24 homeserver evcc[18765]: [lp-3  ] INFO 2022/10/09 07:15:24 start charging ->
Oct  9 07:15:33 homeserver evcc[18765]: [lp-3  ] INFO 2022/10/09 07:15:33 stop charging <-
Oct  9 08:57:00 homeserver evcc[18765]: [lp-1  ] INFO 2022/10/09 08:57:00 start charging ->
Oct  9 10:19:00 homeserver evcc[18765]: [lp-3  ] INFO 2022/10/09 10:19:00 start charging ->
Oct  9 10:19:18 homeserver evcc[18765]: [lp-3  ] INFO 2022/10/09 10:19:18 stop charging <-
Oct  9 10:40:36 homeserver evcc[18765]: [lp-3  ] INFO 2022/10/09 10:40:36 start charging ->
Oct  9 10:40:45 homeserver evcc[18765]: [lp-3  ] INFO 2022/10/09 10:40:45 stop charging <-
Oct  9 10:50:12 homeserver evcc[18765]: [lp-3  ] INFO 2022/10/09 10:50:12 start charging ->
Oct  9 10:50:21 homeserver evcc[18765]: [lp-3  ] INFO 2022/10/09 10:50:21 stop charging <-
Oct  9 10:50:22 homeserver evcc[18765]: #015
Oct  9 10:50:22 homeserver evcc[18765]: 2022/10/09 10:50:22 #033[32m/home/runner/work/evcc/evcc/core/db/db.go:49 #033[33mSLOW SQL >= 200ms
Oct  9 10:50:22 homeserver evcc[18765]: #033[0m#033[31;1m[533.503ms] #033[33m[rows:1]#033[35m UPDATE `transactions` SET `created`="2022-10-07 22:44:09.674",`finished`="2022-10-09 10:50:21.671",`loadpoint`="Heizung",`identifier`="",`vehicle`="Puffer",`meter_start_kwh`=0.000000,`meter_end_kwh`=0.000000,`charged_kwh`=9.996735 WHERE `id` = 1#033[0m
Oct  9 10:54:06 homeserver evcc[18765]: [lp-3  ] INFO 2022/10/09 10:54:06 start charging ->
Oct  9 10:54:15 homeserver evcc[18765]: [lp-3  ] INFO 2022/10/09 10:54:15 stop charging <-

```

Chris591, 2022-10-09:
aber vielleicht ist ja das Problem das das Auto beim abstecken erst unknown wird und dann das stop charging passiert.
```
Oct  8 10:23:12 homeserver evcc[18765]: [lp-1  ] INFO 2022/10/08 10:23:12 vehicle updated: Mustang Mach-e -> unknown
Oct  8 10:23:13 homeserver evcc[18765]: [lp-1  ] INFO 2022/10/08 10:23:13 stop charging <-

```

andig, 2022-10-09:
>  und bei nicht jedem stop charging vom lp-3 wird ein update gemacht.

@Chris591 das ist *sehr* merkwürdig:

	// evChargeStopHandler sends external stop event
	func (lp *LoadPoint) evChargeStopHandler() {
		lp.log.INFO.Println("stop charging <-")
		lp.pushEvent(evChargeStop)

		// soc update reset
		lp.socUpdated = time.Time{}

		// reset pv enable/disable timer
		// https://github.com/evcc-io/evcc/issues/2289
		if !lp.pvTimer.Equal(elapsed) {
			lp.resetPVTimerIfRunning()
		}

		lp.stopSession()
	}

Evtl. wird nicht jedesmal in die DB geschrieben weil irgendein schlauer Mechanismus merkt dass sich nicht geändert hat, aufgerufen wird es allerdings immer!

andig, 2022-10-09:
Sieht aus, als bräuchten wir an der Stelle fürs erste mehr Logging, ich bau was ein.

Chris591, 2022-10-09:
> Evtl. wird nicht jedesmal in die DB geschrieben weil irgendein schlauer Mechanismus merkt dass sich nicht geändert hat, aufgerufen wird es allerdings immer!

das kann gut sein - da wird sich nicht viel geändert haben - bin mir noch unsicher warum er überhaupt ständig start und stop macht. Da muss ich nochmal schauen. lp-3 ist der Pufferspeicher.

Wird das Log nur beim disconnect geschlossen und der neue Eintrag beim Connect angefangen? Würde für den Pufferspeicher gerne Täglich ein Log haben. also müsste ich da ein disconnect und Connect einbauen oder?

andig, 2022-10-09:
Tatsächlich nur beim Disconnect! Scheint aber, den hättest du schon 😂

VolkerK62, 2022-10-09:
>Der aktuelle Ladevorgang taucht nicht im Log und nicht in der Datenbank auf sondern wir erst geschrieben, wenn er endet. Könnte das das Problem sein bei Euch? Falls einer zwischendurch fehlt: wurde da evcc mal beendet?

Hilft evtl.:
Bei dem Vorgang, der bei mir fehlt, wurde das Auto während der Ladung geöffnet. Dadurch wird die Ladung unterbrochen und der Stecker freigegeben. Dann wurde abgesteckt, Auto benutzt und danach wieder angesteckt. Es gab aber keinen Neustart.

`2022-10-08   11:44:40,2022-10-08 16:33:05,Carport,,(T)Raumschiff,1017.300,1020.000,2.700`
Das mit dem Neustart scheint aber zu funktionieren.
Die Ladung endete (mangels Überschuss) bereits um 13:09. Auto blieb aber angesteckt. 
Um 16:33 wurde nach einem update der Raspi neu gestartet. Da endet dann auch die session.

Chris591, 2022-10-09:
dann wäre es bei uns das gleiche wenn das Auto disconnected wird wären es noch lädt dann geht der Eintrag. 
wie bei mir im Log wird da das disconnect vor dem stop charging ausgeführt und damit das Auto unbekannt welches geladen hat oder werden nur das Logging hier in falscher Reihenfolge ausgegeben?

andig, 2022-10-09:
> wie bei mir im Log wird da das disconnect vor dem stop charging ausgeführt

Mal langsam. Hier geht es um fehlende Ladevorgänge. Fehler bei Dir auch einer? Ob das richtige Auto im Vorgang steht ist erstmal egal bzw. wäre ein weiteres Problem.

andig, 2022-10-09:
> Bei dem Vorgang, der bei mir fehlt, wurde das Auto während der Ladung geöffnet. Dadurch wird die Ladung unterbrochen und der Stecker freigegeben. Dann wurde abgesteckt, Auto benutzt und danach wieder angesteckt. Es gab aber keinen Neustart.

@VolkerK62 wo sehe ich das im Log? Lass mich bitte nicht suchen ;)

Chris591, 2022-10-09:
> aber vielleicht ist ja das Problem das das Auto beim abstecken erst unknown wird und dann das stop charging passiert.
>
> ```
> Oct  8 10:23:12 homeserver evcc[18765]: [lp-1  ] INFO 2022/10/08 10:23:12 [...]

das ist der disconnect vom Auto, der bei mir im Ladelog fehlt.
der Pufferspeicher ist im Ladelog - war nur ne Frage wann es neu anfängt.

VolkerK62, 2022-10-09:
@andig  Bin gerade unterwegs. Das ist das kurze Log im Eingangspost. 10:59.

andig, 2022-10-09:
Ohja, ich hab ne Idee. Wenn erst Disconnect kommt und dann StopCharge, dann wird die Session genullt. Ich würde jetzt erstmal sicherstellen, dass sie geschrieben wird; ggf. dann mit dem falschen Fahrzeug. Da müssen wir uns die Reihenfolge nochmal anschauen.

andig, 2022-10-09:
Bitte nochmal in `master` probieren.

VolkerK62, 2022-10-09:
ok. Werd ich morgen ausprobieren.

andig, 2022-10-09:
Ich mach mal zu- sollte gelöst sein, Ursache ist klar!

closed 2022-10-09 by andig

RTTTC, 2022-10-09:
Great troubleshooting and teamwork you all !!

VolkerK62, 2022-10-10:
@andig   hab es gerade ausprobiert. Ladevorgang ist im log vorhanden. Good job ;-)

Chris591, 2022-10-10:
Heute morgen hat er leider erst wieder stop charging und danach car disconnected geloggt. Im lade log steht alles ordentlich drinnen.

```
INFO 2022/10/10 09:23:45 stop charging <-
INFO 2022/10/10 09:23:54 car disconnected
```

VolkerK62, 2022-10-10:
>Ohja, ich hab ne Idee. Wenn erst Disconnect kommt und dann StopCharge, dann wird die Session genullt. 

@Chris591  as designed. Die Reihenfolge ist doch jetzt genau richtig.

Chris591, 2022-10-10:
> > Ohja, ich hab ne Idee. Wenn erst Disconnect kommt und dann StopCharge, dann wird die Session genullt.
> 
> @Chris591 as designed. Die Reihenfolge ist doch jetzt genau richtig.

Ja die frage ist halt ob es wieder passieren kann das es erst ein disconnect passieren kann und es nur Zufall ist in welcher Reihenfolge es passiert. Aber mit dem Change sollte das ja für das Lade log keine Unterschied mehr machen.

Chris591, 2022-10-10:
So nochmal getestet.
Diesmal wieder in falscher Reihenfolge und trotzdem passt das Lade log!!

```
Oct 10 16:21:33 homeserver evcc[7061]: [lp-1  ] INFO 2022/10/10 16:21:33 car disconnected
Oct 10 16:21:34 homeserver evcc[7061]: [lp-1  ] INFO 2022/10/10 16:21:34 vehicle updated: Mustang Mach-e -> unknown
Oct 10 16:21:34 homeserver evcc[7061]: [lp-1  ] DEBUG 2022/10/10 16:21:34 set charge mode: off
Oct 10 16:21:34 homeserver evcc[7061]: [lp-1  ] DEBUG 2022/10/10 16:21:34 pv timer elapse
Oct 10 16:21:34 homeserver evcc[7061]: [lp-1  ] DEBUG 2022/10/10 16:21:34 pv timer inactive
Oct 10 16:21:34 homeserver evcc[7061]: [lp-1  ] DEBUG 2022/10/10 16:21:34 set min current: 6
Oct 10 16:21:34 homeserver evcc[7061]: [lp-1  ] DEBUG 2022/10/10 16:21:34 set max current: 16
Oct 10 16:21:34 homeserver evcc[7061]: [lp-1  ] DEBUG 2022/10/10 16:21:34 set min soc: 0
Oct 10 16:21:34 homeserver evcc[7061]: [lp-1  ] DEBUG 2022/10/10 16:21:34 set target soc: 100
Oct 10 16:21:34 homeserver evcc[7061]: [lp-1  ] DEBUG 2022/10/10 16:21:34 final charge energy: 18.1kWh
Oct 10 16:21:34 homeserver evcc[7061]: [lp-1  ] INFO 2022/10/10 16:21:34 stop charging <-
Oct 10 16:21:34 homeserver evcc[7061]: [lp-1  ] DEBUG 2022/10/10 16:21:34 charger disable
Oct 10 16:21:34 homeserver evcc[7061]: [lp-1  ] DEBUG 2022/10/10 16:21:34 wake-up timer: stop

```

andig, 2022-10-10:
Supi :) Another job well-done, danke für die Fehlersuche!

Chris591, 2022-10-11:
Gerne ;-)
Danke fürs schnelle fixen und alles andere was ihr hier macht!
Seit dem Update sehe ich aber keine SQL Logs mehr. wie kann ich die wieder mit loggen lassen?
also die hier:
```
Oct  9 21:59:38 homeserver evcc[17059]: #033[0m#033[31;1m[641.473ms] #033[33m[rows:1]#033[35m UPDATE `transactions` SET `created`="2022-10-09 21:59:19.892",`finished`="2022-10-09 21:59:37.893",`loadpoint`="Heizung",`identifier`="",`vehicle`="Puffer",`meter_start_kwh`=0.000000,`meter_end_kwh`=0.000000,`charged_kwh`=0.097105 WHERE `id` = 5#033[0m
```

andig, 2022-10-11:
db: trace