evcc-io/evcc #4770: evcc not working on macOS 10.11 (El Capitan)

kaikreuzer, 2022-10-09:
I'm running a Mac mini, which does not support any newer macOS than El Capitan.
Trying to launch evcc results in 
```
dyld: Symbol not found: _clock_gettime
  Referenced from: /usr/local/bin/evcc
  Expected in: flat namespace
```
This seems to be an issue with Go versions >=1.16.2 (see https://stackoverflow.com/questions/66741383/dyld-symbol-not-found-clock-gettime-after-installing-golang).

Is there any way to compile it with an older Go version? If not, you might at least want to add this information to the system requirements.

goebelmeier, 2022-10-09:
I am very unsure if it is really necessary to point out in the system requirements that unsupported operating system versions are unsupported 😁

andig, 2022-10-09:
Ich fürchte das ist wontfix. Wir brauchen Go 1.18.

andig, 2022-10-09:
@kaikreuzer ich hab jetzt erst verstanden was Du mit Requirements meinst- gerne ein PR gegen das docs Repo!

closed 2022-10-09 by andig

kaikreuzer, 2022-10-09:
>  that unsupported operating system versions are unsupported

macOS wird ja als supported gelistet. Ist aber halt leider dann doch nicht immer der Fall - von daher halte ich den Hinweis für hilfreich.

> gerne ein PR gegen das docs Repo!

Erledigt: https://github.com/evcc-io/docs/pull/237