evcc-io/evcc #4777: [Question] [Improve Documentation] Home Energy Management Systeme (HEMS)

m33x, 2022-10-09:
Hi,

cool project. Thank you!

One question / documnetation issue: The [documentation ](https://docs.evcc.io/docs/reference/configuration/hems/)only talks about the "SMA Sunny Home Manager 2.0". Unfortunately, I only have the **SMA Sunny Home Manager 1.0**, can you please specify whether the **old 1.0 version** is supported or not? If not, can you please add this to the documentation. 

Thank you!

premultiply, 2022-10-10:
SHM 1.0 should be supported as HEMS.
But SHM 1.0 is not supported for (grid) metering.

andig, 2022-10-10:
@premultiply sollten wir dann das Template splitten?

premultiply, 2022-10-10:
Brauchen wir m.M.n. nicht, da das sich nur auf das Metering bezieht.
Beim HEMS wird kein Template verwendet.

andig, 2022-10-12:
Dann hier zu :)

closed 2022-10-12 by andig