evcc-io/evcc #4788: minCurrent für 3p anders als für 1p einrichten?

tguenthert, 2022-10-11:
Überwiegend läuft evcc prima und stabil. Bei der Phasenumschaltung (go-echarger) habe ich allerdings beim runter schalten von 3p auf 1p ein Problem. Unser Auto (EQ Smart) mag offensichtlich nicht mit 3p bis 7A geladen werden. Geht erst ab 8A. Für 3p müsste ich einen Befehl haben (minCurrent?) der die minimale Ladeleistung auf 8A setzt. Da springt das System auch rein, wenn es von 1p auf 3p hochschaltet. Beim runter schalten geht 3p natürlich erst mal in von 8A auf 7A oder auf 6A runter. Dann hängt das System. Sobald ich dann manuell auf 1 Phase schalte, läuft es wieder. In 1p funktioniert 6A und 7A prima. Die würden wir gerne auch weiterhin nutzen.

Daher die Frage: kann ich für 3p minCurrent (bei uns 8A) anders definieren als für 1p (bei uns 6A)?

Danke für jede Hilfe!

Joegi05, 2022-10-11:
Hallo,

ich habe genau das gleiche Problem mit meinem Twingo 3. Wir hatten das aber vor Kurzem schon einmal hier im Forum. [https://github.com/evcc-io/evcc/issues/3528](https://github.com/evcc-io/evcc/issues/3528)

Viele Grüße

tguenthert, 2022-10-11:
So wie es aussieht aber erst einmal keine Lösung - schade.

VolkerK62, 2022-10-11:
Evtl kannst du dir selber eine API-Routine basteln.  Z.B. mit NodeRed. 
Bei evcc-MQTT gibt es `phasesActive` (ich vermute, dass das drin ist, was drauf steht)
Dann kannst du in Abhängigkeit von dem Wert automatisch per API `mincurrent` auf 8 bzw 6 setzen.

andig, 2022-11-03:
Siehe Auflösung zu https://github.com/evcc-io/evcc/issues/4641. Vorerst wontfix bis wir eine gute Idee haben wie das zu lösen wäre.

closed 2022-11-03 by andig