evcc-io/evcc #4789: SMA Homemanager 2.0 - Error Messages when other connections are present

Percy2Live, 2022-10-11:
### Describe the bug

Connection to the SMA HM 2.0 works fine, as long as evcc is the only system which accesses the HM. As soon as a second application accesses the HM, error messages appear in evcc.
evcc as well as the second application (in this case an iobroker instance) both show the HM values correctly

### Steps to reproduce

Connect multiple applications to the HM

[image]

### Configuration details

uri: 192.168.178.2:7070 # uri for ui
interval: 10s # control cycle interval
sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE2ODE0MTg1MzQsImlhdCI6MTY0OTg4MjUzNCwiaXNzIjoiZXZjYy5pbyIsInN1YiI6IlBlcmN5MkxpdmUifQ.9Ec41CZe7CogYzKUy1ocYqbEA18bqLZrQfZssLZgHIo

log: debug

meters:
- name: Homemanager
  type: template
  template: sma-home-manager
  usage: grid
  host: 192.168.178.15 # IP-Adresse oder Hostname 
    
- name: Wechselrichter
  type: template
  template: sma-inverter
  usage: pv
  host: 192.168.178.14 # IP-Adresse oder Hostname
  password: 12345# Anlagenpasswort / Installateurspasswort # Optional 

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.178.153  
  name: wbPascal
- type: template
  template: go-e-v3 
  host: 192.168.178.155  
  name: wbSandra

vehicles:
- type: template
  template: nissan 
  title: Leaf Pascal 
  user: XXXXX  
  password: XXXX
  vin: XXXXX
  capacity: 40  
  name: ev1

loadpoints:

- title: Ladepunkt Sandra
  charger: wbSandra
  vehicles: 
  - ev1
  mode: pv
  resetOnDisconnect: false
  phases: 3 # ev phases (default 3)
  enable: # pv mode enable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: -1600 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 10m # threshold must be exceeded for this long
    threshold: 300 # maximum import power (W)
  guardDuration: 10m # switch charger contactor not more often than this (default 5m)
  minCurrent: 6 # minimum charge current (default 6A)
  maxCurrent: 32 # maximum charge current (default 16A)
  soc:
    poll:
      mode: charging
      interval: 60m
    min: 0
    target: 100
    estimate: true

- title: Ladepunkt Pascal
  charger: wbPascal
  vehicles: 
  - ev1
  mode: pv
  resetOnDisconnect: false
  phases: 3 # ev phases (default 3)
  enable: # pv mode enable behavior
    delay: 5m # threshold must be exceeded for this long
    threshold: -1600 # grid power threshold (in Watts, negative=export). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 10m # threshold must be exceeded for this long
    threshold: 300 # maximum import power (W)
  guardDuration: 10m # switch charger contactor not more often than this (default 5m)
  minCurrent: 6 # minimum charge current (default 6A)
  maxCurrent: 32 # maximum charge current (default 16A)
  soc:
    poll:
      mode: charging
      interval: 60m
    min: 0
    target: 100
    estimate: true

site:
  title: Home Ladestation # display name for UI
  meters:
    grid: Homemanager # grid meter
    pvs:
      - Wechselrichter # list of pv inverters/ meters
    battery: # battery meter
  prioritySoC: # give home battery priority up to this soc (empty to disable)
  bufferSoC: # ignore home battery discharge above soc (empty to disable)
  residualPower: #Wintermodus: Wie viel Netzbezug ist ok, damit trotzdem geladen wird (z.B. -1000W)
  
# tariffs are the fixed or variable tariffs
# cheap (tibber/awattar) can be used to define a tariff rate considered cheap enough for charging
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    # either static grid price
    type: fixed
    price: 0.264 # EUR/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.127 # EUR/kWh
influx:
  url: http://192.168.178.23:8086
  database: evcc
  user: admin
  password: XXXXX

### Log details

`/app # evcc --log debug
[main  ] INFO 2022/10/11 23:51:15 evcc 0.104.2
[main  ] INFO 2022/10/11 23:51:15 using config file: /etc/evcc.yaml
[main  ] WARN 2022/10/11 23:51:15 `uri` is deprecated and will be ignored. Use `network` instead.
[main  ] INFO 2022/10/11 23:51:15 listening at :7070
[lp-1  ] WARN 2022/10/11 23:51:24 Configuring soc.target at loadpoint is deprecated and must be applied per vehicle
[lp-1  ] WARN 2022/10/11 23:51:24 vehicles option is deprecated
[lp-1  ] WARN 2022/10/11 23:51:24 locking phase config to 3p for switchable charger
[lp-2  ] WARN 2022/10/11 23:51:24 Configuring soc.target at loadpoint is deprecated and must be applied per vehicle
[lp-2  ] WARN 2022/10/11 23:51:24 vehicles option is deprecated
[site  ] INFO 2022/10/11 23:51:24 site config:
[site  ] INFO 2022/10/11 23:51:24   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2022/10/11 23:51:24     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2022/10/11 23:51:24     pv 1:      power ✓ energy ✓ currents ✓
[site  ] INFO 2022/10/11 23:51:24   vehicles:
[site  ] INFO 2022/10/11 23:51:24     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[lp-1  ] INFO 2022/10/11 23:51:24 loadpoint 1:
[lp-1  ] INFO 2022/10/11 23:51:24   mode:        pv
[lp-1  ] INFO 2022/10/11 23:51:24   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2022/10/11 23:51:24   meters:      charge ✓
[lp-1  ] INFO 2022/10/11 23:51:24     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2022/10/11 23:51:24 loadpoint 2:
[lp-2  ] INFO 2022/10/11 23:51:24   mode:        pv
[lp-2  ] INFO 2022/10/11 23:51:24   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-2  ] INFO 2022/10/11 23:51:24   meters:      charge ✓
[lp-2  ] INFO 2022/10/11 23:51:24     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2022/10/11 23:51:24 phase timer inactive
[lp-1  ] DEBUG 2022/10/11 23:51:24 pv timer inactive
[lp-1  ] DEBUG 2022/10/11 23:51:24 max charge current: 6A
[lp-2  ] DEBUG 2022/10/11 23:51:24 phase timer inactive
[lp-2  ] DEBUG 2022/10/11 23:51:24 pv timer inactive
[lp-2  ] ERROR 2022/10/11 23:51:24 charger: Get "http://192.168.101.153/status": dial tcp 192.168.101.153:80: connect: no route to host
[site  ] DEBUG 2022/10/11 23:51:24 ----
[main  ] FATAL 2022/10/11 23:51:24 listen tcp :7070: bind: address already in use`

### What type of operating system are you running?

Docker container

### Version

evcc version 0.104.2

Sillium007, 2022-10-12:
`[main ] FATAL 2022/10/11 23:51:24 listen tcp :7070: bind: address already in use`

Are you running 2 instances of evcc?

Sillium007, 2022-10-12:
> no, just one. Maybe this error is due to the concurrent access of the Homemanager?

No. This error indicates that evcc wants to listen to the local port 7070 for incoming requests (basically this is the port locally used to provide the website of your evcc instance). This has nothing to do with outgoing connections to other devices.
Did you shutdown the evcc service prior to running it manually?

Anyway this should not really relate to the initial problem you are mentioning.

I would guess that there is not much evcc can do about it if the system you are using does not like having more then one client. I don't know the SMA Homemanager, but maybe it either does not like being polled by 2 clients or does not like being polled to fast. If it is "only" too fast you could turn down the interval of both evcc and iobroker.
Another thing I can think of is that instead of letting iobroker query the SMA Homemanager, the iobroker could query evcc to get the information.

One small hint: Wrap your config in code tags to make it more readable.