evcc-io/evcc #4790: Fehlermeldung: telemetry benötigt sponsortoken

VolkerK62, 2022-10-12:
### Describe the bug

gelegentlich, nicht immer, erhalte ich nach einem reboot des raspi oder auch nach einem restart von evcc im UI diese Fehlermeldung. Ein erneuter restart behebt dann das Problem.
Als Ahnungsloser kommt es mir so vor, als würde telemetry aktiviert, bevor sponsortoken erkannt wurde.
Hab gerade keinen Zugriff. Konfig und Log müsste ich nachliefern, wenn notwendig.

### Steps to reproduce

-

### Configuration details

-

### Log details

-

### What type of operating system are you running?

Linux

### Version

aktuelles nightly

andig, 2022-10-12:
Der Fehler bricht dann endgültig ab?

VolkerK62, 2022-10-12:
Ja. Ich habe gestern abend den Status nicht mehr gecheckt und heute morgen empfing mich das UI mit dem Abbild meiner Konfig ;-)   Aber, ein erneuter restart von evcc und es lief wieder. Problem ist, es tritt nicht immer auf.

andig, 2022-10-13:
Ich hab noch eine Kleinigkeit geändert, aber keine weitere Idee. Alle Initialisierung passiert beim Start. Ich mache mal zu, wenns ab morgen wieder auftreten sollte bitte nochmal melden.

closed 2022-10-13 by andig