evcc-io/evcc #4814: Mehr Werte von der Bluelink API

emcepe, 2022-10-14:
Ich würde mich über mehr Werte aus Bluelink freuen, z.B. sowas wie:

- location (long/lat)
- location name

Ich weiß, es hat nicht wirklich etwas mit Laden eines E-Autos zu tun, jedoch fragt evcc eh die APIs ab und es wäre IMHO ein leichtes die o.g. Sachen mit abzufragen (zumindest für Bluelink, andere kenne ich nicht) und via MQTT verfügbar zu machen.

Wieso? Bisher habe ich einige Sachen in FHEM drin um festzustellen wer zu Hause ist um dann entsprechend per Pushover an die passenden Geräte die Info zu schicken, dass das Auto angeschlossen werden muss & geladen werden sollte, wenn der SoC unterhalb X ist - das wird bei uns nämlich gerne des Öfteren vergessen :-(

Das funktioniert aber nur bedingt gut, weil weder FHEM noch evcc wissen, ob das Auto ebenso zu Hause ist oder nicht.
Wenn man die o.g. Infos hätte, dann könnte man darauf abfragen und hätte dann immer ein gültiges Ergebnis, da ich evcc ebenso via MQTT in FHEM drin habe.

Möglich? 😀

und wenn man schon dabei ist vielleicht sogar noch mehr Werte wie Energieverbrauch, ob die Klima an/aus ist, die Türen auf/zu sind and whatnot.

Dankeschön.

andig, 2022-10-14:
Position würde passen, mit location name kann evcc nix anfangen. Im Moment haben wir kein Ziel, zusätzliche Werte abzufragen mit denen evcc nix anfangen kann- insofern würde auch Position aktuell nicht verwendet.

andig, 2022-10-14:
Added position in https://github.com/evcc-io/evcc/commit/be9c53d6d4d3a5060459c5cd43b135db294f93a1

closed 2022-10-14 by andig

emcepe, 2022-10-14:
danke fürs Implementieren von Position.

Müsste das mit dem Release 0.105.1 schon per MQTT automatisch auftauchen?

andig, 2022-10-15:
Leider nein. Es fehlt noch eine gute Idee, wie beliebige zusätzliche Meßwerte von Fahrzeugen, Zähler etc abgerufen werden sollten.

andig, 2022-10-15:
Siehe auch https://github.com/evcc-io/evcc/issues/4831#issuecomment-1279726695