evcc-io/evcc #4815: Elli-Support mit 0.105 broken wg. Cert...

pencoe, 2022-10-14:
### Describe the bug

Das Release 0.105 kann nicht mit der Elli-Wallbox reden, weil der Cert-Patch wohl nicht enthalten ist:

Oct 14 14:48:40 RPNagel3 evcc[1625900]: [eebus ] TRACE 2022/10/14 14:48:39 Failed to connect to wss://192.168.48.66:4712/ship/: tls: failed to parse certificate from server: x509: invalid basic constraints b

### Steps to reproduce

Release mit Elli-WB starten...

### Configuration details

chargers:
- type: template
  template: ellipro
  ski: 4F86 4FB4 C192 10AD 78EA 8C50 A07E E0D4 0FAB F5E0
  ip: 192.168.48.66
  name: wallbox1

### Log details

Oct 14 14:48:40 RPNagel3 evcc[1625900]: [eebus ] TRACE 2022/10/14 14:48:39 Failed to connect to wss://192.168.48.66:4712/ship/: tls: failed to parse certificate from server: x509: invalid basic constraints b

### What type of operating system are you running?

Linux

### Version

evcc version 0.105.0

andig, 2022-10-14:
Diese Frickelbox raubt einem den letzten Nerv. Ich hoffe @DerAndereAndi hat eine Idee.

DerAndereAndi, 2022-10-14:
@pencoe Thanks for the hint and sorry for the issue. We forgot to adopt the patching to the release workflow. This should be fixed in 0.105.1. Could you please give it a try?

closed 2022-10-14 by andig

pencoe, 2022-10-14:
> @pencoe Thanks for the hint and sorry for the issue. We forgot to adopt the patching to the release workflow. This should be fixed in 0.105.1. Could you please give it a try?

Top, ist gefixt, danke :-D