evcc-io/evcc #4821: Unhealthy container after upgrade

hyperbart, 2022-10-14:
### Describe the bug

I try to do weekly updates of EVCC.
Running EVCC in a Docker Stack.
After tonights "docker stack rm EVCC" and "docker stack deploy compose.yml EVCC" I noticed EVCC did not report healthy in my monitoring.
The web interface is down.
Weirdly it seems to be running 2 dockers at the same time?

### Steps to reproduce

Remove the docker stack, pull in new images and redeploy.

```
bart@nas:~$ docker stack ps EVCC --no-trunc
ID                          NAME              IMAGE                                                                                        NODE      DESIRED STATE   CURRENT STATE                 ERROR                                                        PORTS
iglmyk6pfi0i6unb77jr7u98o   EVCC_evcc.1       andig/evcc:nightly@sha256:031a0e286f17a1ea11800e0244a0236364eebee63c1b43611920823418017678   nas       Running         Starting about a minute ago
5mujgkufqp8zpfc2w18w2ukvs    \_ EVCC_evcc.1   andig/evcc:nightly@sha256:031a0e286f17a1ea11800e0244a0236364eebee63c1b43611920823418017678   nas       Shutdown        Failed about a minute ago     "task: non-zero exit (1): dockerexec: unhealthy container"
r633cwlo8h46ru67qfpp38xvm    \_ EVCC_evcc.1   andig/evcc:nightly@sha256:031a0e286f17a1ea11800e0244a0236364eebee63c1b43611920823418017678   nas       Shutdown        Failed 4 minutes ago          "task: non-zero exit (1): dockerexec: unhealthy container"
0xjhqsnxo1kaybv2qr2zzfuza    \_ EVCC_evcc.1   andig/evcc:nightly@sha256:031a0e286f17a1ea11800e0244a0236364eebee63c1b43611920823418017678   nas       Shutdown        Failed 7 minutes ago          "task: non-zero exit (1): dockerexec: unhealthy container"
kfbt4svsm2ax6k0ck9g0rjtkh    \_ EVCC_evcc.1   andig/evcc:nightly@sha256:031a0e286f17a1ea11800e0244a0236364eebee63c1b43611920823418017678   nas       Shutdown        Failed 10 minutes ago         "task: non-zero exit (1): dockerexec: unhealthy container"
```

### Configuration details

```
interval: 10s

site:
- title: qsfqsfqsdfqsdf
  meters:
    grid: HuisMeter
    pvs:
    - pvHuis
    - pvTuinHuis
  residualpower: 100

messaging:
  events:
    start:
      title: Laadbeurt gestart
      msg: Er is een laadbeurt gestart op qsfqsfqsdf
    stop:
      title: Laadbeurt gestopt
      msg: |
        Oplader {{.title}} heeft een laadbeurt voltooid
        Er is {{round (divf .chargedEnergy 1000) 2 }} kWh geleverd op {{.chargeDuration}}.
    connect:
      title: Wagen aangesloten op qsfdsqdqdsf
      msg: Er is een auto aangesloten op de lader in qsfsqfqsdf
    disconnect:
      title: Wagen ontkoppeld op qsdfsqdfsqdf
      msg: De wagen is ontkoppeld van de lader in qsfdqsfqsdf
  services:
    - type: telegram
      token: sqfdqsfqsfd:qsfdqsdfsqdf
      chats:
        - qsfdqsfqsf

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.28 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.12 # [currency]/kWh

[... 710 lines cut ...]
EVCC_evcc.1.5mujgkufqp8z@nas    | [lp-1  ] DEBUG 2022/10/14 23:50:40 charge currents: [0 0 0]A
EVCC_evcc.1.r633cwlo8h46@nas    | [lp-1  ] DEBUG 2022/10/14 23:47:21 charger status: A
EVCC_evcc.1.r633cwlo8h46@nas    | [site  ] DEBUG 2022/10/14 23:47:31 ----
EVCC_evcc.1.r633cwlo8h46@nas    | [lp-1  ] DEBUG 2022/10/14 23:47:31 charge power: 0W
EVCC_evcc.1.5mujgkufqp8z@nas    | [lp-1  ] DEBUG 2022/10/14 23:50:40 charger status: A
EVCC_evcc.1.r633cwlo8h46@nas    | [site  ] DEBUG 2022/10/14 23:47:31 pv power: 0W
EVCC_evcc.1.r633cwlo8h46@nas    | [site  ] DEBUG 2022/10/14 23:47:31 grid power: 570W
EVCC_evcc.1.r633cwlo8h46@nas    | [site  ] DEBUG 2022/10/14 23:47:31 site power: 670W
EVCC_evcc.1.r633cwlo8h46@nas    | [lp-1  ] DEBUG 2022/10/14 23:47:31 charge currents: [0 0 0]A
EVCC_evcc.1.r633cwlo8h46@nas    | [lp-1  ] DEBUG 2022/10/14 23:47:31 charger status: A
```

### What type of operating system are you running?

Linux

### Version

Linux nas 5.15.0-50-generic #56-Ubuntu SMP Tue Sep 20 13:23:26 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.1 LTS
Release:        22.04
Codename:       jammy

hyperbart, 2022-10-14:
Although the container in some way seems to be up it is not listening on 7070 anymore:

```
bart@nas:~$ sudo lsof -i -P -n | grep LISTEN | grep 7070
bart@nas:~$
```

hyperbart, 2022-10-14:
Images are up to date:
```
bart@nas:~$ docker image ls | grep andi
andig/evcc                          latest    009715c90545   8 hours ago     53.1MB
andig/evcc                          <none>    c5826aa62924   18 hours ago    53.1MB
andig/evcc                          <none>    063f8a351aac   2 weeks ago     49.7MB
```

rubinho76, 2022-10-15:
Hab das gleiche Problem.
Nach dem Upgrade auf 0.105.1 wird mir der Container als unhealthy angezeigt. 
Der native Webzugang funktioniert jedoch, allerdings über Traefik  nicht mehr, was am unhealthy  liegen könnte.
Das Evcc Log sieht unspektakulär aus.
Mein Docker läuft auf einer Synology DS916+ mit x86_64 SOC.

lg
Rubinho

rubinho76, 2022-10-15:
--Update--
Das Problem ist auch in der aktuellen Nightly und in der 0.105.0 vorhanden. Ein Downgrade auf 0.104.2 behebt das Problem.
@hyperbart  The reason why your webinterface is not working could be Traefik (Do you have Traefik in use?) Traefik does a healthcheck and if it is negative, no proxy will be established.

premultiply, 2022-10-15:
Was genau tut denn der "Healthcheck" technisch?

hyperbart, 2022-10-15:
I don’t have a Traefik running but I do have Cloudflare tunnel. But that does normally nog intervene with other containers. EVCC is only attached to an extra network where the cloudlared container also resides so it can tunnel traffic outwards.

i will try later today what happens if I take away the network, then it should run standalone.

@premultiply : no idea about the healthcheck.

rubinho76, 2022-10-15:
@premultiply Ich kann dir das auch nicht wirklich sagen. Manche Images haben einen, manche nicht. Ich glaube @andig muss ihn bewusst eingebaut haben, oder hat sich einem Basisimage bedient, der ein Healthcheck integriert hat. Was der genau prüft, ist mir noch nicht klar. Im Container-Log sehe ich zumindest nichts.

premultiply, 2022-10-15:
Was gibt denn http://evcc.local:7070/api/health (bzw. mit der IP) zurück im Browser?

rubinho76, 2022-10-15:
Das hier...
`SyntaxError: JSON.parse: unexpected character at line 1 column 1 of the JSON data`
Keine Ahnung, ob das der Fehler ist oder der Healthcheck kaputt ist.

rubinho76, 2022-10-15:
Kommando zurück.
Wenn ich im Browser auf Rohdaten klicke, steht da "ok"

rubinho76, 2022-10-15:
Das hier spuckt mir Docker inspect aus...

Output [main ] INFO 2022/10/15 08:56:56 evcc 0.105.1 [main ] ERROR 2022/10/15 08:56:56 health check failed

closed 2022-10-15 by andig

hyperbart, 2022-10-15:
@premultiply : URL gives nothing because the port is not yet open.
@andig : I just deployed but it still exits with the following error. See below

```
bart@nas:~$ docker stack ps EVCC --no-trunc
ID                          NAME              IMAGE                                                                                        NODE      DESIRED STATE   CURRENT STATE                 ERROR                                                        PORTS
4z4qjj2bp8zp4xh0azkaho4zz   EVCC_evcc.1       andig/evcc:nightly@sha256:eb229e21cfbf01f361d71cff98d43583cbf604e29db26cd6258df6d3acd5bfd8   nas       Running         Starting about a minute ago
rjheacxcjlf61ut55dju6ze55    \_ EVCC_evcc.1   andig/evcc:nightly@sha256:eb229e21cfbf01f361d71cff98d43583cbf604e29db26cd6258df6d3acd5bfd8   nas       Shutdown        Failed about a minute ago     "task: non-zero exit (1): dockerexec: unhealthy container"
```

rubinho76, 2022-10-15:
@hyperbart 
It probably doesn't help you, but for me the new version 0.105.2 works again. Did you delete the old image?

hyperbart, 2022-10-15:
@rubinho76 

By switching from nightly to latest it appears to be working again. Took some time to start the container apparently. (around a minute or so).

I'll check if switching to nightly works too.

hyperbart, 2022-10-15:
Nightly still has the issue (I am not a developer, so don't know if that is to be expected). I once switched to nightly's to be able to enjoy the new GUI by @premultiply  👍 .

Switched back to latest and exactly after a minute the container comes up.

rubinho76, 2022-10-15:
@hyperbart
At the moment the "latest" is the newest version. The nightly is 6 hours old and still has the bug
If you are happy with the stable versions, leave it on latest.

andresp, 2025-08-24:
It appears that this bug still persists in the latest 0.207.4 release. The UI won't listen on port 7070 until after the OCPP chargers in my configuration have connected to the OCPP service, or the timeout for connectiontimeout is reached. Because Kubernetes routes traffic based on health of a pod (container), it won't permit OCPP connections to reach a new (or upgraded) instance of evcc because it is unable to ping the container on /api/health. As you can see, this creates a circular dependency that cannot be resolved: K8s won't route traffic to the new container because the health probe is unreachable, and the health probe is contingent to the OCPP chargers checking in or timing out before it will be responsive.