evcc-io/evcc #4828: service evcc stop/restart exits with return code > 0

emcepe, 2022-10-15:
### Describe the bug

"service evcc stop" and "service evcc restart" exit with return code > 0

### Steps to reproduce

see above

### Configuration details

not needed

### Log details

service evcc stop:
Oct 15 11:54:03 raspberry-pi-4b systemd[1]: Stopping evcc...
Oct 15 11:54:03 raspberry-pi-4b systemd[1]: evcc.service: Main process exited, code=exited, status=1/FAILURE
Oct 15 11:54:03 raspberry-pi-4b systemd[1]: evcc.service: Failed with result 'exit-code'.
Oct 15 11:54:03 raspberry-pi-4b systemd[1]: Stopped evcc.
Oct 15 11:54:03 raspberry-pi-4b systemd[1]: evcc.service: Consumed 1.253s CPU time.

service evcc restart:
Oct 15 11:54:42 raspberry-pi-4b systemd[1]: Stopping evcc...
Oct 15 11:54:42 raspberry-pi-4b systemd[1]: evcc.service: Main process exited, code=exited, status=1/FAILURE
Oct 15 11:54:42 raspberry-pi-4b systemd[1]: evcc.service: Failed with result 'exit-code'.
Oct 15 11:54:42 raspberry-pi-4b systemd[1]: Stopped evcc.
Oct 15 11:54:42 raspberry-pi-4b systemd[1]: Started evcc.

the following diff fixes it:

```
--- /lib/systemd/system/evcc.service    2022-09-29 09:52:00.000000000 +0200
+++ /lib/systemd/system/evcc.service    2022-10-15 11:55:54.296845058 +0200
@@ -11,8 +11,12 @@ StartLimitBurst=10

 [Service]
 ExecStart=/usr/bin/evcc
-Restart=always
+Restart=on-failure
 RestartSec=10
+RestartPreventExitStatus=255
+KillMode=process
+KillSignal=15
+SuccessExitStatus=1

 [Install]
 WantedBy=multi-user.target
```

service evcc stop:
Oct 15 12:16:38 raspberry-pi-4b systemd[1]: Stopping evcc...
Oct 15 12:16:38 raspberry-pi-4b systemd[1]: evcc.service: Succeeded.
Oct 15 12:16:38 raspberry-pi-4b systemd[1]: Stopped evcc.
Oct 15 12:16:38 raspberry-pi-4b systemd[1]: evcc.service: Consumed 6.637s CPU time.

service evcc restart:
Oct 15 12:17:12 raspberry-pi-4b systemd[1]: Stopping evcc...
Oct 15 12:17:12 raspberry-pi-4b systemd[1]: evcc.service: Succeeded.
Oct 15 12:17:12 raspberry-pi-4b systemd[1]: Stopped evcc.
Oct 15 12:17:12 raspberry-pi-4b systemd[1]: Started evcc.

### What type of operating system are you running?

Linux

### Version

0.105.2

andig, 2022-10-15:
Wie funktionieren Start/Stop unter Linux- SIGTERM?

andig, 2022-10-15:
Should be fixed on master

closed 2022-10-16 by andig