evcc-io/evcc #4846: Aktuell wird der Ladestand in Prozent und in km nur alle 15 min von dem Fahrzeug abgefragt. In Zeit dazwischen wird zwar SoC in % durch Hochrechnung angepasst, jedoch nicht die Reichweite in km. Diese bleibt die ganze 15 min unverändert und macht dann erst einen großen Sprung. Dadurch ändert sich bei jedem gewonnenem % auch das Verhältnis zwischen SoC in % und der Reichweite in km. Dabei reduziert sich auch das Ladeziel in km bei jedem gewonnenem %.

andig, 2022-10-16:
Aktuell wird der Ladestand in Prozent und in km nur alle 15 min von dem Fahrzeug abgefragt. In Zeit dazwischen wird zwar SoC in % durch Hochrechnung angepasst, jedoch nicht die Reichweite in km. Diese bleibt die ganze 15 min unverändert und macht dann erst einen großen Sprung. Dadurch ändert sich bei jedem gewonnenem % auch das Verhältnis zwischen SoC in % und der Reichweite in km. Dabei reduziert sich auch das Ladeziel in km bei jedem gewonnenem %. 
Vorschlag: Bei Hochrechnung des Ladestands sowohl in % wie auch im km durchführen

_Originally posted by @11-Andreas in https://github.com/evcc-io/evcc/discussions/4844_

Hofyyy, 2022-10-16:
Mit welchem Algorithmus würdet ihr hier arbeiten? Einfach Linear die letzten Werte vom Auto?

API vom Auto:
- 50% und 100KM

Dann estimator
- 55% -> (55 * 100 / 50) > 110KM estimate?

Hofyyy, 2022-10-16:
Und die Frage warum es im Vehicle mehrere Interfaces gibt. 
Ist es weil nicht jeder Hersteller eine Reichweite o.ä. liefern kann?
Sonst wäre die Frage warum wir die Function Range() nicht auch um Vehicle Interface providen können.

premultiply, 2022-10-16:
> Mit welchem Algorithmus würdet ihr hier arbeiten?

Gute Frage. Aber so in etwa stelle ich mir das vor.
Dann basiert die geschätze Reichweite auf dem geschätzten SoC.

Das setzt aber voraus bzw. geht dann davon aus, dass das Auto intern auch so linear rechnet.

11-Andreas, 2022-10-16:
Wenn ich es richtig beobachtet habe, wird ja auch schon bei Reichweiteberechnung beim  Ladeziel mit Dreisatz gerechnet. Also kann man es auch beim Ladestand verwenden. 
Man muss sogar denselben Algorithmus verwenden.

andig, 2023-02-18:
Derzeit keine Prio.

closed 2023-02-18 by andig