evcc-io/evcc #4848: evcc panic

emcepe, 2022-10-16:
### Describe the bug

I like to set a guest vehicle via the MQTT api but didn't find out how to do it, so I did trial & error :)

if I set "loadpoints/1/vehicle/set" to -1, then:

```
Oct 16 23:24:16 raspberry-pi-4b systemd[1]: evcc.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Oct 16 23:24:16 raspberry-pi-4b systemd[1]: evcc.service: Failed with result 'exit-code'.
Oct 16 23:24:16 raspberry-pi-4b systemd[1]: evcc.service: Consumed 3.409s CPU time.
Oct 16 23:24:16 raspberry-pi-4b evcc[4328]: panic: runtime error: index out of range [-1]
Oct 16 23:24:16 raspberry-pi-4b evcc[4328]: goroutine 63 [running]:
Oct 16 23:24:16 raspberry-pi-4b evcc[4328]: github.com/evcc-io/evcc/server.(*MQTT).listenSetters.func7({0x4b74de2, 0x2})
Oct 16 23:24:16 raspberry-pi-4b evcc[4328]: #011/Users/andig/htdocs/evcc/server/mqtt.go:121 +0xb8
Oct 16 23:24:16 raspberry-pi-4b evcc[4328]: github.com/evcc-io/evcc/provider/mqtt.(*Client).ListenSetter.func1({0x4b74de2, 0x2})
Oct 16 23:24:16 raspberry-pi-4b evcc[4328]: #011/Users/andig/htdocs/evcc/provider/mqtt/client.go:147 +0x44
Oct 16 23:24:16 raspberry-pi-4b evcc[4328]: github.com/evcc-io/evcc/provider/mqtt.(*Client).listen.func1({0x1bdb94c, 0x4927ce0}, {0x1bd9aa0, 0x4fd8660})
Oct 16 23:24:16 raspberry-pi-4b evcc[4328]: #011/Users/andig/htdocs/evcc/provider/mqtt/client.go:165 +0x180
Oct 16 23:24:16 raspberry-pi-4b evcc[4328]: github.com/eclipse/paho%2emqtt%2egolang.(*router).matchAndDispatch.func2()
Oct 16 23:24:16 raspberry-pi-4b evcc[4328]: #011/Users/andig/go/pkg/mod/github.com/eclipse/paho.mqtt.golang@v1.4.1/router.go:214 +0x7b8
Oct 16 23:24:16 raspberry-pi-4b evcc[4328]: created by github.com/eclipse/paho%2emqtt%2egolang.(*router).matchAndDispatch
Oct 16 23:24:16 raspberry-pi-4b evcc[4328]: #011/Users/andig/go/pkg/mod/github.com/eclipse/paho.mqtt.golang@v1.4.1/router.go:173 +0x268
```

so I guess -1 is wrong to set a guest vehicle ;-)

is it possible to set that via MQTT?

### Steps to reproduce

MQTT publish -1 to loadpoints/1/vehicle/set

### Configuration details

not needed

### Log details

not needed

### What type of operating system are you running?

Linux

### Version

0.105.2

andig, 2022-10-17:
> is it possible to set that via MQTT?

Not right now but it should not panic either. @naltatis I like the idea of using -1 for removing the vehicle?

andig, 2022-10-17:
Closed in https://github.com/evcc-io/evcc/commit/43d4221cf94000c35050ea5a56bac8ccd1375146

closed 2022-10-17 by andig

andig, 2022-10-17:
You're welcome- that was an easy one ;)