evcc-io/evcc #4879: SQL logic error: there is already another table or index with this name: sessions (1)

Saharel001, 2022-10-19:
### Describe the bug

```
Okt 19 22:47:24 SK-RPI-01 evcc[26224]: [db    ] TRACE 2022/10/19 22:47:24 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="transactions" -1 <nil>

Okt 19 22:47:24 SK-RPI-01 evcc[26224]: [db    ] TRACE 2022/10/19 22:47:24 ALTER TABLE `transactions` RENAME TO `sessions` 0 SQL logic error: there is already another table or index with t
Okt 19 22:47:24 SK-RPI-01 evcc[26224]: [main  ] FATAL 2022/10/19 22:47:24 failed configuring site: SQL logic error: there is already another table or index with this name: sessions (1)
```

### Steps to reproduce

After update from 0.104.2+1665114631 to new release, evcc start failed

### Configuration details

`# open evcc at http://evcc.local:7070  
network:  
  schema: http  
  host: *****  
  port: 7070  
  
log: trace  
#log: info   
levels:   
  cache: error   
#  lp-1: trace   
#  site: trace   
#  eebus: trace  
  
interval: 10s #  control cycle interval  
  
meters:  
- type: template  
  template: kostal-plenticore   
  id: *****  
  host: *****  
  port: 1502    
  usage: pv    
  modbus: tcpip    
  name: pv1  
- type: template  
  template: kostal-plenticore   
  id: *****  
  host: *****  
  port: 1502    
  usage: battery    
  modbus: tcpip    
  name: battery2  
- type: template  
  template: kostal-ksem-inverter   
  id: *****  
  host: *****  
  port: 1502    
  usage: grid    
  modbus: tcpip    
  name: grid3  
  
chargers:  
- type: template  
  template: ellipro   
  ski: D5E1 516B CBDA 56DC 288D 8B9A 9E58 85EF BF7B 00DD    
  ip: 192.168.178.252    
  name: wallbox5  
  
vehicles:  
- type: template  
  template: skoda   
  title: Octavia iV    
  user: *****  
  password: *****  
  capacity: 11  
  phases: 1  
  name: ev4  
  
loadpoints:  
- title: Garage  
  charger: wallbox5  
  vehicle: ev4  
  mode: pv  
  phases: 0  
  mincurrent: 6  
  maxcurrent: 16  
  resetOnDisconnect: true  
  
  
site:  
  title: PV Nutzung  
  residualPower: 200  
  meters:  
    grid: grid3  
    pvs:  
    - pv1  
    batteries:  
    - battery2  
  
tariffs:   
  currency: EUR # (default EUR)   
  grid:  
    # static grid price  
    type: fixed   
    price: 0.344 # [currency]/kWh   
  feedin:  
    # rate for feeding excess (pv) energy to the grid  
    type: fixed  
    price: 0.064 # [currency]/kWh  
  
sponsortoken: *****  
  
eebus:  
  certificate:  
      private: |  
******************
      public: |  
          -----BEGIN CERTIFICATE-----  
*******************
          -----END CERTIFICATE-----  
  `

### Log details

```
Okt 19 22:47:24 SK-RPI-01 evcc[26224]: [db    ] TRACE 2022/10/19 22:47:24 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="transactions" -1 <nil>

Okt 19 22:47:24 SK-RPI-01 evcc[26224]: [db    ] TRACE 2022/10/19 22:47:24 ALTER TABLE `transactions` RENAME TO `sessions` 0 SQL logic error: there is already another table or index with t
Okt 19 22:47:24 SK-RPI-01 evcc[26224]: [main  ] FATAL 2022/10/19 22:47:24 failed configuring site: SQL logic error: there is already another table or index with this name: sessions (1)
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.105.2

Saharel001, 2022-10-19:
I fix it with delete the db file ~/.evcc/evcc.db and lost all charging logs

andig, 2022-10-20:
In that case not much we could do now. If you still have the old database file please email to info@evcc.io for testing.

closed 2022-10-20 by andig