evcc-io/evcc #4891: ABB Terra AC: Add RS485 to template

premultiply, 2022-10-21:
Kann ich https://github.com/3cky/mbusd verwenden, um serial->tcpip auf demselben host auszuführen auf dem evcc läuft ? https://github.com/evcc-io/evcc/blob/master/templates/definition/charger/abb.yaml#L11 für abb kein rs485serial? Danke

_Originally posted by @gytisgreitai in https://github.com/evcc-io/evcc/discussions/3369#discussioncomment-3936489_

andig, 2022-10-23:
Ich bin nicht sicher, ob ich die Frage verstehe. Hier ein Charger mit Modbus TCP oder über Serial angesprochen werden soll?

gytisgreitai, 2022-10-23:
@andig 

[image]

[USB auf RS485 Adapter](https://www.ebay.de/itm/121974380677)

```yaml
chargers:
  - type: template
    template: abb
    id: 1
    modbus: rs485serial
    device: /dev/ttyUSB0
    name: wallbox6
```

andig, 2022-10-23:
Hat die Terra kein TCP?

premultiply, 2022-10-23:
Nein, momentan ist nur Modbus RTU via RS485 möglich.
Modbus TCP Unterstützung ist nur beworben und angekündigt.

closed 2022-10-24 by premultiply