evcc-io/evcc #4892: Automatic switch back to "PV Charging" after a forced charging has happened

Percy2Live, 2022-10-21:
In the cold time of the year the energy from my PV isn't sufficient for charging the car on all days. So if I know I need to drive 100km tomorrow, I check the battery status and if it's not enough, I select "fast charge" and set the slider for the target SoC to e.g. 40%.

This works fine, but it has 2 downsides: 
1. The car doesn't charge any further than 40%, even if solar energy would be available later on
2. When I forget to switch it back, the next time I plug in the car it does immediately start to charge and it does not charge further than 40% SoC

It would be great to have an option which would tell evcc to automatically switch back to PV charge + 100% SoC after such a forced charge. E.g. a single check box would be enough.

andig, 2022-10-23:
Why not use `resetOnDisconnect`?

VolkerK62, 2022-10-23:
@Percy2Live 
> The car doesn't charge any further than 40%, even if solar energy would be available later on

for this behavior you can use `minsoc` in conjuction with `PV` mode.
https://docs.evcc.io/docs/reference/configuration/vehicles#onidentify
If the soc is below `minsoc` the car is charging in `fast` mode until this value and above only in `PV` mode.

Unfortunately this cannot be set in the UI, but its available via API.

Percy2Live, 2022-10-23:
I'm aware of the minsoc feature but this would mean my car would automatically start to charge up to 60 or 80 percent when I plug it in in the evening, even when I don't need it for the next day. Also it wouldn't be possible to change the value depending on the travel distance.

VolkerK62, 2022-10-24:
not, if you use `resetOnDisconnect: true` in your config file (see comment from andig).
You can set `minsoc: 0` in your config file and change it (if needed) via API.
@andig  @naltatis  what about an UI slider for `minsoc` (similar to `targetsoc`)?

Another option: did you try Target Charging?

naltatis, 2022-10-24:
`minsoc` per vehicle will be coming to the UI. We don't have a concrete timing or details on how it will look like, but I'd assume that this will serve the here described use case quite well.