evcc-io/evcc #4893: VARTA puls neo: modbus exception

vfbler, 2022-10-21:
### Describe the bug

Hallo,

ich bekommen beim Einbinden vom Stromspeicher "VARTA pulse neo" einen Fehler, der schonmal für eine andere VARTA gemeldet wurde, nämlich

_read failed: modbus: exception '2' (illegal data address), function '4'_

Leider geht es weder mit dem VARTA-Template "mit PV" noch mit "ohne PV". Eingebunden habe ich es per "grid" und "battery".

### Steps to reproduce

-

### Configuration details

- type: template
  name: grid1
  template: varta-energiespeicher / varta-energiespeicher-battery-only
  usage: grid / battery
  host: 192.168.99.99 # Hostname
  port: 502 # Port

### Log details

read failed: modbus: exception '2' (illegal data address), function '4'

### What type of operating system are you running?

Linux

### Version

evcc version 0.105.2

premultiply, 2022-10-22:
Bitte ergänze die vollständige Ausgabe von `evcc meter --log trace`

vfbler, 2022-10-22:
Habe es eingefügt. "bat1" habe ich zum Test mit "varta-energiespeicher-battery-only" eingebunden, "grid1" mit "varta-energiespeicher". "pv1" ist der Wechselrichter der PV-Anlage, der funktioniert.

vfbler, 2022-10-22:
Hallo,

konnte es anscheinend hiermit zum Laufen bringen. "type" muss "holding" sein. Die Adressen habe ich aus [https://forum.iobroker.net/topic/44510/aufruf-sammlung-an-modbus-anbindungen/19?_=1625838642611](url)

- type: custom
  name: grid1
  power:
    source: modbus
    uri: 192.168.99.99:502
    id: 255
    register:
      address: 1078
      type: holding
      decode: int16
    scale: -1
- type: custom
  name: battery1
  power:
    source: modbus
    uri: 192.168.99.99:502
    id: 255
    register:
      address: 1066
      type: holding
      decode: int16
    scale: -1
  soc:
    source: modbus
    uri: 192.168.99.99:502
    id: 255
    register:
      address: 1068
      type: holding
      decode: uint16
- type: custom
  name: pv1
  power:
    source: modbus
    uri: 192.168.99.99:502
    id: 255
    register:
      address: 1088
      type: holding
      decode: uint16
    scale: -1

andig, 2022-10-24:
Ansonsten: die Varta Registerdoku schweigt sich drüber aus ob Input oder Holding. PR folgt.

closed 2022-10-24 by andig

vfbler, 2022-10-24:
> Dürfte ich noch aus dem Nightly um folgendes bitten:
> 
> ```
> evcc dump sunspec --id 255 192.168.99.99:502
> ```
> 
> Hintergrund: der Initialisierungsdialog sieht sehr nach SUNSPEC aus- warum nutzen wir dann nicht einfach das?

evcc dump sunspec --id 255 192.168.99.99:502
Error: unknown flag: --id
Usage:
  evcc dump [flags]

Flags:
      --cfg   Dump config file

Global Flags:
  -c, --config string   Config file (default "~/evcc.yaml" or "/etc/evcc.yaml")
  -h, --help            Help
  -l, --log string      Log level (fatal, error, warn, info, debug, trace) (default "info")
      --log-headers     Log headers
      --sqlite string   SQlite database file

unknown flag: --id