evcc-io/evcc #4894: SOC-Estimator berechnet SOC neu, obwohl kein Ladevorgang stattfindet | vehicle SOC wird ohne Ladevorgang aktualisiert, trotz poll-mode charging

blenni, 2022-10-21:
### Describe the bug

In Verbindung mit einer KEBA-Wallbox berechnet der SOC-Estimator eine SOC-Erhöhung auf Basis der im vorigen Ladevorgang geladenen Energie.
Keine Ahnung, ob das Verhalten KEBA-spezifisch ist, aber die KEBA-Wallbox meldet so lange die Energie, die im vorigen Ladevorgang geladen wurde, bis der nächste neue Ladevorgang nach Abstecken gestartet (!) wurde. Nur Ab- und wieder Anstecken reicht nicht aus, um den Zähler auf 0 zu setzen.

[image]
_Angesteckt mit 63%, kein Ladevorgang hat seitdem stattgefunden. Trotzdem werden die 7,9 kWh angezeigt._

```
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 17:53:09 charge power: 0W
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 17:53:09 charge currents: [0 0 0]A
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 17:53:09 charger status: B
evcc[2315]: [lp-1  ] INFO 2022/10/21 17:53:09 car connected
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 17:53:09 pv timer elapse
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 17:53:09 pv timer inactive
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 17:53:09 vehicle status: A
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 17:53:09 vehicle soc: 63%
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 17:53:09 vehicle range: 31km
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 17:53:09 pv charge current: 0A = 0A + -1.58A (362W @ 1p)
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 17:53:09 climater active: false, target temp: 21.0°C, outside temp: 0.0°C
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 17:53:09 pv enable timer reset
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 17:53:09 pv timer inactive
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 17:53:09 climater active: false, target temp: 21.0°C, outside temp: 0.0°C
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 17:53:19 charge power: 0W
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 17:53:29 charge power: 0W
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 17:53:29 vehicle odometer: 10319km
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 17:53:29 charge currents: [0 0 0]A
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 17:53:29 charger status: B
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 17:53:29 next soc poll remaining time: 29m39s
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 17:53:29 pv charge current: 0A = 0A + -1.56A (360W @ 1p)
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 17:53:29 climater active: false, target temp: 21.0°C, outside temp: 0.0°C
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 17:53:29 climater active: false, target temp: 21.0°C, outside temp: 0.0°C
```

Dann nach Ablauf der soc poll remaining time passiert folgendes:
```
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 18:23:09 charge currents: [0 0 0]A
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 18:23:09 charger status: B
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 18:23:09 next soc poll remaining time: 0s
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 18:23:09 pv charge current: 0A = 0A + -1.52A (349W @ 1p)
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 18:23:09 climater active: false, target temp: 21.0°C, outside temp: 0.0°C
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 18:23:09 climater active: false, target temp: 21.0°C, outside temp: 0.0°C
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 18:23:19 charge power: 0W
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 18:23:29 charge power: 0W
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 18:23:29 charge currents: [0 0 0]A
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 18:23:29 charger status: B
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 18:23:30 soc estimated: 100.00% (vehicle: 63.00%)
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 18:23:30 vehicle soc: 100%
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 18:23:30 vehicle range: 31km
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 18:23:30 pv charge current: 0A = 0A + -1.53A (351W @ 1p)
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 18:23:30 climater active: false, target temp: 21.0°C, outside temp: 0.0°C
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 18:23:30 climater active: false, target temp: 21.0°C, outside temp: 0.0°C
evcc[2315]: [lp-1  ] DEBUG 2022/10/21 18:23:39 charge power: 0W
```
Siehe um 18:23:30 `soc estimated: 100.00% (vehicle: 63.00%)`

Mir ist auch aufgefallen, dass mein poll-mode auf charging steht, in diesem Zustand also gar nicht aktualisiert werden sollte.
Kann das was mit https://github.com/evcc-io/evcc/issues/3981 zu tun haben? Meine Dauer zählt nämlich auch weiter, obwohl kein Ladevorgang stattfindet.

### Steps to reproduce

- Ladevorgang mit einer Energiemenge beenden, die einen deutlichen SOC-Hub bedeuten würde.
- Fahrzeug abstecken und ggf. fahren
- Fahrzeug anstecken und NICHT laden (PV-Modus)
- soc poll remaining timer abwarten
- **SOC springt nach oben**

### Configuration details

[evcc.yaml.txt](https://github.com/evcc-io/evcc/files/9842743/evcc.yaml.txt)

### Log details

Siehe Ausschnitte oben

### What type of operating system are you running?

Linux

### Version

`evcc version 0.105.2`

Hofyyy, 2022-10-22:
Ich hab da eine idee, gucke morgen mal rein.

Hofyyy, 2022-10-22:
Hallo Blenni,

ich habe ein paar Rückfragen die Du hoffenlich beantworten kannst.

1. Also wenn das Fahrzeug abgesteckt ist sollte "immer" der Charger Status: "A" sein und der Ladevorgang beendet werden.
    Kannst du das bestätigen?
2. Ist es richtig das der Fehler erst ab Version 0.105.* aufgetaucht ist?
    Dann liegt es recht sicher an folgendem Commit: https://github.com/evcc-io/evcc/pull/4656

@andig 
Wollen wir das es im PV mode wieder abgeschaltet ist?, oder muss der Estimator gefixt werden? Es scheint nämlich so als wenn er in dem Fall falsche Werte vorhersagt.

blenni, 2022-10-22:
Hi, danke für die Rückmeldung.

Zu 1: Ja genau, Charge Status wechselt dann auf A und der Ladevorgang stoppt natürlich. Allerdings setzt die Keba ihren Energiezähler dann noch nicht zurück.

Zu 2: Kann sehr gut sein, mir ist es zumindest davor nicht aufgefallen. Und ich habe erst vor ein paar Tagen von der 0.98 auf die 104.2 und dann wieder nach Release auf die 105.2 aktualisiert, hab es jetzt nur damit gesehen. Und es gab davor ja auch schon ein paar schlechtere Tage, wo direkt nach dem Anstecken nicht genug Überschuss vorhanden war. 

Ich schau mir später in Ruhe noch an, was der PR in Bezug auf
_Added special case for poll mode charging(), to force last correct API poll._
macht, vielleicht kann ich ja noch was sinnvolles zu dem Thema beitragen ;)

Hofyyy, 2022-10-22:
Also mein Vorschlag wäre folgender.

 Du stellst auf Poll Mode: Connected und wir beheben alles was wir finden. Und dann gucken wir uns den Sonderfall mit PollMode:Charging an.

Wenn ich raten müsste haben wir hier 2-3 probleme

tmeidinger, 2022-10-23:
Hab den Seiteneffekt auch mit einem go-eCharger (poll mode: always). Tritt bei mir auf, wenn evcc neu gestartet wird (z.B. durch ein Update). Das Fahrzeug bleibt weiterhin mit dem go-eCharger verbunden und der merkt sich auch die bisherige Lademenge. Nach dem Neustart von evcc rechnet er die bisherige Lademenge vom go-eCharger direkt auf den SoC drauf. Mein Workaround ist, den go-eCharger bei einem Update von evcc auch zu rebooten (payload=rst=1).

blenni, 2022-10-23:
Hab jetzt mal auf Poll Mode auf Connected gestellt und Trace Log für die Charger aktiviert.
Mittelfristig würde ich gerne wieder auf Poll Mode Charging wechseln.

@Hofyyy Heute Abend müsste ich das nochmal beobachten können, soll ich auf etwas spezielles achten, wenn du 2-3 Probleme vermutest?

Ich glaube der Fall von @tmeidinger ist mir auch schon aufgefallen.

Hofyyy, 2022-10-23:
Also mein Commit bewirkt das auch beim Connecten noch nachgefragt wird wenn SOc nicht target SOC ist. das löst ein probem beim Ende vom laden, ist aber ein workaround. 

Für reines PV laden, wo das Auto 24 Stunden verbunden sein kann ohne zu laden, kann das von nachteil sein.

Hier könnten wir noch eine bedingung hinzufügen. Wenn Nicht PV. Bei allen anderen lademodi läd er ja immer weiter bis er fertig ist. also minpv oder sofort. 

2.
Das hat aber nichts mit deinen anderen Problemen zu tun. daher glaube ich es sind hier min zwei punkte.

blenni, 2022-10-23:
Das Ändern des Poll Modes hilft erstmal nicht weiter (dazwischen kein Laden gestartet):

```
raspberrypi evcc[29765]: [lp-1  ] INFO 2022/10/23 17:10:27 car connected
raspberrypi evcc[29765]: [lp-1  ] DEBUG 2022/10/23 17:10:27 pv timer elapse
raspberrypi evcc[29765]: [lp-1  ] DEBUG 2022/10/23 17:10:27 pv timer inactive
raspberrypi evcc[29765]: [lp-1  ] DEBUG 2022/10/23 17:10:27 vehicle status: B
raspberrypi evcc[29765]: [lp-1  ] DEBUG 2022/10/23 17:10:27 vehicle soc: 41%
...
raspberrypi evcc[29765]: [lp-1  ] DEBUG 2022/10/23 17:40:47 soc estimated: 89.97% (vehicle: 41.00%)
raspberrypi evcc[29765]: [lp-1  ] DEBUG 2022/10/23 17:40:47 vehicle soc: 89%
```

> Hier könnten wir noch eine bedingung hinzufügen. Wenn Nicht PV. Bei allen anderen lademodi läd er ja immer weiter bis er fertig ist. also minpv oder sofort.

Ja, das würde in meinem Fall vermutlich helfen - nicht aber in dem Fall von @tmeidinger, oder?

Hofyyy, 2022-10-23:
1. Genau, du hast einen Fehler im Poll mode im Zusammenhang mit dem Estimator
Das hat nichts zu tun mit dem Commit https://github.com/evcc-io/evcc/pull/4656
@premultiply  Sieht du das genau so?

Also wenn wir in meinem Commit, den PV MOde ausschließen, was wahrscheinlich eine gute Idee ist,
hat blennie kein Problem mehr.

Allerdings scheint das eigentliche Problem der Pollmode + Estimator zu sein.

premultiply, 2022-11-01:
Das Problem ist "nur", dass die Keba den Session-Zähler nicht beim Connect resettet. Problem bei Keba.
Das muss man im Charger lösen. Notfalls durch Nutzung des Total-Meters und die Sessionverarbeitung evcc überlassen.

andig, 2022-11-01:
Das ist ja schnell gelöst: https://github.com/evcc-io/evcc/pull/5009. Die Sessionenergie rechnet ohnehin der Loadpoint aus wenn der Charger das nicht kann.

PR #5009 (2022-11-01): Keba: remove charge rater as it doesn't reset when reconnecting
Should fix #4894
closed 2022-11-01 by andig