evcc-io/evcc #4931: OCPP phase switching implementation

tarelda, 2022-10-24:
**Is your feature request related to a problem? Please describe.**
I have working proof of concept that I tested against real charger (Alfen Eve). I am working on cleaning it all up to make proper pull request, but I am not very familiar with EVCC and golang, so I need some guidance.

**Describe the solution you'd like**
1. As I discovered EVCC disables and enables charger during phase change. Does this action require complete halt of charging? 
2. Or can this behaviour be changed to simple call to phase switch API and leaving up to charger implementation if it wants to disable/enable charger?
3. What is preferred approach to state storage in EVCC? ATM most is stored inside OCPP struct and somewhere in CP and CS objects, but this doesn't offer any form of persistence. 
4. How can I access loadpoint configuration? For example maximum available current.

andig, 2022-10-24:
This issue mixes different topics. Can we please split the discussion into OCPP and general loadpoint behaviour?

> How can I access loadpoint configuration? For example maximum available current.

From charger? You can but shouldn't if not necessary. Why would you?

tarelda, 2022-10-24:
> This issue mixes different topics. Can we please split the discussion into OCPP and general loadpoint behaviour?

All my questions are directly coupled with the OCPP charger functionality that I am trying to implement. Obviously if you want I can open up another issue regarding loadpoint behaviour in phase switching. 
In this case it is only relevant, because I have wallbox that supports phase switching during transaction and restarting it is kinda pointless. Also proposed configuration key (ConnectorSwitch3to1PhaseSupported) gating functionality is for phase switching during transaction. Otherwise EnablePhaseSwitching is feasible enough to be required. To be honest, even this requirement is kinda redundant, because EVERY OCPP compatible 3 phase charger is required to be able to downgrade charging to 1 phase if cable or vehicle need it. 

> > How can I access loadpoint configuration? For example maximum available current.
> 
> From charger? You can but shouldn't if not necessary. Why would you?

Sanity check. It would be dumb to set 1 phase profile maximum current lower than maxCurrent setting of loadpoint.

andig, 2022-10-24:
Phase switching is already in the code, just needs be uncommented. Start/stop before doing so is a loadpoint topic and not related to ocpp.

andig, 2022-10-24:
Ps.: wäre es einfacher auf deutsch zu schreiben?

closed 2022-10-25 by tarelda