evcc-io/evcc #4937: Support of beta.api.daheimladen.com and api.daheimladen.com

ownakasimon, 2022-10-24:
**Is your feature request related to a problem? Please describe.**
The api of daheimladen.com is a static const variable. 
-> repo/charger/daheimladen/const.go

**Describe the solution you'd like**
There is also a beta environment of daheimladen.com -> beta.api.daheimladen.com. If you put the api_url in the evcc.yaml you can change this value for your evcc instance. If the parameter api_url is empty, the static value „https://api.daheimladen.com/v1“ will be set.

**Describe alternatives you've considered**
I checked already to fix it with DNS Records, issue is the SSL certificate do not allow other IPs for the domain api.daheimladen.com.

**Additional context**
Here an example of the config for daheimladen.com beta space before change: 

```
chargers:
   - name: wallbox_hauswand
     type: template
     template: daheimladen
     token: eyJhbGciOiJIUxxxxxxx # Sie bekommen dieses Access Token vom Daheimladen Support: info@daheimladen.de. # Optional
     stationid: 8627766459579 # Die Seriennummer, welche auf dem Bildschirm der Wallbox angezeigt wird. # Optional 
```

After change:
```
chargers:
   - name: wallbox_hauswand
     type: template
     template: daheimladen
     api_url: https://beta.api.daheimladen.com/v1
     token: eyJhbGciOiJIUxxxxxxx # Sie bekommen dieses Access Token vom Daheimladen Support: info@daheimladen.de. # Optional
     stationid: 8627766459579 # Die Seriennummer, welche auf dem Bildschirm der Wallbox angezeigt wird. # Optional 
```

andig, 2022-10-25:
@utsavanand2 is the beta api something you'd like to support?

StevieC121176, 2022-10-25:
Hallo,

Ein Grund für mich wäre z.Bsp. das aktuell nur bei der Beta Version das WLAN der Daheimladen Wallbox richtig funktioniert. Mit der normalen Firmware lässt sie sich z.Bsp. nicht über EVCC steuern. Es gibt ja eine EVCC Version für die Beta, die ist aber nicht sehr aktuell (V0.98)

andig, 2022-10-25:
Was hat die Firmware mit der Api Url für evcc zu tun? M.e. Gehört das nicht zusammen. 

Waiting for feedback from @utsavanand2

utsavanand2, 2022-10-25:
@andig Hi Andreas, as @StevieC121176 said, they would like to have the latest version of evcc which points to the daheimladen beta api which we use for testing with a small group of users, as this is on a temporary basis, I can provide the users with an updated version of the binary.

To make things more clear, we're currently trying to figure out an issue with v0.16.0 of the ocpp-go library, which seems to work with a small set of charging stations but stops responding when a larger set of stations connect to it https://github.com/lorenzodonini/ocpp-go/issues/136

Once we've fixed the issue in the library, most users would want to switch to the production environment anyway.

StevieC121176, 2022-10-25:
Nein das gehört nicht zusammen, das ist richtig. Nur könnten dann die Leute mit den WLAN Problemen der Wallbox auch die neuen EVCC Versionen benutzen. Bei mir z.Bsp. läuft die Beta Version der Wallbox und ich kann nur die 0.98 Version von EVCC benutzen. Schöner wäre es trotz der Beta Firmware in der Wallbox die aktuellen EVCC Versionen benutzen zu können.

utsavanand2, 2022-10-25:
@StevieC121176 I can provide you with the updated firmware binary which would work with your wallbox in the beta environment :)

andig, 2022-10-25:
Sounds like we have a process ;). Closing as wontfix.

utsavanand2, 2022-10-25:
@ownakasimon @StevieC121176 I've updated the gist for you to install the latest version of evcc 0.105.2
https://gist.github.com/utsavanand2/9706934dd47801ec8c36587c94b07915

ownakasimon, 2022-10-26:
issue is, that i‘m using evcc on homeassistant. I‘m not sure to get this work, only because there is a static url an no possibility as parameter :( @andig @utsavanand2 

thx for the responses!

ownakasimon, 2022-10-29:
@andig could you check this code for this issue?
https://github.com/evcc-io/evcc/pull/4982

andig, 2022-10-29:
> @ownakasimon why should the beta api be supported?
 
 my question is still unanswered?

ownakasimon, 2022-10-30:
@andig, sorry @utsavanand2 did a detailed explaination. You can not use daheimladen api at the moment because of "error 500" if evcc is starting and stopping the charge process. I think there is an old wifi lib and on the beta environment there is already everything fixed, i tested it with postman and scripting tools by my own.

What's your problem with an apiUrl and default value? You get more flexible there and I do not have to change any dockerized deployed files on homeassistant infrastructure. :-)

andig, 2022-10-30:
@utsavanand2 do you want users be able to point to the beta api?