evcc-io/evcc #4939: EVCC not ending charge after target SoC is reached, also starting charge above target SoC, in PV mode when PV power is 0W

Arquiteto, 2022-10-24:
### Describe the bug

My EVCC installed as an add-on controling charge for Nissan Leaf via Shelly 1PM is sometimes stuck on charging above target SoC, especially after long charge. Just now it reached 82% charge when target was 80%. It didn't react to changing mode from min+PV to off, it didn't react to manually switching off the relay (it turned it back on). It only reacted to changing vehicle from Leaf to guest, but it started charging again after I changed it back to Leaf - even tough my default mode is PV and there is no sun at the moment.

### Steps to reproduce

1. Start charging the car
2. Wait until SoC is over target
3. Charging doesn't stop
4. Try to turn it off by switching off relay, changing mode to off, changing vehicle to guest and back again

### Configuration details

```
site:
- title: Dom # display name for UI
  meters:
    grid: my_grid # grid meter reference
    pvs: 
    - goodwe # first pv meter reference
    - sofar
  residualPower: 0

log: info
levels:
  lp-1: debug
 
loadpoints:
- title: Garage
  charger: shelly_charger
  vehicle: leaf
  meter: shelly_meter
  minCurrent: 10
  maxCurrent: 10
  phases: 1
  disable: 
    threshold: 800
  enable: 
    threshold: 0

meters:

  
  # Shelly z HA REST API 
  - name: my_grid
    type: custom
    power:
      source: calc
      add:
        - source: http
          uri: http://192.168.xxx.xxx:8123/api/states/sensor.moc_oddawana
          method: GET # default HTTP method
          headers:
          - content-type: application/json
          - authorization: "Bearer 123456789"
          insecure: false
          jq: .state
          scale: -1
          timeout: 10s
        - source: http
          uri: http://192.168.xxx.xxx:8123/api/states/sensor.moc_pobierana
          method: GET # default HTTP method
          headers:
          - content-type: application/json
          - authorization: "Bearer 123456789"
          insecure: false 
          jq: .state
          timeout: 10s
    energy:
      source: calc
[... 524 lines cut ...]
[lp-1  ] DEBUG 2022/10/24 23:23:55 charge power: 2258W
[lp-1  ] DEBUG 2022/10/24 23:23:55 charger status: C
[lp-1  ] DEBUG 2022/10/24 23:23:55 soc estimated: 85.10% (vehicle: 85.00%)
[lp-1  ] DEBUG 2022/10/24 23:23:55 vehicle soc: 85%
[lp-1  ] DEBUG 2022/10/24 23:23:55 vehicle range: 120km
[lp-1  ] DEBUG 2022/10/24 23:23:55 targetSoC reached: 85.0% > 80%
[lp-1  ] DEBUG 2022/10/24 23:23:55 climater active: true, target temp: 22.0°C, outside temp: 22.0°C
[lp-1  ] DEBUG 2022/10/24 23:23:55 climater active
[lp-1  ] DEBUG 2022/10/24 23:24:05 charge power: 2258W
[lp-1  ] DEBUG 2022/10/24 23:24:05 charger status: C
[lp-1  ] DEBUG 2022/10/24 23:24:05 soc estimated: 85.10% (vehicle: 85.00%)
[lp-1  ] DEBUG 2022/10/24 23:24:05 vehicle soc: 85%
[lp-1  ] DEBUG 2022/10/24 23:24:05 vehicle range: 120km
[lp-1  ] DEBUG 2022/10/24 23:24:05 targetSoC reached: 85.0% > 80%
[lp-1  ] DEBUG 2022/10/24 23:24:05 climater active: true, target temp: 22.0°C, outside temp: 22.0°C
[lp-1  ] DEBUG 2022/10/24 23:24:05 climater active
```

### What type of operating system are you running?

Docker container

### Version

0.105.2 as hassio-addon

Arquiteto, 2022-10-24:
And here is the log of what happens when I change vehicle back from guest to Nissan

```
`lp-1  ] DEBUG 2022/10/24 23:47:25 charge power: 0W
[lp-1  ] DEBUG 2022/10/24 23:47:25 charger status: B
[lp-1  ] DEBUG 2022/10/24 23:47:35 charge power: 0W
[lp-1  ] DEBUG 2022/10/24 23:47:35 charger status: B
[lp-1  ] INFO 2022/10/24 23:47:39 vehicle updated: unknown -> Nissan
[lp-1  ] DEBUG 2022/10/24 23:47:39 set charge mode: pv
[lp-1  ] DEBUG 2022/10/24 23:47:39 pv timer elapse
[lp-1  ] DEBUG 2022/10/24 23:47:39 pv timer inactive
[lp-1  ] DEBUG 2022/10/24 23:47:39 set target soc: 80
[lp-1  ] DEBUG 2022/10/24 23:47:39 charge power: 0W
[lp-1  ] DEBUG 2022/10/24 23:47:40 charger status: B
[lp-1  ] DEBUG 2022/10/24 23:47:51 pv charge current: 0A = 0A + -5.13A (1181W @ 1p)
[lp-1  ] DEBUG 2022/10/24 23:47:54 climater active: true, target temp: 22.0°C, outside temp: 22.0°C
[lp-1  ] DEBUG 2022/10/24 23:47:54 charger enable
[lp-1  ] DEBUG 2022/10/24 23:47:54 wake-up timer: start
[lp-1  ] DEBUG 2022/10/24 23:47:54 charge power: 0W
[lp-1  ] DEBUG 2022/10/24 23:47:54 charger status: B
[lp-1  ] DEBUG 2022/10/24 23:47:56 pv charge current: 0A = 0A + -5.08A (1169W @ 1p)
[lp-1  ] DEBUG 2022/10/24 23:47:56 climater active: true, target temp: 22.0°C, outside temp: 22.0°C
[lp-1  ] DEBUG 2022/10/24 23:47:56 charge power: 0W
[lp-1  ] DEBUG 2022/10/24 23:47:56 charger status: B
[lp-1  ] DEBUG 2022/10/24 23:47:57 pv charge current: 0A = 0A + -5.08A (1169W @ 1p)
[lp-1  ] DEBUG 2022/10/24 23:47:57 climater active: true, target temp: 22.0°C, outside temp: 22.0°C
[lp-1  ] DEBUG 2022/10/24 23:47:58 charge power: 0W
[lp-1  ] DEBUG 2022/10/24 23:47:59 charger status: C
[lp-1  ] DEBUG 2022/10/24 23:47:59 charge start energy: 189kWh
[lp-1  ] INFO 2022/10/24 23:47:59 start charging ->
[lp-1  ] DEBUG 2022/10/24 23:48:00 pv charge current: 1.23A = 10A + -8.77A (2016W @ 1p)
[lp-1  ] DEBUG 2022/10/24 23:48:00 climater active: true, target temp: 22.0°C, outside temp: 22.0°C
[lp-1  ] DEBUG 2022/10/24 23:48:05 charge power: 2255W
[lp-1  ] DEBUG 2022/10/24 23:48:05 charger status: C
[lp-1  ] DEBUG 2022/10/24 23:48:06 pv charge current: 1.23A = 10A + -8.77A (2016W @ 1p)
[lp-1  ] DEBUG 2022/10/24 23:48:06 climater active: true, target temp: 22.0°C, outside temp: 22.0°C
[lp-1  ] DEBUG 2022/10/24 23:48:15 charge power: 2255W
[lp-1  ] DEBUG 2022/10/24 23:48:15 charger status: C
[lp-1  ] DEBUG 2022/10/24 23:48:19 vehicle status: C
[lp-1  ] DEBUG 2022/10/24 23:48:19 vehicle soc: 83%
[lp-1  ] DEBUG 2022/10/24 23:48:19 vehicle range: 119km
[lp-1  ] DEBUG 2022/10/24 23:48:19 targetSoC reached: 83.0% > 80%
[lp-1  ] DEBUG 2022/10/24 23:48:19 climater active: true, target temp: 22.0°C, outside temp: 22.0°C
[lp-1  ] DEBUG 2022/10/24 23:48:19 climater active`
```

andig, 2022-10-25:
Kein Wunder da bei Deinem Auto die Klimatisierung läuft:

> climater active: true

closed 2022-10-25 by andig