evcc-io/evcc #4954: Add Modbus Proxy

andig, 2022-10-25:
Ist vielleicht keine Kernkompetenz, aber wir könnten mittelfristig auch mal überlegen, den Modbus Proxy mit in evcc einzubauen. Der würde dann upstream immer ModbusTCP sprechen und downstream das Protokoll der jeweiligen Verbindung. Je geöffneter Verbindung müssten wir einen Port upstream anbieten. Das wäre insofern einfacher, als dass keine zusätzlichen Komponenten notwendig wären und die evcc Config "clean" bliebe.

_Originally posted by @andig in https://github.com/evcc-io/evcc/discussions/3088#discussioncomment-3960590_

andig, 2022-10-26:
Nicht nötig falls wir das hier umsetzen sollten- dann kann ja jeder selbst über Modbus lesen.

MarkusGH, 2022-10-26:
Klar kann man MBMD dann auch als Client auf den Proxy zugreifen lassen.
Aber wenn EVCC schon als primärer Zugriffsweg zum WR (und im Fall SolarEdge auch zum Meter am Hausanschluss) fungiert wäre es halt nett die ohnehin gelesenen Daten auch per MQTT zu publizieren.

MarkusGH, 2022-10-26:
Stimmt auch wieder. Dann wäre der Export über MQTT eher ein Job für die Komponente, die verallgemeinert Wechselrichter ausliest und hat hiermit nichts zu tun.

andig, 2022-10-30:
Der Proxy selbst funktioniert schon laut Testskript. Es fehlt noch eine geeignete Konfiguration, Vorschläge willkommen.

andig, 2022-11-02:
@MarkusGH willst du mal testen?

MarkusGH, 2022-11-03:
Leider nein - keine Zeit. Ich werde die Funktion wohl ohnehin nicht nutzen, da ich mbmd zum Zugriff auf den Wechselrichter verwende. Hätte der direkte Zugriff via solaredge-hybrid bzw. solaredge-inverter denn Vorteile gegenüber dem Lesen von Power/Energy und Currents per MQTT?

closed 2022-11-03 by andig