evcc-io/evcc #4957: Ocpp: support stack level 0

andig, 2022-10-25:
:
```
Es wurde tatsächlich ein Fehler bei den Logs gefunden, aber scheint dies nicht von uns unterstützt zu werden: 
StackLevel:1
we only do support StackLevel: 0
```
Ist es möglich das in evcc anzupassen?

_Originally posted by @FloEisen in https://github.com/evcc-io/evcc/discussions/4762#discussioncomment-3962075_

tarelda, 2022-10-25:
Change [in this line] (https://github.com/evcc-io/evcc/blob/2905c0a3eef625b9157bac87b231d23ea014af65/charger/ocpp.go#L358) 1 to 0. 0 is lowest allowed number for this field.
Effectively means that this charger doesn't support charging profile composition.

andig, 2022-10-25:
Können wir dann trotzdem Max Profile und Tx Profile auf das gleiche Level setzen?

tarelda, 2022-10-25:
> Können wir dann trotzdem Max Profile und Tx Profile auf das gleiche Level setzen?

Yes. They have different meaning for charge point.

tarelda, 2022-10-28:
After 59e5ad7 I think we can close this.

tarelda, 2022-10-28:
Also yesterday, I found [this example ](https://github.com/mobilityhouse/ocpp/issues/372#issuecomment-1250080881). Only difference between our procedure is that we don't clear old profiles (but I have that in workings).

closed 2022-10-28 by andig

andig, 2022-10-28:
Do you know what would

    call.ClearChargingProfilePayload

translation into in terms of OCPP requests?

tarelda, 2022-10-28:
@andig 25ec859

Two interesting observations regarding my Alfen:
- soft reset interrupts ongoing transaction, but doesn't go back to three phases but keeps last phases

- clear charging profiles causes charger to use max available current (still one phase instead of three) and leaves this in the log: 
[image]

andig, 2022-10-29:
> soft reset interrupts ongoing transaction, but doesn't go back to three phases but keeps last phases

You're only doing that at start, right? Then it shouldn't matter. It is loadpoint responsibility to select correct phases at startup. As explained, loadpoint starts in "don't know" mode and should try to figure it out.