evcc-io/evcc #4965: MQTT Zähler Gesamte Einspeisung

TobiG92, 2022-10-26:
### Discussed in https://github.com/evcc-io/evcc/discussions/4959

<div type='discussions-op-text'>

<sup>Originally posted by **TobiG92** October 26, 2022</sup>
Hallo, ich lesen meinen SHM2.0 mit evcc aus und nutze MQTT um diese Daten an mein Smarthome weiter zu reichen. Der aktuelle Bezug, die aktuelle Einspeisung und der gesamte Bezug aus dem Netz stehen zu Verfügung. Lediglich fehlt mir die gesamte Einspeisung. Besteht die Möglichkeit diese Variable über MQTT bereit zu stellen? 

Also sowas wie „grid_feed-in_energy“. 

Vorab vielen Dank für das tolle Projekt. 

</div>

premultiply, 2022-10-27:
Derzeit verarbeitet evcc pro **meter** nur einen `energy` Zählerstand.
Hier wäre ein (optionaler) zweiter Zählerstand erforderlich.

TobiG92, 2022-10-27:
Der SHM 2.0 und viele andere Zähler liefern diesen Wert ja. Wäre echt super wenn das ergänzt werden könnte.

andig, 2022-10-27:
Mehrfach diskutiert, derzeit haben wir keine gute Idee dafür.

closed 2022-10-27 by andig