evcc-io/evcc #4967: Huawei Sun2000 mit zwei Batterien: Ladeleistung wird nur von Batterie 1 berücksichtigt.

l33tn00b, 2022-10-27:
### Describe the bug

Bei der Anzeige der Ladeleistung des Batteriesystems (rot umrahmt in der folgenden Grafik) berücksichtigt das Dashboard nur eine Batterie.
[image]
Ich habe hier allerdings zwei Batterien (in Hauwei-Sprech "ESU") im Verbund (als "ESS") am Wechselrichter hängen, die zum gegebenen Zeitpunkt etwa das doppelte an Leistung ziehen. 
Damit ist auch der In-Haus-Verbrauch (der sich als Differenz aus erzeugter Leistung und Ladeleistung des Batteriesystems ergibt) zu hoch (um den fehlenden Betrag der Ladeleistung der zweiten Batterie).

Bug oder Feature?

Habe versucht, eine zweite Batterie anzulegen, das hat am Verhalten nichts geändert.

### Steps to reproduce

Sun2000 + 2 Batterien im Systemverbund über SDongle mit dem Leistungsmesser im Sicherungskasten.

### Configuration details

meters:
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: <ip>
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: <ip>
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: <ip>
  port: 502
  usage: battery
  modbus: tcpip
  name: battery3
- type: template
  template: huawei-dongle-powersensor
  id: 1
  host: <ip>
  port: 502
  usage: battery
  modbus: tcpip
  name: battery4

### Log details

[main  ] INFO 2022/10/27 17:07:31 evcc 0.105.2
[main  ] INFO 2022/10/27 17:07:31 using config file: /home/pi/evcc.yaml
[main  ] INFO 2022/10/27 17:07:31 listening at :7070
[site  ] INFO 2022/10/27 17:07:36 site config:
[site  ] INFO 2022/10/27 17:07:36   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2022/10/27 17:07:36     grid:      power ✓ energy ✓ currents ✓
[site  ] INFO 2022/10/27 17:07:36     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2022/10/27 17:07:36     battery 1: power ✓ energy ✗ currents ✗ soc ✓
[site  ] INFO 2022/10/27 17:07:36   vehicles:
[site  ] INFO 2022/10/27 17:07:36     vehicle 1: range ✓ finish ✗ status ✓ climate ✗ wakeup ✗
[lp-1  ] INFO 2022/10/27 17:07:36 loadpoint 1:
[lp-1  ] INFO 2022/10/27 17:07:36   mode:        pv
[lp-1  ] INFO 2022/10/27 17:07:36   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2022/10/27 17:07:36   meters:      charge ✓
[lp-1  ] INFO 2022/10/27 17:07:36     charge:    power ✓ energy ✗ currents ✓
[lp-1  ] DEBUG 2022/10/27 17:07:36 phase timer inactive
[lp-1  ] DEBUG 2022/10/27 17:07:36 pv timer inactive

### What type of operating system are you running?

Linux

### Version

evcc version 0.105.2

VolkerK62, 2022-10-27:
und du hast beide Batterien auch in der `site` angegeben?

l33tn00b, 2022-10-27:
Ich glaube, du kannst die Meldung schließen. *plonk*
Danke! War ein Tippfehler.

closed 2022-10-27 by premultiply