evcc-io/evcc #4977: OCPP: startup issues

tarelda, 2022-10-28:
Two interesting observations regarding my Alfen:
- soft reset interrupts ongoing transaction, but doesn't go back to three phases but keeps last phases

- clear charging profiles causes charger to use max available current (still one phase instead of three) and leaves this in the log: 
[image]

_Originally posted by @tarelda in https://github.com/evcc-io/evcc/issues/4957#issuecomment-1295324899_

tarelda, 2022-10-29:
> > soft reset interrupts ongoing transaction, but doesn't go back to three phases but keeps last phases
> 
> You're only doing that at start, right? Then it shouldn't matter. It is loadpoint responsibility to select correct phases at startup. As explained, loadpoint starts in "don't know" mode and should try to figure it out.

@andig Yes. Unfortunately, "figure it out" state doesn't happen, despite having current readings from charger.

tarelda, 2022-10-29:
@andig my drafts are [here](https://github.com/tarelda/evcc/tree/ocpp-startup-issues)

andig, 2022-10-29:
If there is a 1p3p problem then it is probably not related to OCPP, so it can't be fixed in OCPP. Let's figure the problem out. Could you share config/logfile?

tarelda, 2022-10-29:
> If there is a 1p3p problem then it is probably not related to OCPP, so it can't be fixed in OCPP. Let's figure the problem out. Could you share config/logfile?

Sure.

```
interval: 10s

meters:
  - name: home-grid
    type: template
    template: huawei-dongle-powersensor
    usage: grid
    modbus: tcpip
    id: 1
    host: *****
    port: 502
  - name: carport-solar
    type: template
    template: huawei-dongle-powersensor
    usage: pv
    modbus: tcpip
    id: 1
    host: ****
    port: 502

chargers:
  - name: carport-wallbox
    type: template
    template: ocpp
    idtag: ****

site:
  title: Dom
  meters:
    grid: home-grid
    pvs:
      - carport-solar

loadpoints:
  - title: Wiata
    charger: carport-wallbox
    mode: "off"
    phases: 0
    minCurrent: 6
    maxCurrent: 16
```

Regarding logs, due to having anydesk on my dev machine (it also uses :7070) and these startup issues, I accidently left my EV charging and I have to discharge it to get some fresh logs.

andig, 2022-11-01:
> I have to discharge it to get some fresh logs.

Waiting for log to show the problem (after restart). Config looks fine.

tarelda, 2022-11-02:
Also [this line](https://github.com/evcc-io/evcc/blob/25ec859161719187e5276b9265a1c762cd142496/charger/ocpp.go#L252) was commented out for this test.

tarelda, 2022-11-02:
I played a little bit with init sequence and came with this draft -  1097fb85c29ef9238b49a6a8d54a67e947090e42 
Still no luck with that phase mismatch. I think there is no way around loadpoint enforcing 3phases when it guesses that charger is using 3 phases.

andig, 2022-11-03:
@tarelda seems I never sent my answer. I don't see any mismatch (and to recap: if there is one it's not in OCPP so it can't be fixed in OCPP). The real problem is this:

    [site  ] DEBUG 2022/11/02 10:32:11 grid currents: [2.18 8.56 0.31]A
    [site  ] DEBUG 2022/11/02 10:32:12 site power: 2584W
    [lp-1  ] DEBUG 2022/11/02 10:32:12 charger status: B

you are already charging but status is B. The problem here is that after restart the running transaction is not recognised.

tarelda, 2022-11-03:
@andig you are referring to second part of the log, right?

Then look at this:
```
[site  ] DEBUG 2022/11/02 10:33:07 ----
[lp-1  ] DEBUG 2022/11/02 10:33:07 charge power: 4140W
[site  ] DEBUG 2022/11/02 10:33:07 pv power: 0W
[site  ] DEBUG 2022/11/02 10:33:08 grid power: 3732W
[site  ] DEBUG 2022/11/02 10:33:11 grid currents: [2.21 14.3 0.31]A
[site  ] DEBUG 2022/11/02 10:33:12 site power: 3732W
[lp-1  ] DEBUG 2022/11/02 10:33:12 charger status: C
[lp-1  ] DEBUG 2022/11/02 10:33:12 next soc poll remaining time: 59m51s
[lp-1  ] DEBUG 2022/11/02 10:33:12 vehicle soc: 30%
[lp-1  ] DEBUG 2022/11/02 10:33:12 vehicle range: 143km
[lp-1  ] DEBUG 2022/11/02 10:33:12 vehicle target soc: 80%
[lp-1  ] DEBUG 2022/11/02 10:33:12 available power 408W < 4140W min 3p threshold
[lp-1  ] DEBUG 2022/11/02 10:33:12 phase scale1p in 2m8s
[lp-1  ] DEBUG 2022/11/02 10:33:12 pv charge current: 0.591A = 6A + -5.41A (3732W @ 3p)
```

I had idle usage of 8A on second phase during this test. This shows that loadpoint is thinking that is drawing 4140W (6A) from 3 phases, but site shows that only one phase is used (14-8A = 6A). 
If there is not enough surplus then phase scale1p timer goes off. Charging stops. Loadpoint scales 1 phase to 1 phase. 
But if I switch to fast before that it just keeps charging in 1p with 16A which is obviously wrong. 

Status B according to api.go means `Vehicle: present, Charging: not active, Cable: connected` ?

tarelda, 2022-11-03:
@andig First part of the log shows that 1097fb85c29ef9238b49a6a8d54a67e947090e42 closes any ongoing transaction during startup. Current master doesn't.

andig, 2022-11-03:
> But if I switch to fast before that it just keeps charging in 1p with 16A which is obviously wrong.

Where can I see that in the log with enough context? The loadpoint _does_ switch to 3p when going to now it it doesn't know the phases!

tarelda, 2022-11-03:
@andig I will try to make more logs in the evening when car will be back in garage.  

Can I "fool" EVCC to think that it needs to use 3 phases in minpv mode if there is no PV power by setting site residualPower to -2760 (1380+2760=4140)?

andig, 2022-11-03:
I don't think fooling will help here. We need prove that a simple scenario doesn't work:
- charge 1p
- restart evcc
- switch to "now" 
- check it charges 3p

For Ocpp that requires the soft reset afaikt or it won't recognise the running session.

In other words: I'm more and more unsure what this issue really is about. Shall we close and reopen one for specific 1p startup problem?

andig, 2022-11-03:
> I played a little bit with init sequence and came with this draft - [1097fb8](https://github.com/evcc-io/evcc/commit/1097fb85c29ef9238b49a6a8d54a67e947090e42)
> Still no luck with that phase mismatch. I think there is no way around loadpoint enforcing 3phases when it guesses that charger is using 3 phases.

Feel free to open PR. I'm unclear that the expectation is that should do (except for the reset).

tarelda, 2022-11-03:
> I don't think fooling will help here. We need prove that a simple scenario doesn't work:
> 
> * charge 1p
> * restart evcc
> * switch to "now"
> * check it charges 3p
Yes, but I want to get logs for scenario that requires certain amount of energy produced from the PV installation. In the night it is not normally possible. That's why I asked for advice about configuring EVCC.
 
> For Ocpp that requires the soft reset afaikt or it won't recognise the running session.
Not exactly. After soft reset there is another new transaction automatically opened by charger. EVCC can automatically close it because transaction id is known, but it can be denied at the OCPP level by rejecting StartTransactionReq (this is message sent by charge point that is starting to charge something). 

> In other words: I'm more and more unsure what this issue really is about. Shall we close and reopen one for specific 1p startup problem?
I thought that these two issues were related that's why I opened one ticket for both of them.  Ongoing transaction issues has been pinpointed and deserves its own PR, but I don't think there is need yet to open PR for phase related issues.

tarelda, 2022-11-03:
> Feel free to open PR. I'm unclear that the expectation is that should do (except for the reset).

It is very very rough draft. Not PR ready yet. Main goal is to clean up startup process and make it as close as possible to specification.

andig, 2022-11-03:
> Yes, but I want to get logs for scenario that requires certain amount of energy produced from the PV installation.

It doesn't, you can repro with "now" as suggested above. Use `minpv` before restarting, `now` after.

andig, 2022-11-03:
Let's close this one and please open one issue per specific problem- it has become very hard for me to follow with the different topics mixed here.

andig, 2022-11-04:
Looking forward to follow-up issue for 1p3p startup!

closed 2022-11-04 by andig

tarelda, 2022-11-06:
> Looking forward to follow-up issue for 1p3p startup!

You were right. After three to five cycles it indeed calls scale3p. Not worth fixing.