evcc-io/evcc #4987: Start evcc trotz charger Probleme / 3. API Problems

manny4566, 2022-10-30:
Hallo,

just in diesem Moment hat die API vom Easee ein Problem und somit lässt sich dort der Charger Status nicht abfragen:

```
root@evcc-srv:~# evcc -l debug charger
[main  ] INFO 2022/10/30 10:32:44 evcc 0.105.2
[main  ] INFO 2022/10/30 10:32:44 using config file: /etc/evcc.yaml
[main  ] FATAL 2022/10/30 10:32:54 cannot create charger 'wallbox': cannot create charger 'template': cannot create charger 'easee': Get "https://api.easee.cloud/api/chargers/XXXX/site": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
```

Easee ist das Problem bekannt und eine Störung ist bereits veröffentlich worden.
Was mich aber jetzt stört:
Ich habe evcc neugestartet als mir noch nicht bekannt war, dass es auf der API Seite vom Easee ein Problem gibt.
Das führte aber dazu, dass evcc den charger natürlich nicht anlegen konnte und somit das Programm  im Fehlermodus stehen geblieben ist und somit nutzlos ist. 
Es ist abhängig von einer 3. Seite und somit selbst für das Auslesen der PV, Hausverbrach oder Grid Status nutzlos in dem Moment solange ein Teil, in diesem Fall der Charger, nicht ausgelesen / erstellt werden kann. Eventuell gibt es auch das gleiche Problem, wenn ein anderer Teil z.b. das Grid Gerät nicht ausgelesen werden kann.

Webseite:
```
Fehler beim Starten
cannot create charger 'wallbox'
cannot create charger 'template'
cannot create charger 'easee'
Get "https://api.easee.cloud/api/chargers/EHMYY58Y/site"
net/http: request canceled (Client.Timeout exceeded while awaiting headers)

Bitte überprüfe deine Konfigurationsdatei. Sollte dir die Fehlermeldung nicht weiterhelfen, suche in unseren [GitHub Discussions](https://github.com/evcc-io/evcc/discussions) nach einer Lösung.

Hinweis: Ein weiterer Grund, warum du diese Meldung siehst, könnte ein fehlerhaftes Gerät (Wechselrichter, Zähler, ...) sein. Überprüfe deine Netzwerkverbindungen.
```

Feature Request:
Bitte eine Möglichkeit schaffen, dass evcc trotzdem startet und weiterhin soweit nutzbar ist, wenn einzelne Teile ein Problem haben bzw. Geräte oder 3. APIs nicht erreichbar sind.

P.s. Ein auskommentieren von dem Charger oder Loadpoint Bereich funktioniert nicht, da beides zwingend notwendig ist für den Start des Programms.

Danke

naltatis, 2022-10-30:
Ja, aktuell müssen alle konfigurierten Geräte beim Start erreichbar sein. evcc startet aber regelmäßig neu bis alle Teile wieder verfügbar sind. Heißt, wenn du eine gültige Konfiguration hattest und durch bspw. Stromausfall ein evcc Neustart getriggert wird, wäre es unproblematisch und evcc würde korrekt hochkommen sobald alle anderen Komponenten erreichbar sind.

Vermutlich wird sich das Startverhalten sowieso noch ändern, wenn wir eine Konfigurierbarkeit über Web UI haben. Eine kurzfristige Zwischenlösung sehe ich hier aber gerade nicht.

@andig korrigier mich gerne wenn du hier eine einfache Lösung siehst.

andig, 2022-10-30:
evcc ist zum laden da. Ohne Charger geht das nicht.

manny4566, 2022-10-30:
> evcc ist zum laden da. Ohne Charger geht das nicht.

Schon klar, trotzdem wäre es interessant die anderen Werte weiterhin nutzbar zu halten durch einen Programmstart und sich nicht von einer 3. Partei/Api und derer Verfügbarkeit so abhängig zu machen, finde ich.

andig, 2022-10-30:
Ja, ist aber nicht der Zweck von evcc. Viel wichtiger ist es vllt. durch Neustart wieder in den Genuss von funktionierendem Laden zu kommen.

closed 2022-10-30 by andig

tyson-112, 2022-12-19:
Hallo, mein Name ist Markus und ich bin neu hier.
Hatte mir die Tage das evcc für meine Konfiguration (WR KOSTAL Plenticore 10, KSEM, BYD HVS 7,7, Easee WB und seit heute den Tibber Pulse) erstellt. Das Programm funzte auch sehr gut, wollte heute etwas an der config spielen und leider geht es mir wie dem Themenersteller, es scheitert beim Starten an der Easee, welche kaum erreichbar ist und mir laufend, auch in der App Fehler gemeldet werden.
Weiß jemand, ob Easee in der Problematik schon einen Schritt weiter ist? Ich habe auch schon den Eisee-Support angeschrieben, auch weil sich die WB über die App nicht mehr neu Starten lässt...
Über eine kurze Info, sollte es seitens Easee etwas neues geben wäre ich echt dankbar.
Gruß
Markus

tyson-112, 2022-12-19:
Selbst wenn das Fahrzeug lädt, so wie jetzt gerade, ist die WB über die App nicht erreichbar....