evcc-io/evcc #4989: SE: wrong register encoding

andig, 2022-10-30:
Ah, got it. The id is fixed in the template. We might improve this in the future! 

Looks like the `float32s` is wrong in the template, too. /cc @premultiply

_Originally posted by @andig in https://github.com/evcc-io/evcc/discussions/4871#discussioncomment-4012854_

premultiply, 2022-10-30:
`float32s` ist hier eigentlich korrekt.

andig, 2022-10-30:
Dann ists kein NaN. Kann also egtl nicht sein.

premultiply, 2022-10-30:
Ist Inf.
Das Bitdecoding erfolgt ja zuerst.

andig, 2022-10-30:
Mag sein. In meinem Test wars ein NaN, aber NUR mit umgedrehtem encoding. Dann haben wir das Thema weiter nicht verstanden.

premultiply, 2022-10-30:
Mist. Es ist wohl weder noch:

Schauen wir nochmal mal auf die Fakten. Auf der Modbusebene sieht es so aus:
```
battery2
--------
[modbus] TRACE 2022/10/21 18:51:36 modbus: send 00 0c 00 00 00 06 02 03 f5 74 00 02
[modbus] TRACE 2022/10/21 18:51:36 modbus: recv 00 0c 00 00 00 07 02 03 04 ff ff ff 7f
[modbus] TRACE 2022/10/21 18:51:36 modbus: send 00 0d 00 00 00 06 02 03 f5 84 00 02
[modbus] TRACE 2022/10/21 18:51:37 modbus: recv 00 0d 00 00 00 07 02 03 04 cc cd 40 9c
Power: 340282346638528859811704183484516925440W
SoC:   5%
```

"Swapped word order" stimmt. Denn so sind auch definitiv die korrekten Werte codiert. Sieht man hier auch beim SoC.

Wenn `ff ff ff 7f` man das als "**swapped** word order" decodiert wird es zu `0xff7fffff`.

Das ist aber nicht -Inf sondern "-Inf + 1", sprich die kleinste mit single precision float (float32) darstellbare negative Zahl:

[image]
https://www.h-schmidt.net/FloatConverter/IEEE754.html

andig, 2022-10-30:
Und jetzt nochmal ohne swapped- dann ists ein schönes NaN. Nur Zufall?

premultiply, 2022-10-30:
Ich befürchte ja, denn es wäre es ja schon sehr seltsam dass der Wert ganz normal als float32s decodiert werden kann.
Denn andernfalls müsste ja im normalbetrieb da auch völliger Unsinn bei rauskommen.
Und auch alle anderen Werte sind bei SE definitiv float32s codiert.

Könnte daher auch einfach ein Bug in der SE-Software sein...
Man hat wohl NaN gemeint aber dabei den Wordswap vergessen.

andig, 2022-10-30:
> Man hat wohl NaN gemeint aber dabei den Wordswap vergessen.

Das hab ich auch schon vermutet. Jetzt ist guter Rat teuer. Ticket beim Hersteller?

closed 2022-10-31 by andig

premultiply, 2023-02-27:
@andig Potenzieller Quickfix für den missratenen SE-NaN-SoC im Standby:
Batterie-SoC intern immer auf Range 0-100% limitieren?
Ref #6477

andig, 2023-02-27:
Im verlinkten Post gings ja um Leistung, da nutzt das also nix. Das Problem muss einfach beim Hersteller gelöst werden, kann ja nicht so schwer sein.

Weitere Option: wir bauen ein neues min/max/ignoreerror Plugin mit dem sich Randbedingungen definieren und Fehler ignorieren lassen. Das würde auch bei https://github.com/evcc-io/evcc/issues/6450 helfen und wäre nicht schwierig. Sollen wir das so ins `calc` mit einbauen?

premultiply, 2023-02-27:
Das stimmt. Hab es irgendwie zu zu später Stunde mit SoC verwechselt.