evcc-io/evcc #4997: E3DC Wallbox: Charged energy resetting

pauxus, 2022-10-31:
### Describe the bug

Sometimes, add non discernible intervals, the charged energy reported by my e3dc flex wallbox (not connected to e3dc system) seems to reset to 0.
This results in:

- The UI showing a negative value for charged energy
- influxdb entries for charged energy switching to negative values
- The view "Ladevorgänge" not showing the relevant entries

[image]

Matching entries in log (no debug):

```
Oct 30 14:45:36 homedroid evcc[2402760]: [lp-1  ] INFO 2022/10/30 14:45:36 vehicle updated: unknown -> Multivan
Oct 30 14:55:53 homedroid evcc[2402760]: [lp-1  ] INFO 2022/10/30 14:55:53 stop charging <-
Oct 31 08:57:03 homedroid evcc[2402760]: [lp-1  ] INFO 2022/10/31 08:57:03 start charging ->
Oct 31 09:14:23 homedroid evcc[2402760]: [lp-1  ] INFO 2022/10/31 09:14:23 stop charging <-
Oct 31 09:19:23 homedroid evcc[2402760]: [lp-1  ] INFO 2022/10/31 09:19:23 start charging ->
Oct 31 11:00:03 homedroid evcc[2402760]: [lp-1  ] INFO 2022/10/31 11:00:03 stop charging <-
```

In the time between start charging at 08:57 and the finish at 11:00 the vehicle (10kWh) was charged up to 85% (targeted 80%).

The influxdb entries for chargedEnergy in the period: 

```
2022-10-31T09:25:03Z | 7969.000000000001
2022-10-31T09:25:23Z | 7987.999999999993
2022-10-31T09:25:43Z | 8006.999999999998
2022-10-31T09:26:03Z | -57509
2022-10-31T09:26:23Z | -57491
2022-10-31T09:26:43Z | -57474.00000000001
2022-10-31T09:27:03Z | -57457
2022-10-31T09:27:23Z | -57440.00000000001
```

For the first load point this happened:
- 10-05
- 10-31

For the second:
- 09-14
- 10-10

Since both occurrences seem to be evenly spaced and the drop always ends in somewhat between -57 and -63 kWh I think there is some kind of overflow or reset in the wallbox. 

Of course this only seems to happen if the overflow actually occurs while charging, do it might happen more often...

`evcc charger`

returns:

```
wallbox5
--------
Power:          0W
Energy:         1.4kWh
Current L1..L3: 0A 0A 0A
Charge status:  B
Enabled:        false
Duration:       1h40m38s
Diagnostic dump:
Firmware:	SL-01.04.23
```

With the last positive value of 8000 (influx) + 1400 (lvcc charger) = 9400, which seems to be exactly the amount charged.

So IMHO evcc should have some sanity check to handle a reset of the energy values reported by the wallbox.

I now have increased the log level and try to gather more information

### Steps to reproduce

None

### Configuration details

network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error
  # vehicle: debug
  lp-1: debug

interval: 10s # control cycle interval

meters:
- type: template
  template: e3dc
  usage: grid
  host: 10.0.70.233
  port: 502
  name: grid1
- type: template
  template: e3dc
  usage: pv
  host: 10.0.70.233
  port: 502
  name: pv2
- type: template
  template: e3dc
  usage: battery
  host: 10.0.70.233
  port: 502
  name: battery3

chargers:
- type: template
  template: wallbe-meter
  host: 10.0.70.235
  port: 502
  name: wallbox5
- type: template
  template: wallbe-meter
  host: 10.0.70.234
  port: 502
  name: wallbox6

vehicles:
- name: opel
  type: template
  template: opel
  title: e-Corsa
  user: xxx
  password: xxx
  capacity: 46
  targetSoC: 80
- name: multivan
  type: template
  template: vw
  phases: 1
  title: Multivan
  user: xxx
  password: xxx
  capacity: 10.4
  targetSoC: 80

loadpoints:
- title: Garage
  charger: wallbox5
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  soc:
    estimate: true
- title: Haustür
  charger: wallbox6
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  soc:
    estimate: true

site:
  title: BiBo
  meters:
    grid: grid1
    pvs:
    - pv2
    batteries:
    - battery3
  residualPower: 100
  bufferSoC: 100
  prioritySoC: 100

influx:
  url: http://localhost:8086
  database: evcc
  token: "xxx"
  org: "xxx"

### Log details

See above

### What type of operating system are you running?

Linux

### Version

evcc version 0.105.2

andig, 2022-11-01:
Pls format config

andig, 2022-11-01:
> So IMHO evcc should have some sanity check to handle a reset of the energy values reported by the wallbox.

e3dc? You mean Wallbe? I'd suggest contacting Wallbe for a fix.

premultiply, 2022-11-02:
Can you may please show the meter configuration from the wallbox web-ui?

At first you should install an up-to-date firmware from Wallbe/Compleo.

pauxus, 2022-11-02:
[image]

Problem with updating the firmware is that:

- the old wallbe firmware page is down since they were acquired by Compleo (what is the current firmware? I have SL-01.04.23)
- I only find conflicting information whether the e3dc wallbox needs/uses a customized firmware
- I am somewhat reluctant, since this might void my warranty