evcc-io/evcc #5002: RCT: segfault on inverter standby

premultiply, 2022-10-31:
Habe eben noch einmal getestet ohne irgendetwas anzufassen... beendet sich im laufenden Betrieb... sobald der zweite Wechselrichter nicht mehr erreichbar ist.

Die Anzeige der Web-UI wird eingefroren... dann lässt sich die Wallbox auch nicht mehr steuern.

Anbei der Log:
Oct 31 17:32:31 raspberrypi evcc[666]: [site  ] ERROR 2022/10/31 17:32:31 pv meter 1: dial tcp 192.168.178.87:8899: connect: no route to host
Oct 31 17:32:32 raspberrypi evcc[666]: panic: runtime error: invalid memory address or nil pointer dereference
Oct 31 17:32:32 raspberrypi evcc[666]: [signal SIGSEGV: segmentation violation code=0x1 addr=0x2c pc=0xed1580]
Oct 31 17:32:32 raspberrypi evcc[666]: goroutine 109 [running]:
Oct 31 17:32:32 raspberrypi evcc[666]: github.com/mlnoga/rct.(*Connection).Send(0x4afc440, 0x46d8918)
Oct 31 17:32:32 raspberrypi evcc[666]:         /home/runner/go/pkg/mod/github.com/mlnoga/rct@v0.1.2-0.20220320164346-9f2daa4d6734/connection.go:61 +0x38
Oct 31 17:32:32 raspberrypi evcc[666]: github.com/mlnoga/rct.(*Connection).Query(0x4afc440, 0xb5317b78)
Oct 31 17:32:32 raspberrypi evcc[666]:         /home/runner/go/pkg/mod/github.com/mlnoga/rct@v0.1.2-0.20220320164346-9f2daa4d6734/connection.go:100 +0x80
Oct 31 17:32:32 raspberrypi evcc[666]: github.com/mlnoga/rct.(*Connection).QueryFloat32(0x4afc440, 0xb5317b78)
Oct 31 17:32:32 raspberrypi evcc[666]:         /home/runner/go/pkg/mod/github.com/mlnoga/rct@v0.1.2-0.20220320164346-9f2daa4d6734/connection.go:119 +0x24
Oct 31 17:32:32 raspberrypi evcc[666]: github.com/evcc-io/evcc/meter.(*RCT).CurrentPower(0x47089d0)
Oct 31 17:32:32 raspberrypi evcc[666]:         /home/runner/work/evcc/evcc/meter/rct.go:110 +0x50
Oct 31 17:32:32 raspberrypi evcc[666]: github.com/evcc-io/evcc/core.(*Site).updateMeter.func1()
Oct 31 17:32:32 raspberrypi evcc[666]:         /home/runner/work/evcc/evcc/core/site.go:311 +0x2c
Oct 31 17:32:32 raspberrypi evcc[666]: github.com/avast/retry-go/v3.Do(0x4a954b0, {0x296df68, 0x2, 0x2})
Oct 31 17:32:32 raspberrypi evcc[666]:         /home/runner/go/pkg/mod/github.com/avast/retry-go/v3@v3.1.1/retry.go:107 +0x12c
Oct 31 17:32:32 raspberrypi evcc[666]: github.com/evcc-io/evcc/core.(*Site).updateMeters(0x4d48540)
Oct 31 17:32:32 raspberrypi evcc[666]:         /home/runner/work/evcc/evcc/core/site.go:345 +0x76c
Oct 31 17:32:32 raspberrypi evcc[666]: github.com/evcc-io/evcc/core.(*Site).sitePower(0x4d48540, 0x0)
Oct 31 17:32:32 raspberrypi evcc[666]:         /home/runner/work/evcc/evcc/core/site.go:410 +0x4c
Oct 31 17:32:32 raspberrypi evcc[666]: github.com/evcc-io/evcc/core.(*Site).update(0x4d48540, {0x1be14d4, 0x481c240})
Oct 31 17:32:32 raspberrypi evcc[666]:         /home/runner/work/evcc/evcc/core/site.go:485 +0x124
Oct 31 17:32:32 raspberrypi evcc[666]: github.com/evcc-io/evcc/core.(*Site).Run(0x4d48540, 0x46bc440, 0x2540be400)
Oct 31 17:32:32 raspberrypi evcc[666]:         /home/runner/work/evcc/evcc/core/site.go:573 +0x230
Oct 31 17:32:32 raspberrypi evcc[666]: github.com/evcc-io/evcc/cmd.runRoot.func2()
Oct 31 17:32:32 raspberrypi evcc[666]:         /home/runner/work/evcc/evcc/cmd/root.go:222 +0x3c
Oct 31 17:32:32 raspberrypi evcc[666]: created by github.com/evcc-io/evcc/cmd.runRoot
Oct 31 17:32:32 raspberrypi evcc[666]:         /home/runner/work/evcc/evcc/cmd/root.go:221 +0x1240
Oct 31 17:32:32 raspberrypi systemd[1]: evcc.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Oct 31 17:32:32 raspberrypi systemd[1]: evcc.service: Failed with result 'exit-code'.
Oct 31 17:32:32 raspberrypi systemd[1]: evcc.service: Consumed 26.370s CPU time.
Oct 31 17:32:42 raspberrypi systemd[1]: evcc.service: Scheduled restart job, restart counter is at 2.
Oct 31 17:32:42 raspberrypi systemd[1]: Stopped evcc.
Oct 31 17:32:42 raspberrypi systemd[1]: evcc.service: Consumed 26.370s CPU time.
Oct 31 17:32:42 raspberrypi systemd[1]: Started evcc.
Oct 31 17:32:43 raspberrypi evcc[996]: [main  ] INFO 2022/10/31 17:32:43 evcc 0.105.2 (2c4d1755)
Oct 31 17:32:43 raspberrypi evcc[996]: [main  ] INFO 2022/10/31 17:32:43 using config file: /etc/evcc.yaml
Oct 31 17:32:43 raspberrypi evcc[996]: [main  ] INFO 2022/10/31 17:32:43 listening at :7070
Oct 31 17:32:46 raspberrypi evcc[996]: [main  ] FATAL 2022/10/31 17:32:46 cannot create meter 'grid2': cannot create meter 'template': cannot create meter 'rct': dial tcp 192.168.178.87:8899: connect: no>
Oct 31 17:32:46 raspberrypi evcc[996]: [main  ] FATAL 2022/10/31 17:32:46 will attempt restart in: 5m0s
Oct 31 17:33:06 raspberrypi evcc[996]: [main  ] FATAL 2022/10/31 17:33:06 evcc was stopped. OS should restart the service. Or restart manually.
Oct 31 17:33:06 raspberrypi systemd[1]: evcc.service: Main process exited, code=exited, status=1/FAILURE
Oct 31 17:33:06 raspberrypi systemd[1]: evcc.service: Failed with result 'exit-code'.
Oct 31 17:33:16 raspberrypi systemd[1]: evcc.service: Scheduled restart job, restart counter is at 3.
Oct 31 17:33:16 raspberrypi systemd[1]: Stopped evcc.
Oct 31 17:33:16 raspberrypi systemd[1]: Started evcc.
Oct 31 17:33:16 raspberrypi evcc[1008]: [main  ] INFO 2022/10/31 17:33:16 evcc 0.105.2 (2c4d1755)
Oct 31 17:33:16 raspberrypi evcc[1008]: [main  ] INFO 2022/10/31 17:33:16 using config file: /etc/evcc.yaml
Oct 31 17:33:16 raspberrypi evcc[1008]: [main  ] INFO 2022/10/31 17:33:16 listening at :7070
Oct 31 17:33:20 raspberrypi evcc[1008]: [main  ] FATAL 2022/10/31 17:33:20 cannot create meter 'grid2': cannot create meter 'template': cannot create meter 'rct': dial tcp 192.168.178.87:8899: connect: n>
Oct 31 17:33:20 raspberrypi evcc[1008]: [main  ] FATAL 2022/10/31 17:33:20 will attempt restart in: 5m0s

_Originally posted by @orbus64 in https://github.com/evcc-io/evcc/discussions/4994#discussioncomment-4020660_

/cc @andig @mlnoga

andig, 2022-10-31:
Send on closed connection

andig, 2022-11-01:
Closed in https://github.com/evcc-io/evcc/commit/aff876a2efd7f47fb1c289c8d47306306d8e964b

closed 2022-11-01 by andig