evcc-io/evcc #5012: Fehlerhafte Anzeige Ladevorgänge in der UI

bicaluv, 2022-11-01:
### Describe the bug

Vergleiche ich die Ladesitzungen, wie sie z.B. mittels Telegram aufgegeben werden mit denen die in der UI unter "Ladevorgänge" angezeigt werden, stimmen diese nicht überein. Es fehlen Einträge von Ladevorgängen, die stattgefunden haben.
Auszug Telegram:
[image]

Auszug aus der Evcc.db:
```
30|2022-10-29 10:40:31.718102292+02:00|2022-10-29 17:48:31.816687726+02:00|Carport||CITIGOe iV|0.0|0.0|0.0
31|2022-10-30 09:32:55.647327413+01:00|2022-10-30 11:48:46.20020886+01:00|Carport||CITIGOe iV|0.0|0.0|3.45536111111111
32|2022-10-31 09:27:00.386825929+01:00|2022-10-31 11:12:30.796909515+01:00|Carport||CITIGOe iV|0.0|0.0|0.0
```
Auszug UI:
[image]

Was hierbei auffällt: Es gibt in der Evcc.db einige Einträge, die nicht in der UI angezeigt werden, wohl weil die Lademenge (`charged_kwh`) 0 ist.

### Steps to reproduce

Bin nicht sicher, warum charged_kwh auf 0 steht.

### Configuration details

```
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070
  
interval: 10s # control cycle interval

# log settings
log: info
levels:
  site: info
  lp-1: info
  lp-2: info
  cache: error
  db: error

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for examples see https://github.com/andig/evcc-config#meters
meters:
- type: template
  template: sma-home-manager
  usage: grid
  host: 192.168.2.240
  name: sma-grid
- type: template
  template: sma-inverter
  usage: pv
  host: 192.168.2.241
  password: 1234  # Wird nur benötigt, wenn das Standardpasswort geändert wurde
  name: sma-pv

hems:
  type: shm
  allowcontrol: false

# charger definitions
# name can be freely chosen and is used as reference when assigning charger to vehicle
# for examples see https://github.com/andig/evcc-config#chargers
chargers:
- type: template
  template: go-e
  host: goeCharger
  name: go-eCharger

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for examples see https://github.com/andig/evcc-config#vehicles
vehicles:
- type: skoda
  title: CITIGOe iV
  user: 1234
  password: 1234
  vin: 1234
  capacity: 36
  name: Citigo
  phases: 2
  onIdentify: # set defaults when vehicle is identified
    mode: pv # enable PV-charging when vehicle is identified
    minSoC: 30 # charge to at least 30% independent of charge mode
    targetSoC: 80 # limit charge to 80%

# site describes the EVU connection, PV and home battery
site:
  title: Haus # display name for UI
  meters:
    grid: sma-grid # grid meter
    pv: sma-pv # pv meter

# loadpoint describes the charger, charge meter and connected vehicle
loadpoints:
- title: Carport # display name for UI
  charger: go-eCharger # charger
  mode: "off"
  resetOnDisconnect: true # set defaults when vehicle disconnects
  soc:
    # polling defines usage of the vehicle APIs
    # Modifying the default settings it NOT recommended. It MAY deplete your vehicle's battery
    # or lead to vehicle manufacturer banning you from API use. USE AT YOUR OWN RISK.
    poll:
      # poll mode defines under which condition the vehicle API is called:
      #   charging: update vehicle ONLY when charging (this is the recommended default)
      #   connected: update vehicle when connected (not only charging), interval defines how often
      #   always: always update vehicle regardless of connection state, interval defines how often
      mode: connected
      # poll interval defines how often the vehicle API may be polled if NOT charging
      interval: 60m
    # min: 30 # immediately charge to 30% regardless of mode unless "off" (disabled)
    # target: 80 # always charge to 80%
    estimate: true # set false to disable interpolating between api updates (not recommended)
  phases: 3 # ev phases (default 3)
  enable: # pv mode enable behavior
    delay: 1m # threshold must be exceeded for this long
    threshold: 0 # minimum export power (W). If zero, export must exceed minimum charge power to enable
  disable: # pv mode disable behavior
    delay: 3m # threshold must be exceeded for this long
    threshold: 200 # maximum import power (W)
  guardduration: 5m # switch charger contactor not more often than this (default 10m)
  mincurrent: 6 # minimum charge current (default 6A)
  maxcurrent: 16 # maximum charge current (default 16A)

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.2826 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.0944 # [currency]/kWh

    
# push messages
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
    soc: # vehicle soc update event
      title: SoC updated
      msg: Battery charged to ${vehicleSoC:%.0f}%
  services:
    - type: telegram
      token: 123:1234 # bot id
      chats:
      - 123 # list of chat ids

```

### Log details

Will be added upon request

### What type of operating system are you running?

Linux

### Version

evcc version 0.105.2 + nightly (31.10.22 master)

andig, 2022-11-01:
> Vergleiche ich die Ladesitzungen, wie sie z.B. mittels Telegram aufgegeben werden mit denen die in der UI unter "Ladevorgänge" angezeigt werden, stimmen diese nicht überein.

Müssen sie auch nicht. Telegram zeigt start/stop, Datenbank zeigt connect/disconnect. Bitte konkretes Problem.

bicaluv, 2022-11-01:
Ach ja, die Aussage "Datenbank zeigt connect/disconnect." stimmt so nicht. Sie zeigt die Startzeit des ersten Ladens am Tag und das Ende des letzten Ladens am Tag, sofern das Fahrzeug immer verbunden war. nicht wann es angeschlossen wurde.

andig, 2022-11-01:
Und dennoch beenden wir den Ladevorgang im Log erst bei Disconnect. Daher muss es am 31.10. auch keinen geben wenn du am 29. an- und am 1. abgesteckt hast. Ohne Logfile ist die Sache müßig.

bicaluv, 2022-11-01:
Auto wurde am 31.10. um 11:24 abgesteckt.   
aber ja, ich besorge das Log, hast Recht.

bicaluv, 2022-11-02:
So, das sind die Log-Einträge die ich zum 31.10. noch gefunden habe. Das Auto wurde am 30.10. um 23:00 angesteckt und am 31.10. wurde ab 9:27 bis 11:12 geladen, da ab dann das Auto abgesteckt war.  
Und genau dieser Eintrag mit diesem Zeitraum ist zwar in der Evcc.db vorhanden, wird aber nicht angezeigt in der UI, da der Wert in der charged_kwh-Spalte 0 ist. 

 
```
Oct 31 09:14:30 raspberrypi evcc[6194]: [lp-1  ] ERROR 2022/10/31 09:14:30 charger vehicle id: Get "http://192.168.2.108/status": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
Oct 31 09:15:41 raspberrypi evcc[6194]: [lp-1  ] ERROR 2022/10/31 09:15:41 charge meter: Get "http://192.168.2.108/status": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
Oct 31 09:16:52 raspberrypi evcc[6194]: [lp-1  ] ERROR 2022/10/31 09:16:52 charger: Get "http://192.168.2.108/status": net/http: request canceled (Client.Timeout exceeded while awaiting headers)

Oct 31 09:18:38 raspberrypi evcc[6194]: [lp-1  ] ERROR 2022/10/31 09:18:38 charge rater: Get "http://192.168.2.108/status": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
Oct 31 09:18:57 raspberrypi evcc[6194]: [lp-1  ] ERROR 2022/10/31 09:18:57 charger vehicle id: Get "http://192.168.2.108/status": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
Oct 31 09:19:07 raspberrypi evcc[6194]: [lp-1  ] ERROR 2022/10/31 09:19:07 charger: Get "http://192.168.2.108/status": net/http: request canceled (Client.Timeout exceeded while awaiting headers)

Oct 31 09:27:00 raspberrypi evcc[6194]: [lp-1  ] INFO 2022/10/31 09:27:00 start charging ->
Oct 31 09:36:20 raspberrypi evcc[6194]: [lp-1  ] INFO 2022/10/31 09:36:20 stop charging <-
Oct 31 09:48:20 raspberrypi evcc[6194]: [lp-1  ] INFO 2022/10/31 09:48:20 start charging ->

Oct 31 09:53:20 raspberrypi evcc[6194]: [lp-1  ] INFO 2022/10/31 09:53:20 stop charging <-
Oct 31 10:02:01 raspberrypi CRON[4254]: (pi) CMD (python3 /home/pi/bin/check_fork.py >> /home/pi/etc/check_fork.log)
Oct 31 10:10:00 raspberrypi evcc[6194]: [lp-1  ] INFO 2022/10/31 10:10:00 start charging ->

Oct 31 10:54:20 raspberrypi evcc[6194]: [lp-1  ] INFO 2022/10/31 10:54:20 stop charging <-

Oct 31 11:12:30 raspberrypi evcc[6194]: [lp-1  ] INFO 2022/10/31 11:12:30 car disconnected
Oct 31 11:12:30 raspberrypi evcc[6194]: [lp-1  ] INFO 2022/10/31 11:12:30 vehicle updated: CITIGOe iV -> unknown

Oct 31 13:26:43 raspberrypi systemd[1]: Stopping evcc...
Oct 31 13:26:44 raspberrypi systemd[1]: evcc.service: Succeeded.
Oct 31 13:26:44 raspberrypi systemd[1]: Stopped evcc.
Oct 31 13:26:44 raspberrypi systemd[1]: Started evcc.
Oct 31 13:26:44 raspberrypi evcc[8680]: [main  ] INFO 2022/10/31 13:26:44 evcc 0.105.2_mm (5b140f40)
Oct 31 13:26:44 raspberrypi evcc[8680]: [main  ] INFO 2022/10/31 13:26:44 using config file: /home/pi/etc/evcc.yaml
Oct 31 13:26:44 raspberrypi evcc[8680]: [main  ] INFO 2022/10/31 13:26:44 listening at :7070

Oct 31 13:26:50 raspberrypi evcc[8680]: [semp  ] WARN 2022/10/31 13:26:50 SEMP_BASE_URL unspecified, using http://192.168.2.30:7070 instead
Oct 31 13:26:51 raspberrypi evcc[8680]: [site  ] INFO 2022/10/31 13:26:51 site config:
Oct 31 13:26:51 raspberrypi evcc[8680]: [site  ] INFO 2022/10/31 13:26:51   meters:      grid ✓ pv ✓ battery ✗
Oct 31 13:26:51 raspberrypi evcc[8680]: [site  ] INFO 2022/10/31 13:26:51     grid:      power ✓ energy ✓ currents ✓
Oct 31 13:26:51 raspberrypi evcc[8680]: [site  ] INFO 2022/10/31 13:26:51     pv 1:      power ✓ energy ✓ currents ✓
Oct 31 13:26:51 raspberrypi evcc[8680]: [site  ] INFO 2022/10/31 13:26:51   vehicles:
Oct 31 13:26:51 raspberrypi evcc[8680]: [site  ] INFO 2022/10/31 13:26:51     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
Oct 31 13:26:51 raspberrypi evcc[8680]: [lp-1  ] INFO 2022/10/31 13:26:51 loadpoint 1:
Oct 31 13:26:51 raspberrypi evcc[8680]: [lp-1  ] INFO 2022/10/31 13:26:51   mode:        off
Oct 31 13:26:51 raspberrypi evcc[8680]: [lp-1  ] INFO 2022/10/31 13:26:51   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
Oct 31 13:26:51 raspberrypi evcc[8680]: [lp-1  ] INFO 2022/10/31 13:26:51   meters:      charge ✓
Oct 31 13:26:51 raspberrypi evcc[8680]: [lp-1  ] INFO 2022/10/31 13:26:51     charge:    power ✓ energy ✗ currents ✓
Oct 31 13:26:52 raspberrypi evcc[8680]: [lp-1  ] INFO 2022/10/31 13:26:52 car disconnected
```

andig, 2022-11-02:
Was soll ich daraus lesen: Du hast einen Ladevorgang ohne Meßwerte weil der Zähler nicht da ist. Wundert mich nicht, wenn das als 0 in der DB landet.

andig, 2022-11-02:
Mach doch mal bitte debug log und db auf trace, dann haben wir auch eine Chance zu sehen was da passiert. Ich schließe hier bis aussagekräftiges Log vorliegt.

bicaluv, 2022-11-02:
Ok, Log ist entsprechend aktiviert und ich melde mich bei bedarf.  
Dass die Ladevorgänge ohne Messwerte waren verstehe ich nicht, denn im Telegram wurden diese ja ausgegeben. Siehe Screenshot 1. Post.

andig, 2022-11-02:
Relevant wäre auch was im Log zu `chargedEnergy` steht- das bräuchte aber `cache: trace` und wird sehr viel.

bicaluv, 2022-11-03:
Anbei der Trace von heute:  
```
Nov  2 22:08:40 raspberrypi evcc[5884]: [lp-1  ] INFO 2022/11/02 22:08:40 car connected
Nov  2 22:08:40 raspberrypi evcc[5884]: [push  ] DEBUG 2022/11/02 22:08:40 telegram: sending to xxxx
Nov  2 22:08:40 raspberrypi evcc[5884]: [main  ] DEBUG 2022/11/02 22:08:40 vehicle status: B (CITIGOe iV)
Nov  2 22:08:40 raspberrypi evcc[5884]: [lp-1  ] INFO 2022/11/02 22:08:40 vehicle updated: unknown -> CITIGOe iV
Nov  2 22:08:41 raspberrypi evcc[5884]: [push  ] DEBUG 2022/11/02 22:08:41 telegram: sending to xxxx

Nov  2 23:18:40 raspberrypi evcc[5884]: [lp-1  ] INFO 2022/11/02 23:18:40 start charging ->
Nov  2 23:18:40 raspberrypi evcc[5884]: [db    ] TRACE 2022/11/02 23:18:40 INSERT INTO `sessions` (`created`,`finished`,`loadpoint`,`identifier`,`vehicle`,`meter_start_kwh`,`meter_end_kwh`,`charged_kwh`) VALUES ("2022-11-02 23:18:40.056","0000-00-00 00:00:00","Carport","","CITIGOe iV",0.000000,0.000000,0.000000) RETURNING `id` 1 <nil>
Nov  2 23:18:40 raspberrypi evcc[5884]: [push  ] DEBUG 2022/11/02 23:18:40 telegram: sending to xxxx

Nov 03 02:26:00 raspberrypi evcc[5884]: [push  ] DEBUG 2022/11/03 02:26:00 telegram: sending to xxxx
Nov 03 02:26:10 raspberrypi evcc[5884]: [lp-1  ] INFO 2022/11/03 02:26:10 stop charging <-
Nov 03 02:26:10 raspberrypi evcc[5884]: [db    ] TRACE 2022/11/03 02:26:10 UPDATE `sessions` SET `created`="2022-11-02 23:18:40.056",`finished`="2022-11-03 02:26:10.065",`loadpoint`="Carport",`identifier`="",`vehicle`="CITIGOe iV",`meter_start_kwh`=0.000000,`meter_end_kwh`=0.000000,`charged_kwh`=16.961678 WHERE `id` = 33 1 <nil>
Nov 03 02:26:10 raspberrypi evcc[5884]: [push  ] DEBUG 2022/11/03 02:26:10 telegram: sending to xxxx
Nov 03 08:05:30 raspberrypi evcc[5884]: [db    ] TRACE 2022/11/03 08:05:30 SELECT * FROM `sessions` WHERE charged_kwh>=0.05 ORDER BY created desc 14 <nil>
Nov 03 08:48:50 raspberrypi evcc[5884]: [lp-1  ] INFO 2022/11/03 08:48:50 car disconnected
Nov 03 08:48:50 raspberrypi evcc[5884]: [db    ] TRACE 2022/11/03 08:48:50 UPDATE `sessions` SET `created`="2022-11-02 23:18:40.056",`finished`="2022-11-03 08:48:50.031",`loadpoint`="Carport",`identifier`="",`vehicle`="CITIGOe iV",`meter_start_kwh`=0.000000,`meter_end_kwh`=0.000000,`charged_kwh`=0.000000 WHERE `id` = 33 1 <nil>
Nov 03 08:48:50 raspberrypi evcc[5884]: [lp-1  ] INFO 2022/11/03 08:48:50 vehicle updated: CITIGOe iV -> unknown
Nov 03 08:48:50 raspberrypi evcc[5884]: [push  ] DEBUG 2022/11/03 08:48:50 telegram: sending to xxxx
Nov 03 08:56:03 raspberrypi evcc[5884]: [db    ] TRACE 2022/11/03 08:56:03 SELECT * FROM `sessions` WHERE charged_kwh>=0.05 ORDER BY created desc 13 <nil>
```

Was ich interessant finde, dass der Eintrag 33 in der Evcc.db einen Wert für charged_kwh hat und dann, wenn das Auto abgesteckt wird, wird der Wert erneut mit 0 überschrieben. Hat das mit dem Übergang zu unknown zu tun?

andig, 2022-11-03:
Super, das ist eine heisse Spur! Du hast einen Go-E? Scheint als würde folgendes passieren:
- Auto wird abgezogen
- evcc geht von C -> A
- dabei wird der Ladevorgang letztmalig mit Energie und Zählerstand aktualisiert

...nur dass Energie bei Dir schon 0 ist weil die Go-E schneller war.

Das lässt sich super per Workaround lösen!

bicaluv, 2022-11-03:
Ja, ist eine Go-E.

andig, 2022-11-03:
Closed in https://github.com/evcc-io/evcc/commit/7d3619f8f7a5346beabd2c1dc1867b9bf877e06d

closed 2022-11-03 by andig

bicaluv, 2022-11-04:
Hi sandig, ich muss ich noch ein Trace zeigen, bei dem der Wert für charged_kwh noch immer nicht stimmt. Das Fahrzeug wurde 15:29 bis 17:00 im "Schnell"-Modus von 22 auf 48 % geladen. Auch im Telegram-Output steht "Finished charging 0.0kWh in 1h30m50s", falls das weiterhilft.  
Warum das evtl. beim Go-E wieder nicht weitergegeben wird, weiß ich nicht. Timing-Problem?

```
Nov  4 15:28:38 raspberrypi evcc[616]: [lp-1  ] INFO 2022/11/04 15:28:38 car connected
Nov  4 15:28:38 raspberrypi evcc[616]: [main  ] DEBUG 2022/11/04 15:28:38 vehicle status: A (CITIGOe iV)
Nov  4 15:28:48 raspberrypi evcc[616]: [main  ] DEBUG 2022/11/04 15:28:48 vehicle status: A (CITIGOe iV)
Nov  4 15:28:58 raspberrypi evcc[616]: [main  ] DEBUG 2022/11/04 15:28:58 vehicle status: A (CITIGOe iV)
Nov  4 15:29:08 raspberrypi evcc[616]: [main  ] DEBUG 2022/11/04 15:29:08 vehicle status: A (CITIGOe iV)
Nov  4 15:29:18 raspberrypi evcc[616]: [main  ] DEBUG 2022/11/04 15:29:18 vehicle status: A (CITIGOe iV)
Nov  4 15:29:28 raspberrypi evcc[616]: [main  ] DEBUG 2022/11/04 15:29:28 vehicle status: A (CITIGOe iV)
Nov  4 15:29:38 raspberrypi evcc[616]: [main  ] DEBUG 2022/11/04 15:29:38 vehicle status: B (CITIGOe iV)
Nov  4 15:29:38 raspberrypi evcc[616]: [lp-1  ] INFO 2022/11/04 15:29:38 vehicle updated: unknown -> CITIGOe iV
Nov  4 15:29:58 raspberrypi evcc[616]: [lp-1  ] INFO 2022/11/04 15:29:58 start charging ->
Nov  4 15:29:58 raspberrypi evcc[616]: [db    ] TRACE 2022/11/04 15:29:58 INSERT INTO `sessions` (`created`,`finished`,`loadpoint`,`identifier`,`vehicle`,`meter_start_kwh`,`meter_end_kwh`,`charged_kwh`) VALUES ("2022-11-04 15:29:58.197","0000-00-00 00:00:00","Carport","","CITIGOe iV",0.000000,0.000000,0.000000) RETURNING `id` 1 <nil>

Nov  4 17:00:48 raspberrypi evcc[616]: [lp-1  ] INFO 2022/11/04 17:00:48 stop charging <-
Nov  4 17:00:48 raspberrypi evcc[616]: [db    ] TRACE 2022/11/04 17:00:48 UPDATE `sessions` SET `created`="2022-11-04 15:29:58.197",`finished`="2022-11-04 17:00:48.642",`loadpoint`="Carport",`identifier`="",`vehicle`="CITIGOe iV",`meter_start_kwh`=0.000000,`meter_end_kwh`=0.000000,`charged_kwh`=0.000000 WHERE `id` = 36 1 <nil>
Nov  4 17:00:48 raspberrypi evcc[616]: [lp-1  ] INFO 2022/11/04 17:00:48 car disconnected
Nov  4 17:00:48 raspberrypi evcc[616]: [db    ] TRACE 2022/11/04 17:00:48 UPDATE `sessions` SET `created`="2022-11-04 15:29:58.197",`finished`="2022-11-04 17:00:48.691",`loadpoint`="Carport",`identifier`="",`vehicle`="CITIGOe iV",`meter_start_kwh`=0.000000,`meter_end_kwh`=0.000000,`charged_kwh`=0.000000 WHERE `id` = 36 1 <nil>
Nov  4 17:00:48 raspberrypi evcc[616]: [lp-1  ] INFO 2022/11/04 17:00:48 vehicle updated: CITIGOe iV -> unknown
Nov  4 17:01:19 raspberrypi evcc[616]: [db    ] TRACE 2022/11/04 17:01:19 SELECT * FROM `sessions` WHERE charged_kwh>=0.05 ORDER BY created desc 15 <nil>
```

andig, 2022-11-05:
Probier mal bitte 

    evcc meter

während der Ladevorgang läuft. Wenn da auch ne 0 kommt dann Ticket bei Go-E. Wenn nein bitte nochmal detailliertes Log wo ich sehen kann ab wann der Wert verschwindet als neues Issue.

Ich hab bei Dir auch keine Zählerstände gesehen- die Box hat nur das v1 Api?

bicaluv, 2022-11-05:
Echt `evcc meter` und nicht `evcc charger`?  
Die Box meldet sich mit Hardware Version v2. Bzgl. des APIs weiß ich es nicht; hat zumindest keine Phasensteuerung.  
Logs und Tests etc. erstelle ich morgen.

andig, 2022-11-05:
Sorry, charger!