evcc-io/evcc #5017: Kleine Verbesserungsvorschläge für Ansicht Ladevorgänge

blenni, 2022-11-01:
Kurz vorab, ich nutze Version 105.2, hab also nicht die allerletzten Updates für die Ansicht, falls sich da was geändert haben sollte.

2 Themen sind mir aufgefallen, die man ggf. verbessern könnte:

1. Spaltenbreite in den verschiedenen Monaten ungleich (Screenshot vom Laptop, ist in abgeschwächter Form aber auch am Handy so):
[image]

2. Wenn ich die EVCC-Oberfläche aufrufe, bleibt ja jedes mal gespeichert, welcher Ladepunkt gerade ausgewählt ist (gilt nur für das Handy, am PC sehe ich direkt beide Ladepunkte). Wenn ich die Ladevorgänge-Ansicht auswähle, bin ich danach immer auf dem ersten Ladepunkt - gibt es die Möglichkeit, dass man da auch wieder auf den zuletzt gewählten Ladepunkt kommt?

naltatis, 2022-11-08:
@blenni Schau dir mal die aktuelle Version an. Hier hat sich inzwischen ein bisschen was geändert: Start/Endzeit, rechtsbündige kWh, Spaltenbreiten und Gruppierung nach Ladepunkt. Vermutlich löst das schon die meisten angesprochenen Themen. Wenn nicht gib gerne noch mal konkret Feedback.

Thema "Daten für ausgewählten Ladepunkt". Wie du ja schon schreibst gibts auf größeren Screens das Konzept eines "aktuellen Ladepunkts" in der UI ja gar nicht. Momentan haben wir die Ladesessions ja aus der Hauptnavi (also ohne Kontext) verlinkt. Solange der Link an diesem zentralen Ort ist würde ich auch als Nutzer erwarten, dass ich alles und nicht nur ein Subset sehe. Ich kann mir aber gut vorstellen, dass wir die Ladesessions zukünftig direkt aus den Ladepunkten in der Hauptansicht heraus verlinken. Dann macht es total Sinn nicht alles sondern nur das des Ladepunktes anzuzeigen. Ich hab aktuell aber noch keine "gute" Idee wie wir die Verlinkung von da machen, ohne dass es dort nervt oder die UI unnötig überfrachtet. Die Ladesessions sind ja für viele User auch eher unwichtig. Daher würd ich zumindest aktuell kein eigenes Icon oder Textlink dafür machen. Vielleicht später mal über ein Optionsmenü (⠸) wenn wir noch mehr ähnliche Ladepunktfunktionen bekommen.

Danke für dein Feedback. Ich würd dieses Issue jetzt erstmal schließen.

closed 2022-11-08 by naltatis

blenni, 2022-11-08:
Ja, passt.

Mit der 106.3 passt das mit den Spaltenbreiten über die verschiedenen Monate / Ladepunkte bei mir aber immer noch nicht ganz:
[image]

Tut der Funktionalität ja keinen Abbruch, sieht nur noch nicht so perfekt aus wie der Rest der Oberfläche ;)