evcc-io/evcc #5020: Misleading listening log message placement

tarelda, 2022-11-02:
### Describe the bug

Log message informing that EVCC is listening on webui port is misplaced. 

Currently, it is in line 117.  
https://github.com/evcc-io/evcc/blob/09eab09415cf018d9017413bf978702503859342/cmd/root.go#L117 
But listening takes place here:
https://github.com/evcc-io/evcc/blob/09eab09415cf018d9017413bf978702503859342/cmd/root.go#L260

This is especially confusing when devices take longer to set up and you wonder why there is no webui despite log message saying that it is listening on that port. So I suggest moving message from line 117 to just before line 260 to prevent that confusion. 

### Steps to reproduce

Everytime

### Configuration details

Not relevant

### Log details

[main  ] INFO 2022/11/02 10:21:39 evcc 0.105.2 (25ec8591)
[main  ] INFO 2022/11/02 10:21:39 using config file: /etc/evcc.yaml
[main  ] INFO 2022/11/02 10:21:39 listening at :7070
[ocpp  ] DEBUG 2022/11/02 10:21:40 waiting for chargepoint: 8m0s
[main  ] FATAL 2022/11/02 10:29:40 cannot create charger 'wallbox': cannot create charger 'template': cannot create charger 'ocpp': timeout
[main  ] FATAL 2022/11/02 10:29:40 will attempt restart in: 5m0s

### What type of operating system are you running?

Docker container

andig, 2022-11-02:
It's a trade-off. If we put the message later it will be lost in the amount of other logging. We could change to "ui starting at:" to make clearer that this has not been done yet?

We can also speedup the initialisation, see https://github.com/evcc-io/evcc/pull/4830 which should especially help with multiple OCPP devices.

tarelda, 2022-11-02:
When fixing #5025 maybe change to: 
`("ui will start at: %s", uri)`

andig, 2022-11-03:
Closed in https://github.com/evcc-io/evcc/commit/3e4b50a3bcca481ddab226668bd47cfcf63d7c23

closed 2022-11-03 by andig