evcc-io/evcc #5022: Machine ID generation issue

tarelda, 2022-11-02:
### Describe the bug

I don't think machine-id generation issues should prevent evcc startup when **there is no enabled telemetry**.

I know that this is related to this two:
https://github.com/denisbrodbeck/machineid/issues/5
https://github.com/evcc-io/evcc/issues/3027

### Steps to reproduce

Use code base with #5001 that introduced telemetry as opt-in switch during runtime. 

### Configuration details

not relevant

### Log details

Version: 0.105.2 09eab094 2022-11-02_10:17:28
go build -v -tags=release -ldflags='-X github.com/evcc-io/evcc/server.Version=0.105.2 -X github.com/evcc-io/evcc/server.Commit=09eab094 -s -w'
[main  ] INFO 2022/11/02 11:17:28 evcc 0.105.2 (09eab094)
[main  ] INFO 2022/11/02 11:17:28 using config file: /etc/evcc.yaml
[main  ] INFO 2022/11/02 11:17:28 listening at :7070
[main  ] FATAL 2022/11/02 11:17:28 could not get machineid: open /etc/machine-id: no such file or directory; for manual configuration use plant: 7c6a33ec487c3c597dd2fa34ee80c6268780f570340a2f98890b9cc66f4a3819
[main  ] FATAL 2022/11/02 11:17:28 will attempt restart in: 5m0s

### What type of operating system are you running?

Docker container

andig, 2022-11-02:
> I don't think machine-id generation issues should prevent evcc startup when there is no enabled telemetry.

It depends on your other devices. Does on of them use machine id? Config?

tarelda, 2022-11-02:
@andig https://github.com/evcc-io/evcc/issues/4977#issuecomment-1295808344

andig, 2022-11-02:
Mhhm. Can you track this to a commit? 09eab09415cf018d9017413bf978702503859342?

naltatis, 2022-11-02:
The MR changed the way telemetry is initialised. With the new runtime enabling possibility the machine-id is required on startup. I see two possible solutions:

1. Check machine-id when trying to enable telemetry through UI. We'd have to throw an error and instruct the users to add plant to the config and reboot.
2. Generating an instance id (just for telemetry) if no machine id is present and storing it in settings `telemetry.instanceId`

andig, 2022-11-02:
Ich würde den PR fürs erste zurück rollen und Neues Release machen? Das ist ja breaking für Viele...

naltatis, 2022-11-02:
@andig this is the change https://github.com/evcc-io/evcc/commit/09eab09415cf018d9017413bf978702503859342#diff-68cf38ccea78ea9c869b779386f02b261f8dc67332e3f3b4b105871a1ef63a61L97

naltatis, 2022-11-02:
Ich würd lieber den Fix bauen. Das ist keine große Sache.

andig, 2022-11-02:
Wir sollten genau eine InstanceId haben, nicht mehrere. Geht der Fix kurzfristig? Sonst würde ich den PR zurück rollen.

PR #5029 (2022-11-03): Remove docker plant startup requirement
- show error when telemetry enable fails

fixes #5022

[image]
closed 2022-11-03 by andig