evcc-io/evcc #5025: Can't specify interface - "host" in section network (evcc.yaml) is ignored

Grisnir, 2022-11-02:
### Describe the bug

I try to get evcc listening on localhost (127.0.0.1) only, however it seems to ignore whatever IP address or hostname I define as host in evcc.yaml. Even if I put in fantasy data like host: fakehost.doesnt.exist, it starts up and binds to port 7070 on all interfaces.

### Steps to reproduce

Specify host IP address in evcc.yaml, restart evcc, evcc is still binding to port 7070 on all interfaces.

### Configuration details

network:
  schema: http
  host: 127.0.0.1
  port: 7070

### Log details

Nov  2 10:45:23 jarvis systemd[1]: Started evcc.
Nov  2 10:45:23 jarvis evcc[344831]: [main  ] INFO 2022/11/02 10:45:23 evcc 0.105.2
Nov  2 10:45:23 jarvis evcc[344831]: [main  ] INFO 2022/11/02 10:45:23 using config file: /etc/evcc.yaml
Nov  2 10:45:23 jarvis evcc[344831]: [main  ] INFO 2022/11/02 10:45:23 listening at :7070

netstat -ln -t
Proto Recv-Q Send-Q Local Address           Foreign Address         State
tcp6       0      0 :::7070                 :::*                    LISTEN

### What type of operating system are you running?

Linux

### Version

evcc version 0.105.2

tarelda, 2022-11-02:
Introduced by 1a4da3f7cada077410fd7a17cae44a28a2915327
https://github.com/evcc-io/evcc/blob/cc04229768c5b16f43ffb2d7c0c48706ee086a99/cmd/root.go#L128

andig, 2022-11-02:
I think we did change that because binding to localhost was too confusing for people. The current `host` is for those cases where evcc needs to announce its address on the network (e.g. SEMP or other protocols).

What is your use case for binding to specific interfaces? Do you have a suggestion how to do that now in a non-confusing, non-breaking way?

Grisnir, 2022-11-02:
My use case is that I have evcc running on a ubuntu 22.04 server with multiple interfaces, private and public facing as well as vpn. I want to bind it to one interface specifically, so that it's only accessible via my internal network. I'm aware that there are multiple solutions like iptables, firewalld, etc. to restrict access, but quite about every software can have the interface/IP specified where it should listen on.

I usually let all web applications only listen on localhost and deal with the corresponding access with an apache reverse proxy. I've got the reverse proxy for evcc up and running, accessing it via https://server.ip/evcc, so I'd like to turn off all other possibilities to access evcc via http://<any_interface>:7070/. If that's not possible, I'll need to create firewall rules.

Grisnir, 2022-11-02:
First of all thanks for your quick responses and I wanted to let you know that evcc is really awesome, I'm using it for about 1,5 months now.

> Do you have a suggestion how to do that now in a non-confusing, non-breaking way?

I'm not really a programmer and had nothing to do with golang so far. That said, I was playing around a bit and did change the line referenced above by tarelda

> https://github.com/evcc-io/evcc/blob/cc04229768c5b16f43ffb2d7c0c48706ee086a99/cmd/root.go#L128

        httpd := server.NewHTTPd(fmt.Sprintf(":%d", conf.Network.Port), socketHub)
with
        httpd := server.NewHTTPd(fmt.Sprintf("127.0.0.1:%d", conf.Network.Port), socketHub)

just to see if it works, and it did:
netstat -ln -t
Proto Recv-Q Send-Q Local Address           Foreign Address         State      
tcp        0      0 127.0.0.1:7070          0.0.0.0:*               LISTEN     

However, I have no idea if introducing an optional config option for defining the IP does break other things, that's unfortunately way above my head.

tarelda, 2022-11-02:
> Got it. It's probably possible but I don't immediately see an obvious approach:
> 
> > Do you have a suggestion how to do that now in a non-confusing, non-breaking way?
> 
> I think the requirements are:
> 
> * simple use for novice users (i.e. public)
> * make network-reachable names configurable (for inward apis like SEMP)
> * support MDNS announcements
> * don't break installations with existing config
> 
> and last but not least:
> 
> * allow restriction by interface
> 
> Happy to look at best practices from other packages.

@andig [docs are not helping either](https://docs.evcc.io/docs/reference/configuration/uri) . 

IMHO, split hostname with ui listening address.

andig, 2022-11-02:
How would the config look like?

tarelda, 2022-11-02:
```
# without mdns
network:
  schema: http
  host: tarelda.com
  addr: 127.0.0.1
  port: 80
```

```
# with mdns
network:
  schema: http
  host: tarelda.local
  addr: 127.0.0.1
  port: 80
```

Something along these lines, but that's just quick skim through source code.

andig, 2022-11-02:
Mhhm- rather `bind` than `addr`? I still find `addr` ambiguous has it wouldn‘t tell me if I should use that or `host`?

tarelda, 2022-11-02:
> Mhhm- rather `bind` than `addr`? I still find `addr` ambiguous has it wouldn‘t tell me if I should use that or `host`?

http package uses addr

andig, 2022-11-02:
The audience of the config is not Go programmers!

Grisnir, 2022-11-02:
For config examples:
InfluxDB: bind-address
sshd: listen-on
grafana: http_addr
telegraf influxdbv2 listener: 
#   ## Address and port to host HTTP listener on
#   service_address = ":8080"

As telegraf is written in golang, maybe the example for influxdb_v2_listener above is worth a look, how they achieve that:
https://github.com/influxdata/telegraf/blob/master/plugins/inputs/influxdb_v2_listener/influxdb_v2_listener.go

premultiply, 2022-11-02:
+1 for `listen` or `bind` if necessary.
But how to handle IPv4 vs. IPv6? Or by interface name?

andig, 2022-11-03:
Something like this https://www.linuxquestions.org/questions/linux-software-2/why-is-this-process-only-listening-on-tcp6-4175486989/ should work with having only the `bind` option?

andig, 2022-11-03:
Another interesting question is which listeners would need to honor this setting, please chose below:

- [ ] http
- [ ] SEMP (tied to http)
- [ ] modbus proxy (yes)
- [ ] Keba UDP (no???)
- [ ] Ocpp (no???)

The last two are interesting- those won't work if tied to localhost. What to do about those?

DerAndereAndi, 2022-11-03:
EEBUS settings does have something similar already.

```
  interfaces:
  - en5
```

Takes the interface names that should be used. If this is not set, it will use all interfaces. It is used for mDNS.

andig, 2022-11-03:
I don't think we want to go down the rabbit hole of per-service configuration after we've been able to do without any such feature for 2 years? One obvious solution imho is to just but evcc into a Docker container and forward the ports as needed.

premultiply, 2022-11-03:
I think we should leave it all binding to all interfaces.
If anybody needs more sophisticated setups imho it can be done by firewall rules, docker etc.

andig, 2022-11-03:
Either that or keep it simple- which would break SEMP (an acceptable trade-off). The change would be minimally invasive.

andig, 2022-11-07:
Seems we're not able to resolve the open questions. Closing as wontfix for time being.

closed 2022-11-07 by andig