evcc-io/evcc #5028: Fehler nach Update von 0.105.2 auf 0.106.0

matthiashu1963, 2022-11-02:
### Describe the bug

Die Konfiguration mit 0.105.2 hat ausgezeichnet funktioniert. Nach dem Update auf 0.106.0 kann ich nicht mehr starten und erhalte den Fehler: expected 32 or 64 characters machine id, got 4

### Steps to reproduce

starte evcc,

Fehlermeldung : expected 32 or 64 characters machine id, got 4

### Configuration details

network:  
  # schema is the HTTP schema  
  # setting to `https` does not enable https, it only changes the way URLs are generated  
  schema: http  
  # host is the hostname or IP address  
  # if the host name contains a `.local` suffix, the name will be announced on MDNS  
  # docker: MDNS announcements don't work. host must be set to the docker host's name.  
  host: evcc.local  
  # port is the listening port for UI and api  
  # evcc will listen on all available interfaces  
  port: 7070  
  
interval: 20s # control cycle interval standard 10  
sponsortoken: *****  
#telemetry: true # default: false  
plant: *****  
  
# log settings  
log: info  
levels:  
  site: debug  
  lp-1: debug  
  lp-2: debug  
  
meters:  
- name: grid1  
  type: template  
  template: solarlog   
  usage: grid    
  host: 192.168.1.212    
  port: 502    
- name: pv2  
  type: template  
  template: solarlog   
  usage: pv    
  host: 192.168.1.212    
  port: 502    
  
chargers:  
- name: wallbox3   
  type: template  
  template: go-e-v3   
  host: 192.168.1.3    
  
loadpoints:  
- title: Home Langwiesen  
  charger: wallbox3  
  vehicle:  Tessy  
  mode: off  
  phases: 0  
  enable:  
   threshold: 0  
   delay: 2m  
  mincurrent: 6  
  maxcurrent: 16  
  resetOnDisconnect: true # Einstellungen bei Ausstecken des Autos auf Standartwerte zurücksetzen  
  soc:  
#    min: 20 # Lade sofort auf 50%  
#    target: 80 # Lade bis max. 80% SoC  
  
  
site:  
  title: Home Langwiesen  
  meters:  
    grid: grid1  
    pvs:  
    - pv2  
  
vehicles:  
- name: Audi  
  type: template  
  template: etron  
  title: eTron  
  user: *****  
  password: *****  
  vin: *****  
  capacity: 82  
  phases: 3  
  mode: off  
  minSoC: 20  
  targetSoC: 80  
  minCurrent: 6  
  maxCurrent: 16  
  cloud: false  
  
- name: Tessy  
  type: template  
  template: tesla  
  title: Tessy Model Y Performance # Wird in der Benutzeroberfläche angezeigt # display name for UI  
  capacity: 79 # kwH  
#  OnIdentify:  
  targetSoC: 80  
  minSoC: 20  
  accessToken:  xxxxxxx
  refreshToken: xxxxx
#    vin: *****  
  
  
tariffs:  
  currency: CHF # (default EUR)  
  grid:  
    type: fixed  
    price: 0.184 # [currency]/kWh  
  
  feedin:  
    type: fixed  
    price: 0.08 # [currency]/kWh  
  

### Log details

[main  ] INFO 2022/11/02 20:27:38 evcc 0.106.0
[main  ] INFO 2022/11/02 20:27:38 using config file: /home/pi/evcc.yaml
[main  ] INFO 2022/11/02 20:27:38 listening at :7070
[main  ] FATAL 2022/11/02 20:27:38 expected 32 or 64 characters machine id, got 4
[main  ] FATAL 2022/11/02 20:27:38 will attempt restart in: 5m0s
[server] FATAL 2022/11/02 20:27:38 unlinkat /tmp/evcc: operation not permitted

### What type of operating system are you running?

Linux

### Version

evcc -v
evcc version 0.106.0

emcepe, 2022-11-03:
@matthiashu1963:

Bitte die Config in Zukunft (vielleicht diesmal noch per Edit ändern) am Anfang & Ende mit 3 Backticks einschließen, dann ist die Config deutlich besser lesbar :)

Siehe auch: [Creating and highlighting code blocks](https://docs.github.com/en/get-started/writing-on-github/working-with-advanced-formatting/creating-and-highlighting-code-blocks)

matthiashu1963, 2022-11-03:
Danke für den Tipp mit den ``` Backticks. :-)

andig, 2022-11-03:
Die Lösung ist einfach- bitte plant mit der entsprechenden Anzahl Zeichen einstellen! Wenns leer gelassen wir gibts einen Vorschlag vom System.

closed 2022-11-03 by andig