evcc-io/evcc #5033: i/o timeout Fehler

Poppstar1, 2022-11-03:
Hallo zusammen,
ich möchte evcc nutzen um E-Autos über PV-Anlagen aufzuladen. Ich habe ein Sunny SMA Home Manager 2.0, 4 Wechselrichter von SMA, 3 Wallboxen die ABL emH2 sind, wobei 1 Master und 2 Slaves, und ein RS485 zu Ethernet Adapter. Der Adapter ist an eine der Slave Wallboxen an einen Controlanschluss angeschlossen. Ich habe evcc auf einem Raspberry Pi installiert.
Jetzt habe ich das Problem, dass die Fehlermeldung "FATAL: cannot create charger 'my_charger1': cannot create charger 'template': cannot create charger 'abl': read tcp %RaspberryPiIP% -> %AdapterIP% i/o timeout" erscheint, wenn ich die evcc.yaml-Datei teste, und ich nicht mehr weiter weiß. Meine yaml-Datei:
network:
  schema: http
  host: evcc.local
  port: 7070

log: info
levels:
  cache: error

interval: 10s

meters:
- name: my_grid
  type: template
  template: sma-home-manager
  usage: grid  
  host: %Home-ManagerIP%

site:
  title: Gebäude
  meters:
    grid: my_grid
        
loadpoints:
- title: Container
  charger: my_charger1
  mode: pv
- title: Rampe
  charger: my_charger2
  mode: pv
- title: Halle
  charger: my_charger3
  mode: pv

chargers:
- name: my_charger1
  type: template
  template: abl
  modbus: rs485tcpip
  id: 1
  host: %AdapterIP%
  port: 502
- name: my_charger2
  type: template
  template: abl
  modbus: rs485tcpip
  id: 2
  host: %AdapterIP%
  port: 502
- name: my_charger3
  type: template
  template: abl
  modbus: rs485tcpip
  id: 3
  host: %AdapterIP%
  port: 502

sponsortoken: **********

Die Konfiguration des Adapters:
[image]

Außerdem gibt es eine Software der Herstellers von dem Adapter in der mir Geräte, in diesem Fall der Adapter, angezeigt werden. Dort heißt es: TCP Connection not established.

Kann mir jemand weiterhelfen?

premultiply, 2022-11-03:
Ja, wie immer:
Sämtliche "Intelligenz" und Protokollübersetzung des Konverters abschalten.
evcc spricht kein Modbus TCP und die Wallbox auch kein Modbus RTU sondern beide müssen transparent miteinander kommunizieren können.

Poppstar1, 2022-11-03:
Danke für die schnelle Antwort, werde das heute Mittag mal probieren.

Poppstar1, 2022-11-03:
Die Meldung Timeout kommt immer noch. Sind die Werte Flow Control = None, Multi-Master = Enable, Workmode = TCP Server korrekt?

premultiply, 2022-11-03:
Multi-Master und alles was mit Modbus zu tun hat bestimmt nicht.